zwave-js/zwave-js #2676: Coolcam Light switch 2 channel (EU-2) "does not support the CC assosiation" error

davidwaze, 2021-05-24:
(reopened the bug as it as uspposed to be solved into this version and i still have the problem)

Build/Run method

[ X ] Manually built (git clone - npm install - npm run build )
zwavejs2mqtt version: 4.3.1
zwave-js version: 7.5.1

Describe the bug
Interview fails, stay in "nodeinfo", log indicate [Node 196] Error during node interview: Node 196 (endpoint 1) does not support
the Command Class Association

Device : 600-4235-3 (0x0258-0x0003-0x108b)

Tried : multiple interviews without any success / remove device / factory reset device

Thanks for your support.

[logs.txt](https://github.com/zwave-js/node-zwave-js/files/6531259/logs.txt)
[image]

AlCalzone, 2021-05-24:
I'm not sure where this is coming from - can you please try the following?
Open `node_modules/zwave-js/build/lib/driver/Driver.js`, go to line 689, which should look like this:
```js
                this.controllerLog.logNode(node.id, `Error during node interview: ${e.message}`, "error");
```
and replace it with
```js
                this.controllerLog.logNode(node.id, `Error during node interview: ${e.stack}`, "error");
```

While you're at it, verify that `node_modules/zwave-js/build/lib/commandclass/MultiChannelAssociationCC.js`, line 503 looks like this:
```js
        if (assocAPI.isSupported() && assocGroupCount > mcGroupCount) {
```

Then restart, re-interview the node again and send me a log of that.

davidwaze, 2021-05-24:
Certainly a rookie question but i have the packaged version how to do to go inside then see the files ?

I tried to ipen it with 7zip but it fails

Thanks

AlCalzone, 2021-05-24:
If I can trust this
> Manually built (git clone - npm install - npm run build )

you should have a node_modules folder after `npm install` which contains the mentioned files.

davidwaze, 2021-05-25:
> I'm not sure where this is coming from - can you please try the following?
> Open `node_modules/zwave-js/build/lib/driver/Driver.js`, go to line 689, which should look like this: [...]

Hello,

I've installed the "unpackaged" version. zwavejs2mqtt 4.3.1 / zwave-js 7.52

I've replaced line into file Driver.js (line 689 replaced message by stack)

In file MultiChannelAssociationCC.js, line 503 look the same as yours.

And i tried to interview again the device, here's the result 
[logs.txt](https://github.com/zwave-js/node-zwave-js/files/6541770/logs.txt)

Thanks for your support.

J2Kbr, 2021-05-25:
I am having the same issue described by davidwaze (Coolcam Light switch EU-2):

`2021-05-25T20:43:45.675Z CNTRLR   [Node 006] Error during node interview: Node 6 (endpoint 1) does not support the Command Class Association!`

Running on Supervisioned Home Assistant in a Synology NAS VM.

Z-Wave JS Server: 1.6.0
Node Z-Wave JS: 7.5.2

AlCalzone, 2021-05-25:
> And i tried to interview again the device, here's the result

Perfect - the error was somewhere else. PR is up.

J2Kbr, 2021-05-25:
Not sure if will help, but the EU-2 manual indicates **Command Class Association** and **Command Class Multi Channel Association** are supported by the endpoints:

[image]

AlCalzone, 2021-05-25:
@J2Kbr interesting, the device tells us it doesn't - at least in the logs I've seen.
Nonetheless, this should be fixed on master now - I'll release 7.5.3 in a couple of minutes.

J2Kbr, 2021-05-26:
I am sorry to say, but the problem still present on:

Z-Wave JS Server: 1.7.0
Node Z-Wave JS: 7.6.0

AlCalzone, 2021-05-26:
(╯°□°）╯︵ ┻━┻

AlCalzone, 2021-05-26:
@J2Kbr can you do the first part I described here
https://github.com/zwave-js/node-zwave-js/issues/2676#issuecomment-847116566
and post a log where the error occurs?

J2Kbr, 2021-05-26:
Thank you for reopening.

After re-interview device, here is the log from `Home Assistant -> Supervisor -> Z-Wave JS -> Log`:

```
supported CCs:
                                  · Z-Wave Plus Info
                                  · Version
                                  · Manufacturer Specific
                                  · Device Reset Locally
                                  · Powerlevel
                                  · Association
                                  · Association Group Information
                                  · Multi Channel Association
                                  · Multi Channel
                                  · All Switch
                                  · Binary Switch
                                  · Configuration
                                  controlled CCs:
2021-05-26T22:07:48.880Z CNTRLR   [Node 006] Interview stage completed: NodeInfo
2021-05-26T22:07:48.882Z CNTRLR   [Node 006] Interviewing Manufacturer Specific...
2021-05-26T22:07:48.882Z CNTRLR » [Node 006] querying manufacturer information...
2021-05-26T22:07:49.000Z CNTRLR « [Node 006] received response for manufacturer information:
                                    manufacturer: Shenzhen Neo Electronics Co., Ltd. (0x0258)
                                    product type: 0x03
                                    product id:   0x8b
2021-05-26T22:07:49.001Z CNTRLR   [Node 006] Interviewing Version...
2021-05-26T22:07:49.001Z CNTRLR » [Node 006]   querying the CC version for Version...
2021-05-26T22:07:49.052Z CNTRLR   [Node 006]   supports CC Version (0x86) in version 2
2021-05-26T22:07:49.052Z CNTRLR » [Node 006] querying node versions...
2021-05-26T22:07:49.134Z CNTRLR « [Node 006] received response for node versions:
                                    library type:      Routing Slave (0x06)
                                    protocol version:  4.33
                                    firmware versions: 2.60
                                    hardware version:  16
2021-05-26T22:07:49.134Z CNTRLR » [Node 006] querying CC versions...
2021-05-26T22:07:49.134Z CNTRLR » [Node 006]   querying the CC version for Binary Switch...
2021-05-26T22:07:49.253Z CNTRLR   [Node 006]   supports CC Binary Switch (0x25) in version 1
2021-05-26T22:07:49.253Z CNTRLR   [Node 006]   skipping query for All Switch (0x27) because max implemented vers
                                  ion is 0
2021-05-26T22:07:49.253Z CNTRLR » [Node 006]   querying the CC version for Z-Wave Plus Info...
2021-05-26T22:07:49.302Z CNTRLR   [Node 006]   supports CC Z-Wave Plus Info (0x5e) in version 2
2021-05-26T22:07:49.303Z CNTRLR » [Node 006]   querying the CC version for Manufacturer Specific...
2021-05-26T22:07:49.356Z CNTRLR   [Node 006]   supports CC Manufacturer Specific (0x72) in version 2
2021-05-26T22:07:49.357Z CNTRLR   [Node 006]   skipping query for Device Reset Locally (0x5a) because max implem
                                  ented version is 1
2021-05-26T22:07:49.358Z CNTRLR   [Node 006]   skipping query for Powerlevel (0x73) because max implemented vers
                                  ion is 0
2021-05-26T22:07:49.359Z CNTRLR » [Node 006]   querying the CC version for Association...
2021-05-26T22:07:49.408Z CNTRLR   [Node 006]   supports CC Association (0x85) in version 2
2021-05-26T22:07:49.409Z CNTRLR » [Node 006]   querying the CC version for Association Group Information...
2021-05-26T22:07:49.485Z CNTRLR   [Node 006]   supports CC Association Group Information (0x59) in version 1
2021-05-26T22:07:49.485Z CNTRLR » [Node 006]   querying the CC version for Multi Channel Association...
2021-05-26T22:07:49.532Z CNTRLR   [Node 006]   supports CC Multi Channel Association (0x8e) in version 3
2021-05-26T22:07:49.532Z CNTRLR » [Node 006]   querying the CC version for Multi Channel...
2021-05-26T22:07:49.579Z CNTRLR   [Node 006]   supports CC Multi Channel (0x60) in version 4
2021-05-26T22:07:49.580Z CNTRLR » [Node 006]   querying the CC version for Configuration...
2021-05-26T22:07:49.649Z CNTRLR   [Node 006]   supports CC Configuration (0x70) in version 1
2021-05-26T22:07:49.665Z CNTRLR   [Node 006] Embedded device config loaded
2021-05-26T22:07:49.669Z CNTRLR   [Node 006] Interviewing Z-Wave Plus Info...
2021-05-26T22:07:49.670Z CNTRLR » [Node 006] querying Z-Wave+ information...
2021-05-26T22:07:49.841Z CNTRLR « [Node 006] received response for Z-Wave+ information:
                                  Z-Wave+ version: 1
                                  role type:       AlwaysOnSlave
                                  node type:       Node
                                  installer icon:  0x1d02
                                  user icon:       0x1d02
2021-05-26T22:07:49.847Z CNTRLR   [Node 006] Interviewing Multi Channel...
2021-05-26T22:07:49.847Z CNTRLR » [Node 006] querying device endpoint information...
2021-05-26T22:07:49.955Z CNTRLR « [Node 006] received response for device endpoints:
[... 11 lines cut ...]
                                  supported CCs:
                                    · Z-Wave Plus Info
                                    · Version
                                    · Multi Channel Association
                                    · Binary Switch
2021-05-26T22:07:50.140Z CNTRLR   [Node 006] all endpoints identical, skipping capability query for endpoint #2.
                                  ..
[... cut ...]
2021-05-26T22:07:50.313Z CNTRLR   [Node 006] Interviewing Multi Channel Association...              [Endpoint 1]
2021-05-26T22:07:50.314Z CNTRLR » [Node 006] querying number of multi channel association groups... [Endpoint 1]
2021-05-26T22:07:50.431Z CNTRLR « [Node 006] supports 2 multi channel association groups            [Endpoint 1]
2021-05-26T22:07:50.432Z CNTRLR   [Node 006] Error during node interview: Node 6 (endpoint 1) does not support t
                                  he Command Class Association!
```

AlCalzone, 2021-05-27:
Sorry but that log alone doesn't help. Above I asked for an edit to the file so the error message contains a stack trace.

AlCalzone, 2021-05-27:
I don't know either. It's one of those:
```
/mnt/data/docker/overlay2/612f50ad1d11752037fcd1bf7a14adaa0b490b96fe460e5f7ce4ff740cdce567/merged/usr/src/node_modules/zwave-js/build/lib/driver/Driver.js

/mnt/data/docker/overlay2/44e6f59bcef06dc9d21eec570d80da042464c0ea62285b7ffaebef9f9ab0c9db/diff/usr/src/node_modules/zwave-js/build/lib/driver/Driver.js

/var/lib/docker/overlay2/612f50ad1d11752037fcd1bf7a14adaa0b490b96fe460e5f7ce4ff740cdce567/merged/usr/src/node_modules/zwave-js/build/lib/driver/Driver.js

/var/lib/docker/overlay2/44e6f59bcef06dc9d21eec570d80da042464c0ea62285b7ffaebef9f9ab0c9db/diff/usr/src/node_modules/zwave-js/build/lib/driver/Driver.js
```

J2Kbr, 2021-05-27:
Here is the log (only the relevant part) after editing `Driver.js`. Hope it helps find the cause of the issue.

```
2021-05-27T17:15:27.485Z CNTRLR   [Node 006] Error during node interview: ZWaveError: Node 6 (endpoint 1) does n
                                  ot support the Command Class Association!
                                      at Object.get (/usr/src/node_modules/zwave-js/src/lib/node/Endpoint.ts:326
                                  :12)
                                      at Proxy.withOptions (/usr/src/node_modules/zwave-js/src/lib/commandclass/
                                  API.ts:239:12)
                                      at MultiChannelAssociationCC.refreshValues (/usr/src/node_modules/zwave-js
                                  /src/lib/commandclass/MultiChannelAssociationCC.ts:649:56)
                                      at MultiChannelAssociationCC.interview (/usr/src/node_modules/zwave-js/src
                                  /lib/commandclass/MultiChannelAssociationCC.ts:463:14)
                                      at interviewEndpoint (/usr/src/node_modules/zwave-js/src/lib/node/Node.ts:
                                  1490:5)
                                      at ZWaveNode.interviewCCs (/usr/src/node_modules/zwave-js/src/lib/node/Nod
                                  e.ts:1653:20)
                                      at ZWaveNode.interview (/usr/src/node_modules/zwave-js/src/lib/node/Node.t
                                  s:1214:8)
                                      at Driver.interviewNode (/usr/src/node_modules/zwave-js/src/lib/driver/Dri
                                  ver.ts:895:10)
                                      at /usr/src/node_modules/zwave-js/src/lib/driver/Driver.ts:841:7
```

J2Kbr, 2021-05-27:
I edited the file `Endpoint.ts` to also print the `property` causing _throw_, as the following:

```
throw new ZWaveError(
    `Node ${this.nodeId}${
        this.index === 0 ? "" : ` (endpoint ${this.index})`
    } does not support the Command Class ${ccName}! [${property}]`,
    ZWaveErrorCodes.CC_NotSupported,
);
```

This is the result:
`2021-05-27T18:04:11.382Z CNTRLR   [Node 006] Error during node interview: ZWaveError: Node 6 (endpoint 1) does n                          ot support the Command Class Association! [commandOptions]`

The `property` is: **commandOptions**

Hopefully this info will be of some help.

J2Kbr, 2021-05-27:
Success!! 

Adding: `property !== "commandOptions" &&` in the conditional at `Endpoint.ts` makes the interview succeed with the 
Coolcam Light switch 2 channel (EU-2)! :)

AlCalzone, 2021-05-27:
let me check where that comes from

closed 2021-05-28 by AlCalzone

davidwaze, 2021-06-01:
Working fine !!!!!!!!!! thank you for all !

PR #2720 (2021-05-28): fix: whitelist commandOptions as unchecked CCAPI property
fixes: #2676
(for real this time)