zwave-js/zwave-js #2677: Create a compat flag to Ring Keypad v1 due to duplicate entry control notification sequences

xhemp, 2021-05-24:
I’ve never seen this before but my Ring keypad v1 has been seeing duplicate sequences even when connected to mains (before that, it would only happen when on battery):
`
[Node 057] Received duplicate Entry Control Notification (sequence number 1),ignoring...
[Node 057] Received duplicate Entry Control Notification (sequence number 2),ignoring...
[Node 057] Received duplicate Entry Control Notification (sequence number 3),ignoring...

`
Would it be doable to do a compat flag (ie: ignoreEntryControlSequenceNumber) for it to ignore the duplicates instead of patching the drivers on every release (like I’m doing it now, thanks @marcus-j-davies for the help)?

blhoward2, 2021-05-24:
We have tried to avoid creating device specific compat flags this far. Does this notification event change the state of an entity in HA? Or does it come across as a value_notification or something?

xhemp, 2021-05-24:
Problem is that it doesn't show what I actually pressed on the keypad so HA doesn't even know what is coming from it.  I didn't send in any logs as I would like to know if you would even consider it as if not, I would be patching the driver file every time.  I am using the latest version of zwavejs (7.5.1) and I am attaching a debug I've done a couple of weeks ago.
[keypadv1-debug.txt](https://github.com/zwave-js/node-zwave-js/files/6532473/keypadv1-debug.txt)

marcus-j-davies, 2021-05-24:
@blhoward2,

I think some context might help here.

**Entry Control** is relatively new in Z-Wave JS, I know of very few devices, that provide this CC, its the ring Keypad, that first made use of it - as far as I am aware.

Now then,
The **Entry Control** specification states, that the `sequenceNumber` part of notifications must be unique, but between the range of 0-255 🤷.

The ring pad keypad, has already shown it is not compliant, as there is compat flag for `disableStrictEntryControlDataValidation` already in the config.

It turns out, the device likes to send duplicated `sequenceNumber`s  also.
Z-Wave JS ignores these duplicates (providing they have been used in the last 5 notifications) - and rightfully so.

However, it means, the notifications are not being forwarded to the front end 

Much like `disableStrictEntryControlDataValidation`, it maybe of discussion, to also create a  `ignoreEntryControlSequenceNumber` to deal with Entry Control devices  that don't follow the proper spec.

As mentioned, the way around it currently, is to comment out the duplication checking altogether (for this CC).
which is not the best approach obviously, but if a new compat flag could be added, that can be applied to other device configs that may come out, with the same disregard to spec.

blhoward2, 2021-05-24:
> 
> 
> 
> 
> Problem is that it doesn't show what I actually pressed on the keypad so HA doesn't even know what is coming from it.  I didn't send in any logs as I would like to know if you would even consider it as if not, I would be patching the driver file every time.  I am using the latest version of zwavejs (7.5.1) and I am attaching a debug I've done a couple of weeks ago and a dump of that node.
> 
> [dump_ringkeypad_v1.json.txt](https://github.com/zwave-js/node-zwave-js/files/6532500/dump_ringkeypad_v1.json.txt)
> 
> [keypadv1-debug.txt](https://github.com/zwave-js/node-zwave-js/files/6532473/keypadv1-debug.txt)
> 
> 

Just to confirm, are these logs from the project code or your patched version?

xhemp, 2021-05-24:
> Just to confirm, are these logs from the project code or your patched version?

The debug was done before I patched it.

blhoward2, 2021-05-24:
So I guess one question would be, should this be part of the current compat flag versus a separate one? I haven't looked but what data validation are we disabling as this seems like data validation too?

marcus-j-davies, 2021-05-24:
`Entry Control` specs, states that the ASCII/DATA payload of notifications needs to have padding
(as its a security payload, to introduce entropy I suppose).

But this device (and maybe more) don't add the required padding, hence the compat flag.
Having it apart of the current flag - would indeed work, and will remove the need for another flag.

but I'm not sure if it should be 2 separate areas that need some assistance - there maybe devices hat  follow the padding requirements , but not the `sequenceNumber `runs

xhemp, 2021-05-24:
> @xhemp Did this ever work under any version? Or is the change you describe from switching to mains power?

I have never seen this particular  issue with the device on mains, first time I've seen was thia morning (I had updated from 7.5.0 to 7.5.1 but don't think it's related). If the keypad is on batteries, it happens all the time which makes it unusable.

marcus-j-davies, 2021-05-24:
For what it's worth, I just don't  think the sequenceNumber rules are strictly followed, and end up acting differently, if mains powered.

The device it's self, has issues with the requirements in data validation.

I can't say, if other Entry Control devices have the same problems, but it won't be too much of a surprise if not.

AlCalzone, 2021-05-24:
@xhemp could you share a bit more logging from **before** the first notification gets ignored? It does seem like the device is counting up. Maybe something caused it to lose power and restart from 1?

xhemp, 2021-05-24:
> @xhemp could you share a bit more logging from **before** the first notification gets ignored? It does seem like the device is counting up correctly. Maybe something caused it to lose power and restart from 1?

Trying now, patch has been undone and I'm running on debug only on that device, it may take sometime to reappear, though.

xhemp, 2021-05-24:
Tested for the last 20 minutes and the duplicated showed up again (a few times), in the end of the log file, you can see I try to arm away then arm home and it didn't work then disarmed just worked fine.
[new_log.txt](https://github.com/zwave-js/node-zwave-js/files/6533518/new_log.txt)

marcus-j-davies, 2021-05-24:
@AlCalzone,

Here is another (Mine went from Sequence 1 -> 2 -> 1) towards the end
[zwavelog-2.txt](https://github.com/zwave-js/node-zwave-js/files/6533525/zwavelog-2.txt)

AlCalzone, 2021-05-24:
Yeah, that looks like some kind of reset in the device or something. I'll disable this check as part of `disableStrictEntryControlDataValidation`, as the device is more or less unusable without it.

marcus-j-davies, 2021-05-24:
Specifications  state a range from 1-255, but I wonder if there is any rule as to when the counter is reset. 🤔.
I'm not hot on terminology with specifications, so could be clutching  at straws

PR #2682 (2021-05-24): feat(compat): disable sequence number check in existing compat flag
With this PR, the existing compat flag `disableStrictEntryControlDataValidation` also disables the sequence number check.

fixes: #2677
closed 2021-05-24 by AlCalzone