zwave-js/zwave-js #2683: Weiser - SmartCode 10 Touchscreen Electronic Deadbolt

Chris-V, 2021-05-24:
**Which device is missing?**

Weiser SmartCode 10 Touch Contemporary

**What are the IDs?**

Manufacturer ID: 0x0090  
Product type: 0x0006
Product ID: 0x0642

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
Here is the ZWave Alliance page for this device:
https://products.z-wavealliance.org/products/2351?selectedFrequencyId=2

Weiser is the Canadian brand for Kwikset. AFAICT, this is the same device as this one: https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0090/916.json, with a different manufacturer and product name / code. Is it better to copy the Kwikset's config?

marcus-j-davies, 2021-05-24:
There is a very similar device (GED1800). 

https://devices.zwave-js.io/?jumpTo=0x0090:0x0006:0x0440:0.0

Do you know if they are the same?

marcus-j-davies, 2021-05-24:
Ok,

Does all the config params match?

Chris-V, 2021-05-24:
Yes, they match the configs listed in the manual: https://products.z-wavealliance.org/ProductManual/File?folder=&filename=MarketCertificationFiles/2351/Install%20Guide%20GED2150%2004112017.pdf

Should the Kwikset config be duplicated for Weiser?

marcus-j-davies, 2021-05-24:
Yes,

Fancy making a PR?
if not, I can do this now?

Chris-V, 2021-05-24:
I don't have the repo cloned. If you are already set up you can go ahead and PR this. Thanks a lot :)

closed 2021-05-24 by AlCalzone