zwave-js/zwave-js #2692: [bug] NAS-WR01ZE: kWh meter value displayed as negative number

andre-richter, 2021-05-25:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt version: 4.3.1
zwave-js version: 7.5.1

### Describe the bug

The `kWh` meter of the `NAS-WR01ZE` (https://devices.zwave-js.io/?jumpTo=0x0258:0x0200:0x1027:2.32) smart plug is reported/displayed as a negative number.

After resetting the accumulated meter values, sane positive values are reported for sometime, but at some point it flips to negative.

zwavejs2mqtt.log:
```
2021-05-24 20:46:26.806 DEBUG SOCKET: Event ZWAVE_API emitted to FHb1fJMDSyU3lbGJAAAB
2021-05-24 20:46:26.808 INFO ZWAVE: Calling api refreshCCValues with args: [ 42, 50, [length]: 2 ]
2021-05-24 20:46:26.971 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.13 => -21474836.13
2021-05-24 20:46:27.120 INFO ZWAVE: Node 42: value updated: 50-0-value-66049 27.85 => 27.49
2021-05-24 20:46:27.266 INFO ZWAVE: Node 42: value updated: 50-0-value-66561 234.62 => 234.93
2021-05-24 20:46:27.410 INFO ZWAVE: Node 42: value updated: 50-0-value-66817 0.25 => 0.25
```
zwavejs.log:
```
2021-05-24T20:46:26.811Z CNTRLR » [Node 042] querying meter value (type = Electric, scale = kWh, rate type = Con
                                  sumed)...
2021-05-24T20:46:26.975Z CNTRLR » [Node 042] querying meter value (type = Electric, scale = W, rate type = Consu
                                  med)...
2021-05-24T20:46:27.123Z CNTRLR » [Node 042] querying meter value (type = Electric, scale = V, rate type = Consu
                                  med)...
2021-05-24T20:46:27.268Z CNTRLR » [Node 042] querying meter value (type = Electric, scale = A, rate type = Consu
                                  med)...
```

The negative value also gets picked up by Home Assistant.

### To Reproduce

Let the device report for some time.

### Expected behavior

A sane number being reported.

### Additional context

*I am unsure if this is a SW issue or a quirk with the device itself, since it looks like bit 31 is being flipped.*
The screenshot below was made not too long after a reset of the accumulated values. If I take the negative number and flip bit 31 from 1 to 0, we get `35`, which would be `0.35` kWh I presume, and that would make sense given that a reset was done earlier.

This is happening on multiple of the same devices I own, not only a single one.

[image]

AlCalzone, 2021-05-25:
@andre-richter any chance you could catch this in a zwave-js log with loglevel debug? I wonder if it is **only** that bit.

andre-richter, 2021-05-25:
@AlCalzone, are you particularly interested in the event of flipping from sane to negative value or something else?
I could set up that experiment later, and increase the reporting interval to report every minute or so.

In the meantime, another experiment completed, giving more insight. I have one of those devices connected to a consumer that constantly drains ~`27 W`. I had set the reporting interval to 10 minutes before resetting:
```
2021-05-25 08:38:43.108 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0 => 0
2021-05-25 08:48:43.034 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0 => 0
2021-05-25 08:58:42.973 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0 => 0
2021-05-25 09:08:42.914 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0 => -21474836.46
2021-05-25 09:18:42.857 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.46 => -21474836.46
2021-05-25 09:28:42.795 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.46 => -21474836.44
2021-05-25 09:38:43.739 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.44 => -21474836.44
2021-05-25 09:48:43.679 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.44 => -21474836.42
2021-05-25 09:58:43.618 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.42 => -21474836.42
2021-05-25 10:08:43.557 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.42 => -21474836.4
2021-05-25 10:18:43.503 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.4 => -21474836.4
2021-05-25 10:28:43.444 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.4 => -21474836.4
2021-05-25 10:38:43.388 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.4 => 0.1
2021-05-25 10:48:43.329 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0.1 => 0.1
2021-05-25 10:58:43.271 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0.1 => -21474836.36
```

Exactly after 2 hours, it incidentally switched back to a sane value. 
We would expect to see `2 h * 27 W = 54 Wh`, or `0.05 kWh` respectively. The reported value is **double**, which seems in line with what was reported here: https://community.home-assistant.io/t/energy-consumption-for-neo-coolcam-plug/219844/8.

Taking the last timestamp, which is after 140 minutes, we would expect `140 minutes * 27 W = 63 Wh`, aka `0.06 kWh`

The reported value is `-21474836.36`.
Naively converting this (not sure if this is valid, aka what the actual number representation used in the zwave value is):
```
-21474836.36 * 100 = -2147483636
-2147483636 & 0x7fff_ffff = 12
12 / 100 = 0.12 kWh
```

Which would again be *double* the expected value.

AlCalzone, 2021-05-25:
I'm interested in the raw commands that get sent - best case some correct ones and some flipped ones. Make sure to set the loglevel of the zwave-js logger to debug, so we can see this.

> Naively converting this (not sure if this is valid, aka what the actual number representation used in the zwave value is):

Seems correct, in Z-Wave floats are just encoded as integers plus precision (# of decimal places).

Based on the history of values I think this device has some big problems with encoding values that aren't divisible by 0.1.
```
        0.00 => 0.00 ✔
-21474836.46 => 0.02
-21474836.44 => 0.04
-21474836.42 => 0.06
-21474836.40 => 0.08
        0.10 => 0.10 ✔
-21474836.36 => 0.12
```

> Which would again be double the expected value.

Not much we can do about that if the device reports this.

andre-richter, 2021-05-25:
> Based on the history of values I think this device has some big problems with encoding values that aren't divisible by 0.1.
> 
> ```
>         0.00 => 0.00 ✔
> -21474836.46 => 0.02
> -21474836.44 => 0.04
> -21474836.42 => 0.06
> -21474836.40 => 0.08
>         0.10 => 0.10 ✔
> -21474836.36 => 0.12
> ```

Some more from this run:
```
2021-05-25 11:08:43.212 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.36 => -21474836.36
2021-05-25 11:18:43.157 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.36 => -21474836.34
2021-05-25 11:28:43.095 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.34 => -21474836.34
2021-05-25 11:38:43.037 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 -21474836.34 => 0.16
2021-05-25 11:48:42.981 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0.16 => 0.16
2021-05-25 11:58:42.924 INFO ZWAVE: Node 42: value updated: 50-0-value-65537 0.16 => 0.18
```

> > Which would again be double the expected value.
> 
> Not much we can do about that if the device reports this.

So some correcting math on the value on retrieval is not feasible? I can hack it into my home-assistant installation, but before I go there, I'm trying to understand to what extend zwave_js is the correct place to correct such awful quirks.

Trying to set up your experiment now.

AlCalzone, 2021-05-25:
> I'm trying to understand to what extend zwave_js is the correct place to correct such awful quirks.

Generally we do not try to fix application level device quirks in the driver for two reasons:
1. While these values seem obviously wrong, it is impossible to tell automatically whether the value was reported like this on purpose or not
2. It is not our job to clean up the manufacturers' mess. Sure, we work around some things to make the devices usable at all, but fiddling around with the reported values doesn't belong in this project.

Other projects (like SmartThings) have done these things by writing device-specific integrations, which in the end just leads to the manufacturers saying _"but it works in XYZ 🤷🏻‍♂️"_ when in reality they are just too lazy to fix their shit. The stuff above would have been caught by QA if they bothered to test it. Maybe having a majority of their users complain because their stuff does not work with a standards-compliant software will change that mentality.

Some manufacturers do care - I've already seen a case where we had a beta firmware version to test just hours after a bug report.

andre-richter, 2021-05-25:
I don't know what is different this time. Currently having a run from `0` to `0.14 kWh` for almost 3 hours without any negative value. Will update once there is a repro.

andre-richter, 2021-05-26:
@AlCalzone logs at: https://send.tresorit.com/a#wHJq_Ng_rc3XDLPmKNw6XQ

AlCalzone, 2021-05-26:
Oh, the messages are encrypted. I need your network key to decode them - you can send it via mail to the one in my profile.

andre-richter, 2021-05-26:
info@zwave-js.io ?

AlCalzone, 2021-05-26:
It actually is just the one bit that flips. This seems to be the same in the other bug reports that can be found all over Google.

If you ask me, you should complain to the manufacturer about this and ask for a firmware update or return it or something.

andre-richter, 2021-05-26:
Fair enough, I'll work around it in SW on my local HA overlay. I have little hope to see FW updates for such budget devices, but lets see (the first of the two support email addresses already replied my email wasn't delivered because the inbox quota was full :sweat_smile: )

closed 2021-05-26 by andre-richter

Minims, 2021-12-30:
@AlCalzone Do you think a workaround like that can be implemented ?  
 https://github.com/domoticz/open-zwave/pull/3

blhoward2, 2021-12-30:
As a rule we don't implement device-specific workarounds. Doing so is part of what led to OZW's bloated code. You can template a sensor entity in HA on your own to fix this.

andre-richter, 2022-01-14:
I would love to have something like that as well. 👍👍

In the current case, all we would need is to apply some integer math on the raw device value before passing it on to home assistant or whoever reads it.

A hook in zwavejs2mqtt sounds reasonable?

andre-richter, 2022-01-14:
I have a similar use case for a roller shutter that never reaches fully closed or open in some situations.
I keep o fixing that locally in HA (https://github.com/andre-richter/core/commit/485b3e78a8be12d54c20d75fc0b7891b325bd310) but a hook in the server would do here as well.

AlCalzone, 2022-01-14:
> there might be a better approach from a overall smart home architecture perspective than making it a problem of the users (either buy new devices or implementing workarounds)

This may sound cynical but I disagree. It really should become a problem of the **manufacturers** who release these buggy devices and don't offer any support whatsoever. The huge bunch of workarounds that other drivers implemented (some still do, see SmartThings) make it too easy for manufacturers to reject responsibility and claim _"but it works in XYZ, cannot be our fault!"_

I can understand the sentiment of making things work when you're using unofficial APIs etc, but Z-Wave is a standard, devices have to be certified and these things still happen too much. Only when we stop solving the manufacturers' mistakes for them and make them fix them, then we can get to a point where it becomes a benefit for the smart home community.

Unfortunately, I don't have the power to make manufacturers do that, so all I can do here is stay strong and say _"I will not fix this"_ (and neither should the other projects), even if it is annoying for users. In the end, the **users have the power to make a change**:
1. When there are issues with a device, **contact the manufacturer support**. Open Source projects are not free tech support for manufacturers.
2. **Hold them accountable**. If there is clearly an issue with the device firmware, demand them to fix it. Even if they initially claim its not their fault. European law recently even made it mandatory to provide updates for ~5 years.
3. **Vote with your wallet**:
    - Don't buy from manufacturers that don't provide any support
    - Don't buy from manufacturers that don't provide firmware updates or make you pay for them. Nowadays there is really not a reason not to provide them online.
    - Don't buy from manufacturers that tend to seek fault elsewhere (_"The device is certified, cannot be our fault..."_)
    - Inform yourself before you buy
    - DO buy from manufacturers that have reliable devices and give good support.

We can help with the latter - we recently added the ability to display comments in our device configuration DB, but it isn't used much yet:
[image]
https://devices.zwave-js.io/?jumpTo=0x0258:0x0100:0x1027:0.0

---

Side note: I currently have devices from 3 large manufacturers. Of these 3, I'd only buy from one again and my DO-NOT-BUY list is growing rapidly just by looking at the issues that are reported here.

andre-richter, 2022-01-14:
I just checked https://zwave-js.github.io/zwavejs2mqtt/#/usage/setup?id=device-values-configuration some more and it seems this would be almost a solution. I think not all problems could be solved with math operations only.

And sometimes you might not want this for ALL devices of the same type (that's what it does?), but just some specific devices?

@AlCalzone since this is in the realm of zwavejs2mqtt, would you be opposed to exploring this more as well? Or is this anyways not "your domain" but @robertsLando's ? :)

AlCalzone, 2022-01-14:
@robertsLando Can do what he likes

robertsLando, 2022-01-14:
For the ones using mqtt devices configuration table can already be used to create a custom send and write functions

ahochsteger, 2022-01-14:
> For the ones using mqtt devices configuration table can already be used to create a custom send and write functions

@robertsLando would that only affect values sent via MQTT or also affect values sent to HA using the server port integration (posibly not).

KloudJack, 2022-01-25:
I've created a [Template Sensor](https://www.home-assistant.io/integrations/template/) that does corrective calculations on the values reported by NAS-WR01ZE devices:

https://gist.github.com/KloudJack/4489ffcbeae58b88072f391dca55b40d

I've been using this successfully with my power sockets until a firmware fix is released, not holding my breath though as this issue has been known for at least 2 years.

heapxor, 2022-03-22:
Hi,
 i am not sure if the template fix the issue? Once applied i can see results as 
[image]

the bottom one is the value from the template.

MarijnS95, 2022-03-22:
@heapxor That gist also mentions this in the top-level comment:

```yaml
# In addition to random most significant bit flips in the 32-bit energy consumption value reported
# which results in massive negative values, the NAS-WR01ZE also generally computes the energy consumed incorrectly.
# In brief practical tests the actual consumption is actually ~1/4 of the value reported.
```

It seems to have deduced a correction factor of 0.26 which when multiplied with the original value of `18.48` results in `4.8048`:

```jinja2
        {% set correctionFactor = 0.26 %}
...
          {{ ( value * correctionFactor ) | round(roundDecimals) }}
```

heapxor, 2022-03-22:
@MarijnS95 ah okay, now i see the point, in that case it seems to be okay. thx.

heapxor, 2022-03-28:
hello,
i think this is another bug node-zwave-js bug? https://community.home-assistant.io/t/fgwp102-zw5-v3-2-bug-in-reports/406169

blhoward2, 2022-03-28:
Don't comment on a closed issue. Open a new bug report and fill out the template.

heapxor, 2022-03-28:
okay its here > https://github.com/zwave-js/node-zwave-js/issues/4408 

thanks

joshuisken, 2024-07-08:
> I've created a [Template Sensor](https://www.home-assistant.io/integrations/template/) that does corrective calculations on the values reported by NAS-WR01ZE devices: [...]

This gist has disappeared, anyone have a copy?