zwave-js/zwave-js #2700: [bug] Yale Conexis L1 dead on 4.4.0

lolouk44, 2021-05-26:
### Version

**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [x] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt version: <version here, you can find it on control panel>
zwave-js version: <version here, you can find it on control panel>

### Describe the bug

Although an HA user, issue doesn't appear to be HA related.
My container got updated to :latest tag today, since my Yale Conexis ends up dead.
I can bring it back to Alive by manually unlocking it, but when I manually lock it back up it ends up dead again
Reverted to 4.3.1 and it works again...

### To Reproduce

Steps to reproduce the behavior:

1. Manually unlock Conexis
2. Manually lock Conexis

### Expected behavior

lock does not become dead

### Additional context

I have tried to heal the network and tried to update the lock's data to no avail.
I've also tried to restart the server/container.
Here are some logs if that helps:
```
2021-05-25 18:12:41.117 INFO APP: GET /api/settings 304 64.731 ms - -
2021-05-25 18:12:41.129 INFO APP: GET /api/auth-enabled 304 2.097 ms - -
2021-05-25 18:12:56.212 INFO ZWAVE: Calling api isFailedNode with args: [ 1, [length]: 1 ]
2021-05-25 18:12:56.224 INFO ZWAVE: Calling api isFailedNode with args: [ 5, [length]: 1 ]
2021-05-25 18:12:56.239 INFO ZWAVE: Success zwave api call isFailedNode { success: true, message: 'Success zwave api call', result: false }
2021-05-25 18:12:56.250 INFO ZWAVE: Success zwave api call isFailedNode { success: true, message: 'Success zwave api call', result: true }
2021-05-25 18:13:20.520 INFO ZWAVE: Calling api beginHealingNetwork with args: [ [length]: 0 ]
2021-05-25 18:13:20.532 INFO ZWAVE: Controller status: Healing process IN PROGRESS. Healed 1 nodes
2021-05-25 18:13:20.535 INFO ZWAVE: Success zwave api call beginHealingNetwork { success: true, message: 'Success zwave api call', result: true }
2021-05-25 18:13:20.543 INFO ZWAVE: Controller status: Healing process COMPLETED. Healed 1 nodes
2021-05-25 18:13:29.442 INFO ZWAVE: Calling api isFailedNode with args: [ 1, [length]: 1 ]
2021-05-25 18:13:29.453 INFO ZWAVE: Calling api isFailedNode with args: [ 5, [length]: 1 ]
2021-05-25 18:13:29.467 INFO ZWAVE: Success zwave api call isFailedNode { success: true, message: 'Success zwave api call', result: false }
2021-05-25 18:13:29.476 INFO ZWAVE: Success zwave api call isFailedNode { success: true, message: 'Success zwave api call', result: true }
2021-05-25 18:13:54.991 INFO ZWAVE: Calling api isFailedNode with args: [ 5, [length]: 1 ]
2021-05-25 18:13:55.010 INFO ZWAVE: Success zwave api call isFailedNode { success: true, message: 'Success zwave api call', result: true }
2021-05-25 18:14:02.297 INFO ZWAVE: Calling api refreshValues with args: [ 5, [length]: 1 ]
2021-05-25 18:14:09.430 INFO ZWAVE: Success zwave api call refreshValues { success: true, message: 'Success zwave api call', result: undefined }
2021-05-25 18:17:28.099 INFO ZWAVE: Node 5: has returned from the dead
2021-05-25 18:17:28.186 INFO ZWAVE: Node 5: value added 5-98-0-currentMode => 255
2021-05-25 18:17:28.189 INFO ZWAVE: Node 5: value added 5-98-0-targetMode => 0
2021-05-25 18:17:28.191 INFO ZWAVE: Node 5: value added 5-98-0-outsideHandlesCanOpenDoor => false,false,false,false
2021-05-25 18:17:28.191 INFO ZWAVE: Node 5: value added 5-98-0-insideHandlesCanOpenDoor => false,false,false,false
2021-05-25 18:17:28.192 INFO ZWAVE: Node 5: value added 5-98-0-latchStatus => open
2021-05-25 18:17:28.192 INFO ZWAVE: Node 5: value added 5-98-0-boltStatus => locked
2021-05-25 18:17:28.193 INFO ZWAVE: Node 5: value added 5-98-0-doorStatus => closed
2021-05-25 18:17:28.194 INFO ZWAVE: Node 5: value added 5-98-0-lockTimeout => 0
2021-05-25 18:17:28.194 INFO ZWAVE: Node 5: value added 5-98-0-operationType => 1
[... 188 lines cut ...]
2021-05-25 18:24:17.206 INFO ZWAVE: Node 5: value updated: 98-0-doorStatus closed => open
2021-05-25 18:24:18.059 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 0 => 0
2021-05-25 18:24:18.065 INFO ZWAVE: Node 5 CC Notification {
type: 6,
event: 2,
label: 'Access Control',
eventLabel: 'Manual unlock operation',
parameters: undefined
}
2021-05-25 18:24:36.049 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 0 => 255
2021-05-25 18:24:36.051 INFO ZWAVE: Node 5: value updated: 98-0-outsideHandlesCanOpenDoor true,false,false,false => false,false,false,false
2021-05-25 18:24:36.052 INFO ZWAVE: Node 5: value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-05-25 18:24:36.054 INFO ZWAVE: Node 5: value updated: 98-0-latchStatus open => open
2021-05-25 18:24:36.055 INFO ZWAVE: Node 5: value updated: 98-0-boltStatus unlocked => locked
2021-05-25 18:24:36.057 INFO ZWAVE: Node 5: value updated: 98-0-doorStatus open => closed
2021-05-25 18:24:36.058 INFO ZWAVE: Node 5: value updated: 98-0-lockTimeout 0 => 0
2021-05-25 18:24:37.038 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 255 => 255
2021-05-25 18:24:37.040 INFO ZWAVE: Node 5 CC Notification {
type: 6,
event: 1,
label: 'Access Control',
eventLabel: 'Manual lock operation',
parameters: undefined
}
```

lolouk44, 2021-05-28:
Sorry for the delay. How do I get a driver log?
I was testing again and it doesn't seem to be specifically related to 4.4.0
Here is how I can reproduce it:
1) manually unclock (don't pull the handle down)
2) manually lock (pull the handle up)
3) lock becomes available
4) even though the lock shows the correct status, send the lock request via HA twice (give is a couple of seconds in between)
5) lock shows dead in Zwave UI

some more logs in case it helps:

<details>
<summary>Logs</summary>

```
2021-05-28 08:19:38.236 INFO ZWAVE: Node 5: has returned from the dead
2021-05-28 08:19:38.264 INFO ZWAVE: Node 5: value added 5-98-0-currentMode => 0
2021-05-28 08:19:38.267 INFO ZWAVE: Node 5: value added 5-98-0-targetMode => 0
2021-05-28 08:19:38.269 INFO ZWAVE: Node 5: value added 5-98-0-outsideHandlesCanOpenDoor => true,false,false,false
2021-05-28 08:19:38.269 INFO ZWAVE: Node 5: value added 5-98-0-insideHandlesCanOpenDoor => false,false,false,false
2021-05-28 08:19:38.270 INFO ZWAVE: Node 5: value added 5-98-0-latchStatus => open
2021-05-28 08:19:38.270 INFO ZWAVE: Node 5: value added 5-98-0-boltStatus => unlocked
2021-05-28 08:19:38.271 INFO ZWAVE: Node 5: value added 5-98-0-doorStatus => open
2021-05-28 08:19:38.271 INFO ZWAVE: Node 5: value added 5-98-0-lockTimeout => 0
2021-05-28 08:19:38.272 INFO ZWAVE: Node 5: value added 5-98-0-operationType => 1
2021-05-28 08:19:38.273 INFO ZWAVE: Node 5: value added 5-98-0-outsideHandlesCanOpenDoorConfiguration => false,false,false,false
2021-05-28 08:19:38.273 INFO ZWAVE: Node 5: value added 5-98-0-insideHandlesCanOpenDoorConfiguration => false,false,false,false
2021-05-28 08:19:38.274 INFO ZWAVE: Node 5: value added 5-98-0-lockTimeoutConfiguration => undefined
2021-05-28 08:19:38.274 INFO ZWAVE: Node 5: value added 5-112-0-1 => 3
2021-05-28 08:19:38.275 INFO ZWAVE: Node 5: value added 5-112-0-2 => 255
2021-05-28 08:19:38.276 INFO ZWAVE: Node 5: value added 5-112-0-3 => 30
2021-05-28 08:19:38.276 INFO ZWAVE: Node 5: value added 5-112-0-4 => 5
2021-05-28 08:19:38.277 INFO ZWAVE: Node 5: value added 5-112-0-7 => 180
2021-05-28 08:19:38.277 INFO ZWAVE: Node 5: value added 5-113-0-alarmType => 144
2021-05-28 08:19:38.278 INFO ZWAVE: Node 5: value added 5-113-0-alarmLevel => 3
2021-05-28 08:19:38.278 INFO ZWAVE: Node 5: value added 5-113-0-Access Control-Lock state => 11
2021-05-28 08:19:38.279 INFO ZWAVE: Node 5: value added 5-114-0-manufacturerId => 297
2021-05-28 08:19:38.281 INFO ZWAVE: Node 5: value added 5-114-0-productType => 26112
2021-05-28 08:19:38.282 INFO ZWAVE: Node 5: value added 5-114-0-productId => 2
2021-05-28 08:19:38.282 INFO ZWAVE: Node 5: value added 5-128-0-level => 100
2021-05-28 08:19:38.283 INFO ZWAVE: Node 5: value added 5-128-0-isLow => false
2021-05-28 08:19:38.283 INFO ZWAVE: Node 5: value added 5-134-0-libraryType => 3
2021-05-28 08:19:38.284 INFO ZWAVE: Node 5: value added 5-134-0-protocolVersion => 4.61
2021-05-28 08:19:38.284 INFO ZWAVE: Node 5: value added 5-134-0-firmwareVersions => 33.16
2021-05-28 08:19:38.284 INFO ZWAVE: Node 5: value added 5-134-0-hardwareVersion => undefined
2021-05-28 08:19:38.285 INFO ZWAVE: Node 5: value added 5-139-0-dateAndTime => undefined
2021-05-28 08:19:38.418 INFO ZWAVE: Node 5 ready: Yale - SD-L1000-CH (Conexis L1)
2021-05-28 08:19:42.176 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 0 => 0
2021-05-28 08:19:42.178 INFO ZWAVE: Node 5: value updated: 98-0-outsideHandlesCanOpenDoor true,false,false,false => true,false,false,false
2021-05-28 08:19:42.180 INFO ZWAVE: Node 5: value updated: 98-0-insideHandlesCanOpenDoor false,false,false,false => false,false,false,false
2021-05-28 08:19:42.181 INFO ZWAVE: Node 5: value updated: 98-0-latchStatus open => open
2021-05-28 08:19:42.183 INFO ZWAVE: Node 5: value updated: 98-0-boltStatus unlocked => unlocked
2021-05-28 08:19:42.184 INFO ZWAVE: Node 5: value updated: 98-0-doorStatus open => open
2021-05-28 08:19:45.934 INFO ZWAVE: Node 5: value updated: 113-0-alarmType 144 => 21
2021-05-28 08:19:45.937 INFO ZWAVE: Node 5: value updated: 113-0-alarmLevel 3 => 1
2021-05-28 08:19:45.941 INFO ZWAVE: Node 5: value updated: 98-0-currentMode 0 => 255
2021-05-28 08:19:45.946 INFO ZWAVE: Node 5 CC Notification {
type: 6,
event: 1,
label: 'Access Control',
eventLabel: 'Manual lock operation',
parameters: undefined
}
2021-05-28 08:20:19.158 INFO ZWAVE: Node 5 is now dead
2021-05-28 08:20:27.253 ERROR ZWAVE-SERVER: Z-Wave error Failed to send the command after 3 attempts (Status NoAck)
ZWaveError: Failed to send the command after 3 attempts (Status NoAck)
at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:2341:23)
at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:2464:28)
at DoorLockCCAPI.set (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:292:21)
at Proxy.DoorLockCCAPI.<computed> (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/DoorLockCC.ts:140:15)
[... 224 lines cut ...]
2021-05-28 08:29:31.846 INFO ZWAVE: Node 5: value added 5-114-0-productType => 26112
2021-05-28 08:29:31.847 INFO ZWAVE: Node 5: value added 5-114-0-productId => 2
2021-05-28 08:29:31.847 INFO ZWAVE: Node 5: value added 5-128-0-level => 100
2021-05-28 08:29:31.847 INFO ZWAVE: Node 5: value added 5-128-0-isLow => false
2021-05-28 08:29:31.848 INFO ZWAVE: Node 5: value added 5-134-0-libraryType => 3
2021-05-28 08:29:31.848 INFO ZWAVE: Node 5: value added 5-134-0-protocolVersion => 4.61
[... cut ...]
2021-05-28 08:29:36.628 INFO APP: GET /static/fonts/MaterialIcons-Regular.fa3334f.woff2 200 3.306 ms - 82492
2021-05-28 08:29:36.718 INFO APP: GET /api/settings 200 87.519 ms - 12689
2021-05-28 08:29:36.740 INFO APP: GET /api/auth-enabled 304 4.231 ms - -
2021-05-28 08:29:36.844 INFO APP: GET /static/favicons/favicon-32x32.png 200 3.940 ms - 1831
2021-05-28 08:30:15.366 INFO APP: GET /api/settings 304 148.328 ms - -
2021-05-28 08:30:15.403 INFO APP: GET /api/auth-enabled 304 3.041 ms - -
```
</details>

AlCalzone, 2021-05-28:
>  How do I get a driver log?

In zwavejs2mqtt, here's how you do it:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

In the HA integration I'm not sure, but the latest version does have a way to do it via the UI.

lolouk44, 2021-05-28:
Thanks, Here are the logs.
Interestingly the lock has come back alive after a few min...

<details>
  <summary>Logs</summary>

```  
2021-05-28 08:41:41.211 INFO MQTT: MQTT is disabled
2021-05-28 08:41:41.221 INFO ZWAVE: Connecting to /dev/ttyACM0
2021-05-28 08:41:41.238 INFO APP: POST /api/settings 200 113.672 ms - 725
2021-05-28 08:41:41.245 INFO ZWAVE: Zwavejs usage statistics ENABLED
2021-05-28 08:41:41.322 INFO APP: GET /api/auth-enabled 304 2.778 ms - -
2021-05-28 08:41:41.518 INFO ZWAVE: Next update scheduled for: Sat May 29 2021 00:00:00 GMT+0000 (Coordinated Universal Time)
2021-05-28 08:41:44.213 INFO ZWAVE: Zwave driver is ready
2021-05-28 08:41:44.214 INFO ZWAVE: Controller status: Driver ready
2021-05-28 08:41:44.217 DEBUG ZWAVE: Binding to node 1 events
2021-05-28 08:41:44.218 DEBUG ZWAVE: Node 1 has been added to nodes array
2021-05-28 08:41:44.220 DEBUG ZWAVE: Binding to node 5 events
2021-05-28 08:41:44.221 DEBUG ZWAVE: Node 5 has been added to nodes array
2021-05-28 08:41:44.224 INFO ZWAVE: Scanning network with homeid: 0xde677210
2021-05-28 08:41:44.228 INFO ZWAVE: Node 1 is alive
2021-05-28 08:41:44.243 INFO ZWAVE: Node 1 ready: AEON Labs - ZW090 (Z‐Stick Gen5 USB Controller)
2021-05-28 08:41:47.893 INFO ZWAVE: Node 5 is dead
2021-05-28 08:41:47.896 INFO ZWAVE: Controller status: Scan completed
2021-05-28 08:41:47.898 INFO ZWAVE: Network scan complete. Found: 2 nodes
2021-05-28 08:42:04.138 INFO ZWAVE: Node 5: has returned from the dead
2021-05-28 08:42:04.320 DEBUG GATEWAY: Publishing discovery: {
  type: 'lock',
  object_id: 'lock',
  discovery_payload: {
    command_topic: 'zwave/nodeID_5/98/0/targetMode/set',
    state_locked: 255,
    state_unlocked: 0,
    payload_lock: 255,
    payload_unlock: 0,
    value_template: '{{ value_json.value }}',
    state_topic: 'zwave/nodeID_5/98/0/currentMode',
    json_attributes_topic: 'zwave/nodeID_5/98/0/currentMode',
    device: {
      identifiers: [ 'zwavejs2mqtt_0xde677210_node5', [length]: 1 ],
      manufacturer: 'ASSA ABLOY',
      model: 'Yale Conexis L1 (SD-L1000-CH)',
      name: 'nodeID_5',
      sw_version: '33.16'
    },
    name: 'nodeID_5_lock',
    unique_id: 'zwavejs2mqtt_0xde677210_5-98-0-currentMode'
  },
  discoveryTopic: 'lock/nodeID_5/lock/config',
  values: [ '98-0-currentMode', '98-0-targetMode', [length]: 2 ],
  persistent: true,
  ignoreDiscovery: false
}
2021-05-28 08:42:04.156 INFO ZWAVE: Node 5: value added 5-98-0-currentMode => 255
2021-05-28 08:42:04.160 INFO ZWAVE: Node 5: value added 5-98-0-targetMode => 255
2021-05-28 08:42:04.161 INFO ZWAVE: Node 5: value added 5-98-0-outsideHandlesCanOpenDoor => false,false,false,false
2021-05-28 08:42:04.162 INFO ZWAVE: Node 5: value added 5-98-0-insideHandlesCanOpenDoor => false,false,false,false
2021-05-28 08:42:04.163 INFO ZWAVE: Node 5: value added 5-98-0-latchStatus => open
2021-05-28 08:42:04.164 INFO ZWAVE: Node 5: value added 5-98-0-boltStatus => locked
2021-05-28 08:42:04.165 INFO ZWAVE: Node 5: value added 5-98-0-doorStatus => closed
2021-05-28 08:42:04.165 INFO ZWAVE: Node 5: value added 5-98-0-lockTimeout => 0
2021-05-28 08:42:04.166 INFO ZWAVE: Node 5: value added 5-98-0-operationType => 1
2021-05-28 08:42:04.167 INFO ZWAVE: Node 5: value added 5-98-0-outsideHandlesCanOpenDoorConfiguration => false,false,false,false
2021-05-28 08:42:04.167 INFO ZWAVE: Node 5: value added 5-98-0-insideHandlesCanOpenDoorConfiguration => false,false,false,false
2021-05-28 08:42:04.168 INFO ZWAVE: Node 5: value added 5-98-0-lockTimeoutConfiguration => undefined
2021-05-28 08:42:04.169 INFO ZWAVE: Node 5: value added 5-112-0-1 => 3
2021-05-28 08:42:04.170 INFO ZWAVE: Node 5: value added 5-112-0-2 => 255
2021-05-28 08:42:04.171 INFO ZWAVE: Node 5: value added 5-112-0-3 => 30
2021-05-28 08:42:04.173 INFO ZWAVE: Node 5: value added 5-112-0-4 => 5
2021-05-28 08:42:04.174 INFO ZWAVE: Node 5: value added 5-112-0-7 => 180
[... 270 lines cut ...]
    device_class: 'battery',
    state_topic: 'zwave/nodeID_5/128/0/isLow',
    json_attributes_topic: 'zwave/nodeID_5/128/0/isLow',
    device: {
      identifiers: [ 'zwavejs2mqtt_0xde677210_node5', [length]: 1 ],
      manufacturer: 'ASSA ABLOY',
      model: 'Yale Conexis L1 (SD-L1000-CH)',
      name: 'nodeID_5',
      sw_version: '33.16'
    },
    name: 'nodeID_5_battery_islow',
    unique_id: 'zwavejs2mqtt_0xde677210_5-128-0-isLow'
  },
  discoveryTopic: 'binary_sensor/nodeID_5/battery_islow/config',
  values: [ '128-0-isLow', [length]: 1 ],
  persistent: true,
  ignoreDiscovery: false
}
2021-05-28 08:42:48.054 INFO APP: GET /api/settings 200 114.827 ms - 12722
2021-05-28 08:42:48.127 INFO APP: GET /api/auth-enabled 304 3.090 ms - -
2021-05-28 08:42:53.373 INFO GATEWAY: Closing Gateway...
2021-05-28 08:42:53.413 INFO ZWAVE: Client closed
```
</details>
I only have this lock on the network...

AlCalzone, 2021-05-28:
You might have been put on the wrong track by a mistake in the docs. That is still the z2m log, not the driver log.

https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs has now been corrected and should point you towards the correct one.

lolouk44, 2021-05-28:
Thanks. I can't reproduce now 😠 
Can you confirm the Dead/Alive status is not available in HA? It would help to get some timings and potentially more logs.
Would you have any additional troubleshooting advice as I'm starting to think it's not an issue with Zwave-JS and don't want to keep wasting your time if my hunch is true.

AlCalzone, 2021-05-28:
> Can you confirm the Dead/Alive status is not available in HA?

I don't know - I don't use HA.

> don't want to keep wasting your time if my hunch is true.

I should be able to confirm if I get my hands on a driver log ;)

lolouk44, 2021-06-02:
container went down, when trying to restart, it fails to detect the aeotec zwave stick so most likely this is where the failure comes.
I'll close this for now and will reopen with logs if I experience this issue again
Thanks for your help

closed 2021-06-02 by lolouk44