zwave-js/zwave-js #2701: New Event: 'heal failed' for ZWaveNode.

marcus-j-davies, 2021-05-26:
As discussed:

We have the `interview failed` event, that gives some indication of the reasons it failed.
We however, do not have the same event for a failed heal.

Suggestion to add a new `heal failed` event on the ZWaveNode object, that offers the same information
(albeit more related error info).

Whilst we have `heal network progress` which is super helpful, one has to review logs to find out more on the failed collection.

AlCalzone, 2021-05-26:
The failure reason should ideally be part of the progress report.

marcus-j-davies, 2021-05-26:
That would be helpful, but may cause a break, on the current signature?

`(progress: ReadonlyMap<number, HealNodeStatus>)`

AlCalzone, 2021-05-26:
True, but anything else would make for an awkward API