zwave-js/zwave-js #2702: Zipato "Mini Keypad" RFID

ArMour85, 2021-05-26:
**Which device is missing?**

Zipato "Mini Keypad" RFID

**What are the IDs?**

Manufacturer ID: 0x0097  
Product type: 0x6131  
Product ID: 0x4501  

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

https://tutoriels.domotique-store.fr/images/Zipato/wt-rfid.eu%20-%20Zipato%20Mini%20Keypad%20RFiD%20Z-Wave.pdf

AlCalzone, 2021-05-26:
Interesting - we have this with a matching device ID under Wintop:
https://devices.zwave-js.io/?jumpTo=0x0097:0x6131:0x4501:0.0

Gis70, 2022-09-03:
Hello, I have this devices and one thing is missing, it's the following : 
[image]

0x71 parameters should return 5 away or 6 present on the value 9th places in the raw value returned.

How could i obtain it ?

Gis70, 2022-09-15:
I put zwavejs in silly debug mode and this is all that i can get when i'm making action with the Home or Away button.
It seems something is missing ? if not i don"t know how to get the Notification CC on this node, or even with the API.
If you can help me... Thank you

`2022-09-15T18:56:58.177Z SERIAL « 0x01140004000e0e63030000323331320000000000008d                      (22 bytes)
2022-09-15T18:56:58.179Z SERIAL » [ACK]                                                                   (0x06)
2022-09-15T18:56:58.181Z CNTRLR   [Node 014] [translateValueEvent: value updated]
                                    commandClass: User Code
                                    endpoint:     0
                                    property:     userIdStatus
                                    propertyKey:  0
                                    internal:     false
                                    secret:       false
                                    event source: undefined
2022-09-15T18:56:58.182Z CNTRLR   [Node 014] [~] [User Code] userIdStatus[0]: 0 => 0                [Endpoint 0]
2022-09-15T18:56:58.182Z CNTRLR   [Node 014] [translateValueEvent: value updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2022-09-15T18:56:58.184Z CNTRLR   [Node 014] [translateValueEvent: value updated]
                                    commandClass: User Code
                                    endpoint:     0
                                    property:     userCode
                                    propertyKey:  0
                                    internal:     false
                                    secret:       true
                                    event source: undefined
2022-09-15T18:56:58.184Z CNTRLR   [Node 014] [translateValueEvent: value updated]
                                    is root endpoint:        true
                                    is application CC:       true
                                    should hide root values: false
2022-09-15T18:56:58.186Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[UserCodeCCReport]
                                      user id:   0
                                      id status: Available
                                      user code: ****
2022-09-15T18:56:58.194Z SERIAL « 0x01080004000e0284077c                                              (10 bytes)
2022-09-15T18:56:58.195Z SERIAL » [ACK]                                                                   (0x06)
2022-09-15T18:56:58.196Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[WakeUpCCWakeUpNotification]
2022-09-15T18:56:58.200Z CNTRLR « [Node 014] received wakeup notification
2022-09-15T18:56:58.203Z CNTRLR   [Node 014] The node is now awake.
2022-09-15T18:56:59.205Z CNTRLR » [Node 014] Sending node back to sleep...
2022-09-15T18:56:59.216Z SERIAL » 0x010900130e028408240041                                            (11 bytes)
2022-09-15T18:56:59.216Z DRIVER » [Node 014] [REQ] [SendData]
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[WakeUpCCNoMoreInformation]
2022-09-15T18:56:59.218Z SERIAL « [ACK]                                                                   (0x06)
2022-09-15T18:56:59.224Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-09-15T18:56:59.225Z SERIAL » [ACK]                                                                   (0x06)
2022-09-15T18:56:59.226Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-09-15T18:56:59.234Z CNTRLR   [Node 014] The node is now asleep.`

AlCalzone, 2022-09-15:
Looks like you haven't configured the device to recognize that tag yet. This is from the device manual:
[image]

You need to take the user code that is reported in slot 0 and set it in the slot it should be in.