zwave-js/zwave-js #2709: Bug - Yale keyfree connected lock missing

DirkFace, 2021-05-26:
**Checklist:**

-   [✓ ] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [✓ ] 
[node9.txt](https://github.com/zwave-js/node-zwave-js/files/6547801/node9.txt)
[zwave_js_dump.txt](https://github.com/zwave-js/node-zwave-js/files/6547811/zwave_js_dump.txt)

My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [✓ ] I have checked the troubleshooting section and my problem is **not** described there.
-   [✓] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Hi, I've just migrated from smartthings over to HA & managed to move over all my devices that were on zigee or zwave - apart from my front door lock which is a ... https://help.yale.co.uk/wp-content/uploads/2019/06/Keyfree-manual-2014-2.pdf

The zwave module is found when adding the device & is alive / beaming, but returns a Device ID 297-0-1536 (0x0129-0x0600-0x0000)

As I mention, it worked fine thorugh zwave / smartthings & I was using the Rboy device handler.

**Device information**

Which device(s) is/are affected (manufacturer/model)?  
What is/are the node IDs?

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [✓] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [✓] **Yes**, in: _(it worked in smartthings using the Rboy device handler & generic smarthings zwave lock handler

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [✓] Other:  Zstick gen5 USB

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-05-26:
You did not read the template thoroughly. I **really** need the logfile that's being requested in the issue template and not others:

    ATTACH(!) `zwave-js` logfile with DEBUG or VERBOSE loglevel here. Please no links or gists or embedded logs.
    Please make sure to upload the correct log. If you're unsure, the correct one is called `zwave-<number>.log` and starts with

    ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
    ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
      ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
     ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
    ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
    ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝

    If you are using zwavejs2mqtt, this is how you create the logfiles:
    * Go to Settings, Z-Wave section
    * select log level DEBUG
    * enable "log to file"

    For HomeAssistant, this is how you do it:
    Home Assistant -> settings -> Integrations -> Z-Wave JS -> Configure -> Create dump -> zip the json file and post it here.

DirkFace, 2021-05-26:
Sorry. I'm new to this.

I've set the log level to debug & log to file ... it states Default: store/zwavejs2newMqtt.log
If I go into Home Assistant -> settings -> Integrations -> Z-Wave JS -> Configure -> **there is no option to create dump**
But there is 'download a dump of your network to help diagnose issues' which was what I initially attached.

DirkFace, 2021-05-26:
I've tracked down the Default store location & managed to export this file.
Apologies again for my lack of knowledge, but hopefully this is what you may require? (attached)
[zwavejs_2021-05-26.zip](https://github.com/zwave-js/node-zwave-js/files/6549564/zwavejs_2021-05-26.zip)

AlCalzone, 2021-05-26:
Ok we do in fact not have a configuration file for this. It should be working though, except for configuration parameters and the device labels. I do see the usual values for a Lock in your node dump you posted above.

AlCalzone, 2021-05-26:
I've raised https://github.com/zwave-js/node-zwave-js/issues/2712 to track the creation of the missing device file.

closed 2021-05-26 by AlCalzone

DirkFace, 2021-05-27:
Thank you for looking into this for me, appreciate it.