zwave-js/zwave-js #2710: Define in config files what a Basic CC Report maps to

AlCalzone, 2021-05-26:
For some device types the specification explicitly states that the Basic CC mapping target must be mentioned in the user manual. For those, zwave-js cannot know which CC to map the Basic CC to, unless we give it the info. For this, we should add an endpoint-specific configuration option. One example is the Logic Group ZDB5100:
[image]

J2Kbr, 2021-05-26:
Is this related with #2676 and #2659 ?

If yes, here is the manual page of Coolcam Light Switch model EU-2.

Device : 600-4235-3 (0x0258-0x0003-0x108b)

[image]

AlCalzone, 2021-05-26:
No - the behavior for Switches is pretty well defined.

Roagert, 2021-05-27:
Seems there might be some discovery scheme where it cannot recognize this device as well 
Logic Group ZBA7140
[image]
[image]

HA just recognizes the battery and not the 4 switches.,

AlCalzone, 2021-05-27:
Any chance you could share a driver log of a re-interview? Just so we have all the information in one place.

Roagert, 2021-05-27:
Attached

[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6552658/zwavejs2mqtt.log)
[zwavejs_2021-05-27.log](https://github.com/zwave-js/node-zwave-js/files/6552659/zwavejs_2021-05-27.log)