zwave-js/zwave-js #2712: Missing configuration for Yale keyfree connected lock

AlCalzone, 2021-05-26:
**Which device is missing?**

Enter Manufacturer and Device name here

**What are the IDs?**

Manufacturer ID: 0x0129  
Product type: 0x0600  
Product ID: 0x0000  

**Did you check if we really don't have it?**
- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**

https://help.yale.co.uk/wp-content/uploads/2019/06/Keyfree-manual-2014-2.pdf

_Originally reported in https://github.com/zwave-js/node-zwave-js/issues/2709_
/cc @DirkFace

disforw, 2021-06-12:
Is this still needed?

DirkFace, 2021-06-12:
Yes please.

At present I have to physically set it to locked in HA, as it never
registers this.
Also no configuration settings are available, i.e. passagemode, time before
re-locking automatically, etc.

I have a device handler which works with this lock via Smartthings. If you
need it for reference, I could email it to you?

The Yale locks it covers are : (mine is the third one down)

        case "0129-8008-0A00": // YALE NEXTOUCH AUR-NTM-62
        	log.debug "Found YALE NEXTOUCH AUR-NTM-62"
        	break
        case "0129-803A-0508": // Yale YRD156
        	log.debug "Found Yale YRD156"
            break
        *case "0129-0600-0000": // Yale keyfree connected
            log.debug "Found Yale Keyfree Connected Lock"*
            break
        case ~/0129-0001-.*/: // Yale Touch Lever
            log.debug "Found Yale Touch Lever"
            break
        case ~/0129-0002-.*/: // Yale Touch Deadbolt
            log.debug "Found Yale Touch Deadbolt"
            break
        case ~/0129-0003-.*/: // Yale Push Button Lever
            log.debug "Found Yale Push Button Lever"
            break
        case ~/0129-0004-.*/: // Yale Push Button Deadbolt
            log.debug "Found Yale Push Button Deadbolt"
            break
        case ~/0129-0006-.*/: // Yale Keyfree/Conexis L1 Lock
            log.debug "Found Yale Keyfree/Conexis L1 Lock"
            break
        case ~/0129-6600-.*/: // Yale Conexis
        case ~/0129-0066-.*/: // Yale Yale Conexis L1 SD-L1000-CH Safe.co.uk
            log.debug "Found Yale Conexis Lock"
            break
        case ~/0129-0007-.*/: // Yale Keyless Connected YD-01
            log.debug "Found Yale Keyless Connected YD-01"
            break
        case ~/0129-8001-.*/: // Yale nextTouch Wireless Touchscreen
            log.debug "Found Yale nexTouch"
            break
        case ~/0129-8002-.*/: // Yale Assure Lock
        case ~/0129-800B-.*/:
        case ~/0129-800C-.*/:
            log.debug "Found Yale Assure"
            break
        case ~/0129-8004-.*/: // Yale Yale Assure Lock Push Button Deadbolt
            log.debug "Found Yale Assure Lock Push Button Deadbolt"
            break
        case ~/0129-0600-.*/: // Yale Assure Assa Abloy
            log.debug "Found Yale Assure Assa Abloy Lock"
            break

Kind regards,

Dirk Lee.

disforw, 2021-06-12:
PR created, #2842

closed 2021-12-06 by blhoward2