zwave-js/zwave-js #2714: [feat] Proper Multi Channel support

andras-tim, 2021-05-27:
**Is your feature request related to a problem? Please describe.**

The current group assignment has several issues, but I think these are in strong connection, therefore I write all of them into this ticket.

* [ ] The **single-channel** mode specifies only the **Target Node**, however, the **multi-channel** requires the **Target Endpoint** too. If I have a device that supports both of them (e.g. Qubino Flush 2 Relay) then the **Target Endpoint** will specify the communication mode of an assign. So...
  - singngle-channel mode: 
    **Root device (Endpoint 0)** / **Lifeline (Group 0)**
    => **Node 1** / `None` **Target Endpoint**
  - multi-channel mode:
    **Root device (Endpoint 0)** / **Lifeline (Group 0)** 
    => **Node 1** / `0` **Target Endpoint**

  that is so important because...

  - the modes have different value/switch/... identifiers
  - some values/switches/... are not reachable in single-channel mode
  - the multi-channel mode 1.5 - 2 times slower because it has an extra multi-channel packet-encapsulation

  _(Based on my 3 days long tests and this detailed [Sylab: Z-Wave Association Basics](https://www.silabs.com/documents/public/application-notes/APL12956-Z-Wave-Association-Basics.pdf)_ documentation.)
  
  ... So I think the channel mode should be highlight somehow instead of a simple `0` or `None`... because this is not so intuitive.

* [ ] The current .json device descriptors contain only one set of association groups. However, a multi-channel device can have different groups per endpoint, e.g. check the **Association Groups** in the manual of the [Flush 2 Relays](https://qubino.com/manuals/Flush_2_Relay.pdf) on **page 44**. This has a **Root Device** + 3 extra **Endpoints** with totally different(!) groups.

* [x] I would like to display the **Group Index** beside the **Group Names** to be able to easily identify the desired group based on user manuals.

* [ ] Some devices haven't multiple endpoints, however, the manual says it has several. E.g. the _Qubino's Flush Dimmer Plus_ has a root device (Endpoint 0)  + 4 extra endpoints... but the UI shows only one.

**Describe the solution you'd like**
- Please, highlight the single-multi channel mode
- The device descriptor files have to contain different association groups for Entrypoints, according to the user manuals.
- Show all endpoints and group indexes too.

Thank you!

andras-tim, 2021-05-30:
> @andras-tim What version are you using?

I'm using use zwavejs2mqtt: 4.2.1 with zwave-js: 7.5.0

andras-tim, 2021-05-31:
> > The single-channel mode specifies only the Target Node, however..
> 
> I don't understand this, what's the issue? ATM you can specify the target endpoint you prefer based on the suggested groups on UI dialog.

Yes, I can specify the endpoint and the channel too, but I didn't know what is the difference between the **Endpoint None** and the **Endpoint 0** until I dig deeper.

I agreed with @AlCalzone this is a documentation bug, but I think this is a bit more:

* If you use a multi-mode device the single-channel and the multi-channel mode update different values via the lifeline.
* Therefore a status change will update the different value on the "Node" tab
* Therefore after a single-channel <> multi-channel switch you will have dead values on UI

Ideas for fix:
* I think a small note in braces and/or an icon beside the `None` target and the `0` target (of the lifeline) can make the behavior be more understandable.
* And I think, the UI has to have a "limited" **Re-interview Node** to be able to drop dead values, but keep the customized associations (e.g. keep the lifeline's mode switch).

_Side note: And I think, we should standardize the default assign mode of both-mode supported devices because sometimes the lifeline has been assigned to `None` (so update via single-channel) and sometimes to `0` (so update via multi-channel)._

> > I would like to display the Group Index beside the Group Names to be able to easily identify the desired group based on user manuals.
> 
> This is an easy fix, no problem

thank you

andras-tim, 2021-05-31:
@AlCalzone 
> > the multi-channel mode 1.5 - 2 times slower because it has an extra multi-channel packet-encapsulation
> 
> Multi-Channel encapsulation basically just means you have a few bytes extra - so I doubt that it is much slower.

I have read that in docs, but all of my **Qubino R2** and **Dimmer Plus** modules send the state updates approx. in
* **3-4 s** w/ multi-channel associated lifeline
* **1-2 s** w/ multi-channel associated lifeline

(without any other associations on the nodes and in a silent environment)

_Maybe is this Qubino's special behavior?_

> > The current .json device descriptors contain only one set of association groups. However, a multi-channel device can have different groups per endpoint
> 
> Supported since 3 days ago: #2621
> But we need help to identify and edit device files that need this.

Wow! Many thanks! I will help on the Qubino's front.

> 
> > Qubino's Flush Dimmer Plus has a root device (Endpoint 0) + 4 extra endpoints... but the UI shows only one.
> 
> From my experience, the endpoint count of many Qubino devices depends on how they were configured and what was connected at the time of inclusion.

Yes, that is my experience too... but a **Dimmer Plus** has these endpoints:

1. The "Root" endpoint (always existing)
2. Endpoint 1 for **I1** (always existing in the "multi-channel world")
3. Endpoint 2 for **I2** (depends on the value of the _[13-112-0-100] Enable / Disable Endpoints I2_)
4. Endpoint 3 for **I3** (depends on the value of the _[13-112-0-101] Enable / Disable Endpoints I3_)
5. Endpoint 4 for optional external temperature probe (depends on the connectivity of the sensor)

So, at least the "root" and the "Endpoint 1" should be here.

AlCalzone, 2021-07-11:
#2997 is going to fix this. We may need to add an additional compat flag afterwards though.

commit 10e53aff (2021-07-13): feat: rework how endpoints and lifeline associations are handled (#2997)

fixes: #2286
fixes: #2257
fixes: #1989
fixes: #2714
fixes: #1860
fixes: #2910
closed 2021-07-17 by AlCalzone