zwave-js/zwave-js #2719: [bug] Qubino 3-Phase Smart Meter values disappearing after some time

zfubar86, 2021-05-27:
**Checklist:**

- [X] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt version: 4.4.0
zwave-js version: 7.5.2

### Describe the bug

I securely added a Qubino 3-Phase Smart Meter. 
Values were being reported correctly for each phase for a few hours, after that one phase started to report erratically and the others stopped completely.
I re-interviewed the node and most of the values disappeared.

### To Reproduce

Steps to reproduce the behavior:

1. Securely add Qubino 3-Phase Smart Meter.
2. Node added, meter values showing and updating, "Secure: Yes".
2. Wait 3 hours.
3. Re-interview node.
4. Values disappear and node is marked "Secure: No".

### Expected behavior

Node should still be marked as secure, values shouldn't disappear.

Logs for the re-interview are attached.

[node_25.json.txt](https://github.com/zwave-js/zwavejs2mqtt/files/6546481/node_25.json.txt)
[zwavejs.log](https://github.com/zwave-js/zwavejs2mqtt/files/6547721/zwavejs.log)

AlCalzone, 2021-05-27:
```
2021-05-26T15:03:42.877Z CNTRLR   [Node 025] Timed out while waiting for a response from the node
2021-05-26T15:03:42.879Z CNTRLR   [Node 025] The node is not included securely. Continuing interview non-securely.
```

The node did not respond to the query that's used to find out which commands it supports securely. When that happens we can only assume that it isn't included securely. Because of this, the values are not queried again and disappear.

Interview the node again and hope that it responds this time.

closed 2021-05-27 by AlCalzone

zfubar86, 2021-05-27:
> ```
> 2021-05-26T15:03:42.877Z CNTRLR   [Node 025] Timed out while waiting for a response from the node
> 2021-05-26T15:03:42.879Z CNTRLR   [Node 025] The node is not included securely. Continuing [...]

thanks for the analysis!

This node is very close to the controller so I guess that rules out bad comms. I already tried re-nterviewing several times and always get the same results. Can it be a firmware issue? I have two other qubino devices with strange results.

AlCalzone, 2021-05-28:
> This node is very close to the controller so I guess that rules out bad comms. 

It doesn't really. Your controller itself could be in a bad spot or be affected by reflections or interference. Try moving it a few cm or putting it on a USB extension cable.

zfubar86, 2021-05-28:
I bit the bullet and shutdown the main breaker to force a device restart and it solved it. I will report this to qubino!

zfubar86, 2021-05-28:
> I bit the bullet and shutdown the main breaker to force a device restart and it solved it. I will report this to qubino!

Solved it only for a few mins.