zwave-js/zwave-js #2721: [question] Unknown manufacturer 0xXXXX

Ruudje007, 2021-05-28:
Hi, 
this is I think the third time now that I receive the following error for all my devices, this happens every time I update the docker image
`Unknown manufacturer 0xXXXX `

[image]

The first two times I could only fix this by resetting my zwave stick and start al over again. But I would like to avoid this since I takes a lot of time. 

docker-compose.yml 

```
version: '3.7'
services:

  zwavejs2mqtt:
        container_name: zwavejs2mqtt
        image: zwavejs/zwavejs2mqtt:latest
        restart: always
        tty: true
        stop_signal: SIGINT
        environment:
            - SESSION_SECRET=h2D3JZeoMrsV6WCLHiLv
            - TZ=Europe/Amsterdam
        devices:
            - "/dev/ttyACM0:/dev/ttyACM0"
        volumes:
            - zwavejs:/usr/src/app/store
        ports:
            - '8091:8091' # port for web interface
            - '3000:3000' # port for zwave-js websocket server
        networks:
            Arctic:
                ipv4_address: 10.20.30.8

                ipv4_address: 10.20.30.9

networks:
    Arctic:
        ipam:
            driver: default
            config:
                - subnet: 10.20.30.0/16

volumes:
    zwavejs:

```

Log file

```

 ______                       _     ___                  _   _   
|___  /                      (_)   |__ \                | | | |  
   / /_      ____ ___   _____ _ ___   ) |_ __ ___   __ _| |_| |_ 
  / /\ \ /\ / / _` \ \ / / _ \ / __| / /| '_ ` _ \ / _` | __| __|
 / /__\ V  V / (_| |\ V /  __/ \__ \/ /_| | | | | | (_| | |_| |_ 
/_____|\_/\_/ \__,_| \_/ \___| |___/____|_| |_| |_|\__, |\__|\__|
                            _/ |                      | |        
                           |__/                       |_|        

2021-05-28 00:00:08.979 WARN STORE: scenes.json not found
/bin/sh: git: not found
2021-05-28 00:00:11.451 INFO APP: Version: 4.5.0
2021-05-28 00:00:11.453 INFO APP: Application path:/usr/src/app
2021-05-28 00:00:11.582 INFO MQTT: Connecting to mqtt://10.0.4.100:1883
2021-05-28 00:00:11.710 INFO ZWAVE: Connecting to /dev/ttyACM0
2021-05-28 00:00:11.760 INFO APP: Listening on port 8091 host 0.0.0.0 protocol HTTP
2021-05-28 00:00:11.849 INFO MQTT: MQTT client connected
2021-05-28 00:00:16.261 INFO ZWAVE: Next update scheduled for: Sat May 29 2021 00:00:00 GMT+0200 (Central European Summer Time)
[... 327 lines cut ...]
2021-05-28 00:08:58.086 INFO ZWAVE: Node 18: value updated: 49-0-Illuminance 0 => 0
2021-05-28 00:10:25.672 INFO APP: GET /control-panel 200 8.145 ms - 1371
2021-05-28 00:10:25.748 INFO APP: GET /static/css/app.13acfb9370ca06cb5352.css 304 2.599 ms - -
2021-05-28 00:10:25.754 INFO APP: GET /static/js/app.54e6ea30c5784a7fae32.js 304 2.103 ms - -
2021-05-28 00:10:25.914 INFO APP: GET /api/auth-enabled 304 6.194 ms - -
2021-05-28 00:10:25.946 INFO APP: GET /static/logo.png 304 1.141 ms - -
2021-05-28 00:10:26.042 INFO APP: GET /static/fonts/MaterialIcons-Regular.fa3334f.woff2 304 2.718 ms - -
2021-05-28 00:10:26.118 INFO APP: GET /api/settings 200 102.038 ms - 6190
2021-05-28 00:10:26.133 INFO APP: GET /api/auth-enabled 304 2.631 ms - -
2021-05-28 00:11:04.831 INFO ZWAVE: Node 20: value added 20-50-1-value-65537 => 1.11
2021-05-28 00:11:04.836 INFO ZWAVE: Node 20: value updated: 50-1-value-65537 1.1 => 1.11
2021-05-28 00:12:08.743 INFO MQTT: Message received on zwave/8/37/0/targetValue/set, '{"value":false}'
2021-05-28 00:12:08.747 INFO ZWAVE: Writing false to 8-37-0-targetValue
2021-05-28 00:12:08.897 INFO ZWAVE: Node 8: value updated: 37-0-currentValue false => false
2021-05-28 00:12:10.027 INFO ZWAVE: Node 8: value updated: 37-0-currentValue false => false
2021-05-28 00:12:40.652 INFO ZWAVE: Node 5: metadata updated: 49-0-Power
2021-05-28 00:12:40.657 INFO ZWAVE: Node 5: value updated: 49-0-Power 0 => 0
2021-05-28 00:12:56.256 INFO ZWAVE: Node 21: metadata updated: 49-0-Illuminance
2021-05-28 00:12:56.259 INFO ZWAVE: Node 21: value updated: 49-0-Illuminance 4 => 0
2021-05-28 00:13:07.496 INFO ZWAVE: Node 14: metadata updated: 49-0-Illuminance
2021-05-28 00:13:07.499 INFO ZWAVE: Node 14: value updated: 49-0-Illuminance 0 => 0

      
    
```

AlCalzone, 2021-05-28:
> zwavejs_2021-05-28.log

That's the one - but I need it with loglevel "debug"

Ruudje007, 2021-05-28:
@AlCalzone here you go
[zwavejs_2021-05-28 (debug).log](https://github.com/zwave-js/node-zwave-js/files/6562659/zwavejs_2021-05-28.debug.log)

AlCalzone, 2021-05-28:
Ok this actually looks like the `.values.jsonl` and the `.metadata.jsonl` files were both truncated 😐

cdnninja, 2021-05-28:
I have had this same thing happen more than once.  Typically restarting has fixed it for me.  Not sure root cause though.

Ruudje007, 2021-05-28:
@cdnninja i have a simple backup script that shutdowns all my containers, backups them and transfer them to my Synology NAS and restart the containers again. This happens every night. So a restart will not work for me.

AlCalzone, 2021-05-28:
@Ruudje007 can you somehow figure out if the mentioned files broke during a shutdown? I can't wrap my head around why these mostly appendonly files get truncated and why the hell it only happens when Docker is involved.

Ruudje007, 2021-05-28:
What I can do is to compare the files from no vs my backups. What I know is is that those I see these errors almost every time I updated my docker container. 

So my feeling is that during an update something gets corrupted?

AlCalzone, 2021-05-28:
@cdnninja if restarting fixed it, your files weren't broken. You probably ran into the issue where querying the neighbors during startup would lock up the stick.

@Ruudje007 anything that could put us closer to a solution would help. I have a suspicion that somehow multiple processes try to access the db files at the same time.

Ruudje007, 2021-05-28:
@AlCalzone you are totally right that the file is truncated 

So what is changed during that period

- at 2021-05-18 I had version 4.2.1 running 
- at 2021-05-19 I updated to 4.3.0, files are still fine
- at 2021-05-24 I updated to 4.3.1, files are still fine
- at 2021-05-26 files are still fine
- at 2021-05-27 I updated to 4.4.0, I notice the first differences in files (Corrupted)

[image]

Files from 2021-05-26 (Working files):
[_data.zip](https://github.com/zwave-js/node-zwave-js/files/6563594/_data.zip)

Files from 2021-05-27 (Corrupted files):
[_data.zip](https://github.com/zwave-js/node-zwave-js/files/6563597/_data.zip)

AlCalzone, 2021-05-28:
>     * at 2021-05-27 I updated to 4.4.0, I notice the first differences in files (Corrupted)

Was that **after** you started the updated container?

Ruudje007, 2021-05-28:
yes indeed

FYI i restored the corrupted files and it looks like i can access my devices again

AlCalzone, 2021-05-28:
Okay, 4.4.0 had some problems. There might have been multiple driver instances at once trying to access the DB files and blocking each other. Hopefully #2728 fixes that.

Ruudje007, 2021-05-28:
Perfect! 🤩  thank you very much for your help, just bought you a couple of beers 👍

AlCalzone, 2021-05-28:
Cheers - but we still need to verify that the potential fix works.

Ruudje007, 2021-05-28:
Nevertheless you spend a lot of time in this software, so a little appreciation is all i can do. 

Once the version is available i will test and see if everything is working and keep you updated.

AlCalzone, 2021-06-06:
Did you have a chance to test yet?

closed 2021-06-19 by zwave-js-assistant[bot]