zwave-js/zwave-js #2722: Adding support for GreenWave Reality GWR868RF Z-Wave 868

karolzamosc, 2021-05-28:
Hi,

Is it possible to add support for a GreenWave Z-Wave usb stick?

ID_SERIAL: GreenWave_Reality_Inc._GWR868RF_GP03-0916-00076
ID_VENDOR: GreenWave_Reality_Inc.
ID_VENDOR_ENC: GreenWave\x20Reality\x20Inc.\x20
ID_VENDOR_ID: '2703'

[image]
[image]

Regards, 
Charles

marcus-j-davies, 2021-05-28:
Hi,

No amount of configuration files will overcome the stick not responding.
The stick is not responding to a basic call of Getting its version.

Configuration comes after the fact. Config files for a stick however, only improves cosmetics - it's not required.

I can't find any info on this stick, is it a series 700/500?
have you tried a different port / environment ?

karolzamosc, 2021-05-28:
@marcus-j-davies first of all, thanks for your quick answer. I didn't check different ports and environment (I only have a Dell WYSE + Debian + Docker + HA Supervised). Now I can only remote connect to the machine with installed stick and I got this:

  - name: 5-2
    sysfs: '/sys/devices/pci0000:00/0000:00:15.2/0000:05:00.0/usb5/5-2'
    dev_path: /dev/bus/usb/005/002
    subsystem: usb
    by_id: null
    attributes:
      BUSNUM: '005'
      DEVNAME: /dev/bus/usb/005/002
      DEVNUM: '002'
      DEVPATH: '/devices/pci0000:00/0000:00:15.2/0000:05:00.0/usb5/5-2'
      DEVTYPE: usb_device
      DRIVER: usb
      ID_BUS: usb
      ID_MODEL: GWR868RF
      ID_MODEL_ENC: GWR868RF\x20
      ID_MODEL_ID: '0002'
      ID_REVISION: 0009
      ID_SERIAL: GreenWave_Reality_Inc._GWR868RF_GP03-0916-00076
      ID_SERIAL_SHORT: GP03-0916-00076
      ID_USB_INTERFACES: ':020201:0a0000:'
      ID_VENDOR: GreenWave_Reality_Inc.
      ID_VENDOR_ENC: GreenWave\x20Reality\x20Inc.\x20
      ID_VENDOR_ID: '2703'
      MAJOR: '189'
      MINOR: '513'
      PRODUCT: 2703/2/9
      SUBSYSTEM: usb
      TYPE: 2/0/0
      USEC_INITIALIZED: '4275725'
  - name: ttyACM0
    sysfs: >-
      /sys/devices/pci0000:00/0000:00:15.2/0000:05:00.0/usb5/5-2/5-2:1.0/tty/ttyACM0
    dev_path: /dev/ttyACM0
    subsystem: tty
    by_id: /dev/serial/by-id/usb-GreenWave_Reality_Inc._GWR868RF_GP03-0916-00076-if00
    attributes:
      DEVLINKS: >-
        /dev/serial/by-id/usb-GreenWave_Reality_Inc._GWR868RF_GP03-0916-00076-if00
        /dev/serial/by-path/pci-0000:05:00.0-usb-0:2:1.0
      DEVNAME: /dev/ttyACM0
      DEVPATH: >-
        /devices/pci0000:00/0000:00:15.2/0000:05:00.0/usb5/5-2/5-2:1.0/tty/ttyACM0
      ID_BUS: usb
      ID_MM_CANDIDATE: '1'
      ID_MODEL: GWR868RF
      ID_MODEL_ENC: GWR868RF\x20
      ID_MODEL_FROM_DATABASE: uPD720200 USB 3.0 Host Controller
      ID_MODEL_ID: '0002'
      ID_PATH: 'pci-0000:05:00.0-usb-0:2:1.0'
      ID_PATH_TAG: pci-0000_05_00_0-usb-0_2_1_0
      ID_PCI_CLASS_FROM_DATABASE: Serial bus controller
      ID_PCI_INTERFACE_FROM_DATABASE: XHCI
      ID_PCI_SUBCLASS_FROM_DATABASE: USB controller
      ID_REVISION: 0009
      ID_SERIAL: GreenWave_Reality_Inc._GWR868RF_GP03-0916-00076
      ID_SERIAL_SHORT: GP03-0916-00076
      ID_TYPE: generic
      ID_USB_CLASS_FROM_DATABASE: Communications
      ID_USB_DRIVER: cdc_acm
      ID_USB_INTERFACES: ':020201:0a0000:'
      ID_USB_INTERFACE_NUM: '00'
      ID_VENDOR: GreenWave_Reality_Inc.
      ID_VENDOR_ENC: GreenWave\x20Reality\x20Inc.\x20
      ID_VENDOR_FROM_DATABASE: NEC Corporation
      ID_VENDOR_ID: '2703'
      MAJOR: '166'
      MINOR: '0'
      SUBSYSTEM: tty
      TAGS: ':systemd:'
      USEC_INITIALIZED: '7267858'

[image]
[image]
[image]
[image]
[image]

AlCalzone, 2021-05-28:
What makes you think this is a Z-Wave stick? I don't see "Z-Wave" mentioned anywhere, except in the issue title.

closed 2021-05-28 by AlCalzone

karolzamosc, 2021-05-28:
@AlCalzone the seller provides information that this stick is a z-wave bridge.

AlCalzone, 2021-05-28:
If that is indeed true, this seems like you've set up a wrong USB port. Z-Wave Sticks don't need config files or anything to function.