zwave-js/zwave-js #2723: [bug] Failed to execute controller commands after x/3 attempts.

cinadr, 2021-05-28:
### Version

**Checklist:**

- [x] I am **not** using HomeAssistant.
- [x] I have checked the troubleshooting section and my problem is **not** described there.
- [x] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [ ] Docker
- [ ] PKG
- [ ] Snap package
- [x] Manually built (git clone - yarn install - yarn run build ) Windows x64 Server

zwavejs2mqtt version: 4.4.0.5238242
zwave-js version: 7.7.0

### Describe the bug
After a while the zwavejs.log shows 
```
CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try in 1100 ms.
```
and nothing works or responds. After restarting zwavejs2mqtt all comes back to live.
In Zwavejs2mqtt.log:
```
ERROR ZWAVE: Node x error while updating Neighbors: Failed to send the message after 3 attempts
```
or 
```
ERROR ZWAVE: Error while writing true on 5-37-1-targetValue: Failed to send the message after 3 attempts
```
But the undelying zwavejs error always as above.

### To Reproduce

Steps to reproduce the behavior:
Start with yarn start and wait for several hours before this appears. Tested for 3-4 days and it is consequently dies every time. Approx since 4.2 version. (Not sure on what version this occured first).

### Expected behavior

Stabile controller operation.

### Additional context

Controller is AEOTEC ZWAVE USB STICK.

### Attached logs
[logs.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6560646/logs.zip)

AlCalzone, 2021-05-28:
The log seems clean up until the point where it suddenly dies. Did anything else change? USB Port or something?

cinadr, 2021-05-29:
No nothing to mention or relevant.

cinadr, 2021-05-29:
Same with version 4.5.
[logs.zip](https://github.com/zwave-js/node-zwave-js/files/6565229/logs.zip)
What to do to get more info on this? Any option to switch to trace?

AlCalzone, 2021-05-29:
I don't think there's anything to be logged when nothing happens. I guess I need to add a detection when the stick no longer responds and soft-reset it.

> Might Windows be a problem?

I've never had any problems with Windows.

cinadr, 2021-05-30:
One more thing to add here:
I had an UPS behind my server but it is currently removed for battery exchange. Might be some voltage floating behind these errors as it appeared after I removed the UPS. But I have no other errors in the system (unifi switch and POE APs, cameras Zigbee Donge, etc. running from this UPS) Just thinking. 
Thank you.

AlCalzone, 2023-06-22:
From: https://github.com/zwave-js/zwave-js-ui/issues/3146
Soft-reset isn't always possible. In some cases, the serialport needs to be reopened.

trankin, 2023-08-22:
Is there a consistent way for me to be able to detect when the serial device is not responding.  Currently when it fails it's pretty silent.   I could automate restarting the service if I could find a way to detect when it gets into this state.   I'm imagining sending some command that requires serial processing as a response to confirm that communication is working as expected?

AlCalzone, 2023-08-22:
Not yet. I recently added support for tracking controller status in https://github.com/zwave-js/node-zwave-js/pull/6174. Updating that status when the controller is unresponsive is still TODO though.

PR #6244 (2023-09-06): feat: detect an unresponsive stick and reset it
This PR makes use of the previously-added `Unresponsive` controller status by adding a detection for an unresponsive controller (not ACKing serial API commands). When this situation happens, we soft-reset the controller in an attempt to recover. If that does not help either, the driver instance gets destroyed.
This should trigger a restart in most applications, which effectively closes and re-opens the serial port.

fixes: #2723
closed 2023-09-06 by zwave-js-bot