zwave-js/zwave-js #2724: UFairy GR-B3-3 Not properly recognized

AlbertoLopSie, 2021-05-28:
**Checklist:**
-   [x] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [x] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**
On ZWave2MQTT this multichannel module is not recognized properly and is not possible to address any of the 3 channel switches.

A similar problem on my previous implementation using OpenZWave was solved applying [this PR](https://github.com/OpenZWave/open-zwave/pull/1567) I submitted to the OZW author and applied on v1.6.

**Device information**

[UFairy G.R. Tech / 3 Gang In-wall Switch](https://devices.zwave-js.io/?jumpTo=0x0152:0x0500:0x0005:0.0)
35

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [x] **Yes**, in: _OpenZWave 1.4 with the above PR_ applied

**How are you using `node-zwave-js`**

-   [x] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version 2021.5.5
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Add a UFairy GR-B3-3 module to your ZWave network
2. Experience the bug

**Additional context**

**Logfile:**

[zwavejs_2021-05-28.log](https://github.com/zwave-js/node-zwave-js/files/6563172/zwavejs_2021-05-28.log)

[zwave_js_dump.json.zip](https://github.com/zwave-js/node-zwave-js/files/6563138/zwave_js_dump.json.zip)

AlCalzone, 2021-05-28:
If I understand the OZW PR correctly, the device actually supports version 2, but reports version 0. Correct?

AlbertoLopSie, 2021-05-28:
Correct. Its a FW bug. The PR tried to workaround the problem.
__________________________________________________________________
Saludos,
  Alberto

Mis direcciones de mail son ***@***.*** o ***@***.***

AlCalzone, 2021-05-29:
@zwave-js-bot add compat flag to 0x0152:0x0500:0x0005
```
"commandClasses": {
	"add": {
		"0x60": {
			// This device has 3 channels, but it responds to the CC version request with 0
			"isSupported": true,
			"version": 2
		}
	}
}
```

AlbertoLopSie, 2021-05-29:
Great!!

I’ll have to clone the repo and make a docker container… any guide?

____________________________________________
Saludos,
Alberto

AlCalzone, 2021-05-29:
There's an easier way now: https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

Just put the file from the PR branch there. Make sure the log reads "user-provided device config loaded", not "embedded". The docs have a typo there.
You will need to re-interview the device.

AlbertoLopSie, 2021-05-30:
Well... It didn't work...  :-(

Copied the `gr-br-3.json` file to `store/config` result was 
```
2021-05-30T18:43:25.839Z CNTRLR   [Node 035] Embedded device config loaded
```
Copied to `store/config/devices/0x0152` same result
Copied to `store/config/config/devices/0x0152` same result

Las resort was copying the json file straight into the  docker container at 
```
docker cp gr-b3-3.json <container-id>:/usr/src/app/node_modules/@zwave-js/config/config/devices/0x0152/
```
Same result.
```
CNTRLR   [Node 035] Embedded device config loaded
```

Attaching result of `tail -f /srv/docker/zwavejs/zwavejs_2021-05-30.log | grep "Node 035"` while re-interviewing node.

[node035.log](https://github.com/zwave-js/node-zwave-js/files/6566765/node035.log)

Next step is to modify the docker image to patch the `gr-b3-3.json` file before start-up.  I'll let you know the outcome.

AlbertoLopSie, 2021-05-30:
Patching the docker image didn't work either...

```
2021-05-30T19:12:13.928Z CNTRLR » [Node 035]   querying the CC version for Multi Channel...
2021-05-30T19:12:13.933Z DRIVER » [Node 035] [REQ] [SendData]
2021-05-30T19:12:14.119Z DRIVER « [Node 035] [REQ] [ApplicationCommand]
2021-05-30T19:12:14.453Z DRIVER « [Node 035] [REQ] [ApplicationCommand]
2021-05-30T19:12:14.456Z CNTRLR   [Node 035]   does NOT support CC Multi Channel (0x60)
2021-05-30T19:12:14.456Z CNTRLR   [Node 035]   skipping query for All Switch (0x27) because max implemented vers
2021-05-30T19:12:14.457Z CNTRLR   [Node 035] [+] [Version] interviewComplete: true       [Endpoint 0] [internal]
2021-05-30T19:12:14.464Z CNTRLR   [Node 035] Embedded device config loaded
2021-05-30T19:12:14.487Z CNTRLR   [Node 035] Interviewing Z-Wave Plus Info...
2021-05-30T19:12:14.488Z CNTRLR » [Node 035] querying Z-Wave+ information...
```

The modified `gr-b3-3.json` file was copied to the original config folder:
```
/usr/src/app # ls -l /usr/src/app/node_modules/@zwave-js/config/config/devices/0x0152
total 40
-rw-r--r--    1 root     root          1667 May 25 15:01 dmwv1.json
-rw-r--r--    1 root     root          1147 May 25 15:01 gr-105.json
-rw-r--r--    1 root     root          1049 May 25 15:01 gr-201n.json
-rw-r--r--    1 root     root          1057 May 25 15:01 gr-309n.json
-rw-r--r--    1 root     root           711 May 30 16:03 gr-b3-3.json
-rw-r--r--    1 root     root           300 May 25 15:01 gr-m-202n-2.json
-rw-r--r--    1 root     root           372 May 25 15:01 topvico_tp-819zw.json
-rw-r--r--    1 root     root           340 May 25 15:01 tp-807zg.json
-rw-r--r--    1 root     root          1076 May 25 15:01 zse01.json
-rw-r--r--    1 root     root          1141 May 25 15:01 zse02.json
/usr/src/app #
```

However... It didn't do the trick. The 3 channels are not detected .

Any hint?

[node035-2.log](https://github.com/zwave-js/node-zwave-js/files/6566793/node035-2.log)

AlCalzone, 2021-05-30:
> I haven't took a look at the zwave-js code but, in OZW it was not enough to modify config files.

Here it is. However that flag doesn't seem to be applied at the correct time. Looks like you have to wait for https://github.com/zwave-js/node-zwave-js/issues/2273

> Las resort was copying the json file straight into the docker container at
> 
> ```
> docker cp gr-b3-3.json <container-id>:/usr/src/app/node_modules/@zwave-js/config/config/devices/0x0152/
> ```
> 
> Same result.
> 
> ```
> CNTRLR   [Node 035] Embedded device config loaded
> ```

Well - it **is** now embedded ;)

AlCalzone, 2021-05-30:
FYI: the external config not being loaded is a bug in zwavejs2mqtt:
https://github.com/zwave-js/zwavejs2mqtt/pull/1294

AlbertoLopSie, 2021-05-30:
> > I haven't took a look at the zwave-js code but, in OZW it was not enough to modify config files.
> 
> In zwave-js it is. However that flag doesn't seem to be applied at the correct time. Looks like you have to wait for #2273
> 
> > Las resort was copying the json file straight into the docker container at
> > ```
> > docker cp gr-b3-3.json <container-id>:/usr/src/app/node_modules/@zwave-js/config/config/devices/0x0152/
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Same result.
> > ```
> > CNTRLR   [Node 035] Embedded device config loaded
> > ```
> 
> Well - it **is** now embedded ;)

Yeaahh!!  You're right!! 

> In zwave-js it is. However that flag doesn't seem to be applied at the correct time. Looks like you have to wait for #2273

Looks like a real suspect!!  I'll wait then.  

Thanks for your time!!

closed 2021-06-08 by AlCalzone

AlbertoLopSie, 2021-06-10:
I can confirm that the last fixes (#2273 and this) solved the GR-B3-3 problem. 

They are recognized and working as expected.

Thanks!

PR #2732 (2021-06-08): fix(config): force-add support for Multi Channel CC to UFairy GR-B3-3
fixes: #2724
~~blocked by https://github.com/zwave-js/node-zwave-js/issues/2273~~