zwave-js/zwave-js #2725: Yale lock code query disables device, and kills the battery due to looping NodeInfo

psarossy, 2021-05-28:
**Checklist:**

-   [x] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [x] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**
I use ZWaveJS2MQTT but the zwave logic for this I believe comes from  ZWave JS directly hence filing here.

Yale lock node user code update / interview is strobing the device batteries to death.

I have 5 Yale Assure Touchscreen Deadbolts on my network and due to a bug in ZWaveJS2MQTT (?) after restarting, sometimes they have to be re-interviewed to work again (but just adding them the first time was the same hassle) as they all go to "Unknown manufacturer 0xXXXX".

It seems that the NodeInfo loop gets all the relevant details from the lock, then proceeds to query all user codes, I believe there are 254 (?) or so, but it gets to 15-20 when the node doesn't respond quickly enough, and it gives up on the interview, then restarts it several times, even eventually declaring the node dead, or getting stuck in an interview loop until it depletes all the batteries. While this is going on the lock is unresponsive to commands.

This makes the locks effectively useless as I can't control them and can't even use them as "dumb" locks (keypad or auto relock) as the interview eats the 4 AA batteries in a day or two.

Could the code query be deferred so the device can finish NodeInfo without it, and then in a separate, more lenient loop fetch them without maybe restarting from the start every single time?

**Device information**

Which device(s) is/are affected (manufacturer/model)? 
They report in as: Yale | Assure Touchscreen Deadbolt | YRD226 / YRC226 / YRC246 / YRD256 / YRC256 / YRD446

What is/are the node IDs?
The last one that's not completely dead and is in the log is Node 24

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [x] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version was zwavejs2mqtt 0.17.0 now 0.18.0
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

Try to add a Yale lock, or re-interview one.
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6563288/zwavejs2mqtt.log)

AlCalzone, 2021-05-28:
Please make a zwave-js log, loglevel `debug` or `silly` and attach it here as a file.

psarossy, 2021-05-28:
The DEBUG level log I attached in the original report was made based on the instructions in the template:

  If you are using zwavejs2mqtt, this is how you create the logfiles:
  * Go to Settings, Z-Wave section
  * select log level DEBUG
  * enable "log to file"

I don't have another zwave-js log I can find... If the attached on is not what is needed, how to I get it?

psarossy, 2021-05-28:
Adding more logs, trying to recover a YRL256 (Node 37) which is also failing to finish.
[zwavejs2mqtt (2).log](https://github.com/zwave-js/node-zwave-js/files/6563379/zwavejs2mqtt.2.log)

psarossy, 2021-05-28:
Ha, for the YRL256 (Node 37) it just gave up even pulling all the device properties and declared the interview done...

Debug logs from both zwave-js and zwavejs2mqtt starting around 17:20
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/6563417/zwavejs2mqtt-store.zip)

AlCalzone, 2021-05-28:
I don't see the driver log (zwavejs_<date>.log) in your latest upload. Did you upload the correct archive?

psarossy, 2021-05-28:
I guess I'll learn to read one of these days, enabled the file saving now... Logging restarted yet again, waited for a few fail cycles

Swapped the batteries in Node 43, and power cycled Node 50, both of which were showing Unknown manufacturer again.
[image]

Node 50 at least got populated and keeps trying to get the codes, 43 is still unknown but also trying to poll.

Of course now that I enabled logging both are progressing while failing for the past 2 hours.
[zwavejs2mqtt-store (1).zip](https://github.com/zwave-js/node-zwave-js/files/6563557/zwavejs2mqtt-store.1.zip)

Restarted 37 as well as it had the partial info only as mentioned above, as soon as I did that 50 was declared dead, then it starts again from scratch

AlCalzone, 2021-05-28:
It looks like the query for Node 43 is still going on, but seems to be on a good way.

Node 50 is problematic. It toggles between sending invalid security encapsulated frames (I count 50 instances in the log) and not responding at all (19 times). I don't see the reason for the invalid frame right now, but usually that means it is using an invalidated nonce. This shaky communication makes it even harder to query it.

I agree though - blindly querying all user codes during the interview is probably not a good idea.

psarossy, 2021-05-28:
Thanks for the patience :)

Node 50 is right next to 2 powered routers and tried to heal it several times (WD100+ dimmer), but ended up going dead for a second then came back and decided that it's complete now, and I can control it (garage service door), same for 43. 
[image]

Node 24 says it's complete but did not query anything and can't be controlled:
[image]

Starting an interview for it says Status: Unknown an failing... I'll power cycle it again

For Node 37 I ran another  interview, but it still didn't query half the things, and is uncontrollable:
[image]

When trying to start an interview, it immediately completes ProtocolInfo and NodeInfo in the logs but then queries further and failing a bunch, even though variables are coming back from it. Finally starts going into the code fetch phase, and fails. 

Attempting to heal it seems to fail as well, so I started a network-wide heal, which gets stuck at 33 nodes
[zwavejs2mqtt-store (3).zip](https://github.com/zwave-js/node-zwave-js/files/6563678/zwavejs2mqtt-store.3.zip)

AlCalzone, 2021-05-29:
There is definitely something fishy with your controller and/or mesh.

1. the startup has a phase of a few seconds where it seems stuck and does not respond to the driver as it should:

2. Later, when assigning the return routes to some nodes, many of them fail with the status "NoRoute", which indicates a connectivity/routing problem. This could potentially be improved with the network heal, but in your log many nodes failed to update their neighbors - including the dreaded Node 50.

3. The communication with Node 50 is also weird, which sometimes causes the interview to error. Here's an example (with some commentary):
```
zwave-js to node 50: send me a nonce!
2021-05-28T22:21:06.851Z DRIVER » [Node 050] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      70
                                  └─[SecurityCCNonceGet]

stick to zwave-js: the message was sent
2021-05-28T22:21:06.864Z DRIVER « [RES] [SendData]
                                    was sent: true

node 50 to zwave-js: here's your nonce
2021-05-28T22:21:09.228Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0x5c9fa4441a2abade

stick to zwave-js: node 50 did not receive the message
2021-05-28T22:21:10.407Z DRIVER « [REQ] [SendData]
                                    callback id:     70
                                    transmit status: NoAck
```
The same thing happens numerous times.

---

> (Node 24) Starting an interview for it says Status: Unknown an failing...

The controller stick does not respond here (see 1. above), not the node.

> For Node 37 I ran another interview, but it still didn't query half the things

When trying to figure out if it is included securely, the same thing happend as with Node 50. Probably didn't help that at that exact moment, Node 50 decided to flood the network:
```
2021-05-28T22:33:45.461Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xc9ed78920c9086b3
2021-05-28T22:33:45.495Z SERIAL « 0x0110000410320a9880c9ed78920c9086b3bc                              (18 bytes)
2021-05-28T22:33:45.496Z SERIAL » [ACK]                                                                   (0x06)
2021-05-28T22:33:45.498Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xc9ed78920c9086b3
2021-05-28T22:33:45.530Z SERIAL « 0x0110000410320a9880c9ed78920c9086b3bc                              (18 bytes)
2021-05-28T22:33:45.531Z SERIAL » [ACK]                                                                   (0x06)
2021-05-28T22:33:45.532Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xc9ed78920c9086b3
2021-05-28T22:33:45.584Z SERIAL « 0x0110000410320a9880c9ed78920c9086b3bc                              (18 bytes)
2021-05-28T22:33:45.585Z SERIAL » [ACK]                                                                   (0x06)
2021-05-28T22:33:45.587Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xc9ed78920c9086b3
2021-05-28T22:33:45.596Z SERIAL « 0x0110000410320a9880c9ed78920c9086b3bc                              (18 bytes)
2021-05-28T22:33:45.597Z SERIAL » [ACK]                                                                   (0x06)
2021-05-28T22:33:45.599Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xc9ed78920c9086b3
2021-05-28T22:33:45.880Z SERIAL « 0x0110000410320a9880c9ed78920c9086b3bc                              (18 bytes)
2021-05-28T22:33:45.881Z SERIAL » [ACK]                                                                   (0x06)
2021-05-28T22:33:45.883Z DRIVER « [Node 050] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xc9ed78920c9086b3
```

Try again, maybe you have better luck this time.

> it immediately completes ProtocolInfo and NodeInfo

These stages are pretty short - one message to the controller and one to the node. The real fun begins when querying the supported CCs and their capabilities.

---

I can't tell what exactly is wrong but you can try and put the stick on an extension cord and place it away from metallic surfaces. Sometimes even moving it a few cms can be enough.

psarossy, 2021-05-29:
TL;DR it seems moving the RPi didn't do anything adding the USB cable between it and the stick made things work, but can't tell if merely messing with it helped or the cable for sure. The Z-Stick Gen5+ says it's specifically upgraded for Raspberry Pi 4, but it seems being plugged in directly causes issues? 

Changelog on what I tried:

Moved the stick (and the RPi4) from my telco room where it was sitting on top of a plastic housed UPS to upstairs (same place in the X/Y direction, but a floor up) onto wooden workbench away from other potential interference sources.

Node 37 stated polling the codes immediately, and seemingly faster than before, and finished all the way and became published and usable if clicked on, though the HA UI stayed gray on the main view: 
[image]

 and is listed unavailable 
[image]

Tried to reinterview 24 and 50 but both were declared dead.

Powercycled Node 24/50 by pulling a battery and the ZWave module for 10s then reinserted.

Seemed to have no effect, until a few minutes later where I've seen Node 24 suddenly polling the user codes, but the in the middle timed out and aborted the interview. Which I have restated, yet again. It seems to be progressing with `Interview of node 24 has failed` sprinkled in an it doesn't progress past the product identification. This device also sits right next do at least 3 powered dimmer nodes.

Node 50 continues to be declared dead.

I see  No response from controller after 1/3 attempts. around these timestamps as well :/

After all this I found a USB extension cable and installed it between the RPi and the stick, now both 24 and 50 are querying

Waited ~20 min and all locks are now operational, started a full heal again to see if everything just reconciles now, which seems to have completed all but the devices with aggressive power saving modes (recessed door/tilt sensors and Aeotec multis on battery
[zwavejs2mqtt-store (4).zip](https://github.com/zwave-js/node-zwave-js/files/6565273/zwavejs2mqtt-store.4.zip)

AlCalzone, 2021-05-29:
> The Z-Stick Gen5+ says it's specifically upgraded for Raspberry Pi 4, but it seems being plugged in directly causes issues?

The USB ports themselves are known to cause interference, so this is definitely plausible.

> which seems to have completed all but the devices with aggressive power saving modes (recessed door/tilt sensors and Aeotec multis on battery)

I'm not sure what HA's plans for displaying the healing progress are, but at least zwavejs2mqtt should tell you that the heal is pending for the battery-powered nodes. It should work for them too if you wake them up manually.

disforw, 2021-06-10:
The Yale locks are very temperamental when first associating. I ran in to this issue before. Make sure when you first add the lock it is within 10 feet of your controller. I had to completely remove the lock from my door and put it together in my hand in order to successfully add it.

commit b12bfb3c (2021-07-03): fix(cc): do not query all user codes during interview (#2932)

fixes: #2725
fixes: #2727
closed 2021-07-17 by AlCalzone