zwave-js/zwave-js #2735: Sunricher ZV9003T3-RGBW Missing Documented Configuration Parameters

jimcortez, 2021-05-30:
**Checklist:**

-   [X ] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [ X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [ X] I have checked the troubleshooting section and my problem is **not** described there.
-   [ X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I have a Sunricher ZV9003T3-RGBW device connected through zwavejs2mqtt to home-assistant. I am looking to set the "4" parameter to "0" to allow events to get to home-assistant. However, that configuration value is not available. I have ensured that these values are available in the device database and match the github entry. I have tried manually setting the parameter through the home-assistant zwave_js.set_config_parameter service, but get a log saying that parameter value is not available.

At this point I am at a loss, the device seems to have been in the db for quite some time. I have referenced the hard-copy documentation of the device and all the documented parameters seem to be correct. However, they are just not showing up.

I also have a discussion here: https://community.home-assistant.io/t/rgbgenie-zw-3002-scene-controller/311777

**Device information**

* Retail Model: RGBgenie ZW-3002
* Zwave reported model: Sunricher ZV9003T3-RGBW
* Product Site: https://rgbgenie.com/?product=rgbgenie-color-touch-panel-and-dimmer-white-z-wave
* ZwaveJS Device Database: https://devices.zwave-js.io/?jumpTo=0x0330:0x0301:0xa106:0.0
* Documentation link: https://rgbgenie.com/wp-content/uploads/2018/11/ZW-3002-instructions.pdf

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [X ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [X ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X] `HomeAssistant` version 2021.5.5
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Include device as non-secure
2. See non-secure device is paired, all manufacturer values are correct
3. No configuration values mentioned in Zwavejs device DB are available

**Additional context**

Add any other context about the problem here.

**Logfile:**
```
2021-05-30 11:38:24.638 INFO ZWAVE: Connecting to /dev/ttyACM0
2021-05-30 11:38:24.604 INFO MQTT: MQTT is disabled
2021-05-30 11:38:24.678 INFO APP: POST /api/settings 200 234.892 ms - 768
2021-05-30 11:38:24.710 INFO ZWAVE: Zwavejs usage statistics ENABLED
[... 455 lines cut ...]
2021-05-30 11:40:08.038 INFO ZWAVE: Node 76: value added: 134-0-firmwareVersions => 1.10
2021-05-30 11:40:08.041 INFO ZWAVE: Node 76: value added: 134-0-hardwareVersion => 1
2021-05-30 11:40:12.200 INFO ZWAVE: Node 76: metadata updated: 91-0-scene-001
2021-05-30 11:40:12.203 INFO ZWAVE: Node 76: metadata updated: 91-0-scene-002
2021-05-30 11:40:12.206 INFO ZWAVE: Node 76: metadata updated: 91-0-scene-003
2021-05-30 11:40:12.263 INFO ZWAVE: Node 76: interview stage COMMANDCLASSES completed
2021-05-30 11:40:12.267 INFO ZWAVE: Node 76: interview stage OVERWRITECONFIG completed
2021-05-30 11:40:12.295 INFO ZWAVE: Node 76: interview stage NEIGHBORS completed
2021-05-30 11:40:12.299 INFO ZWAVE: Node 76: interview stage COMPLETE completed
2021-05-30 11:40:12.317 INFO ZWAVE: Node 76: value added 76-134-0-libraryType => 3
2021-05-30 11:40:12.319 INFO ZWAVE: Node 76: value added 76-134-0-protocolVersion => 4.61
2021-05-30 11:40:12.321 INFO ZWAVE: Node 76: value added 76-134-0-firmwareVersions => 1.10
2021-05-30 11:40:12.323 INFO ZWAVE: Node 76: value added 76-134-0-hardwareVersion => 1
2021-05-30 11:40:12.325 INFO ZWAVE: Node 76: value added 76-114-0-manufacturerId => 816
2021-05-30 11:40:12.327 INFO ZWAVE: Node 76: value added 76-114-0-productType => 769
2021-05-30 11:40:12.329 INFO ZWAVE: Node 76: value added 76-114-0-productId => 41222
2021-05-30 11:40:12.333 INFO ZWAVE: Node 76: value added 76-91-0-slowRefresh => undefined
2021-05-30 11:40:12.335 INFO ZWAVE: Node 76: value added 76-91-0-scene-001 => undefined
2021-05-30 11:40:12.336 INFO ZWAVE: Node 76: value added 76-91-0-scene-002 => undefined
2021-05-30 11:40:12.338 INFO ZWAVE: Node 76: value added 76-91-0-scene-003 => undefined
2021-05-30 11:40:12.339 INFO ZWAVE: Node 76: value added 76-43-0-sceneId => undefined
2021-05-30 11:40:12.341 INFO ZWAVE: Node 76: value added 76-43-0-dimmingDuration => undefined
2021-05-30 11:40:12.353 INFO ZWAVE: Node 76: interview COMPLETED, all values are updated
2021-05-30 11:40:12.356 INFO ZWAVE: Node 76 ready: Sunricher - ZV9003T3-RGBW (RGBW 3 Scene Wall Controller)
```

AlCalzone, 2021-05-30:
Please make a **zwave-js** (driver) log, loglevel `debug` or `silly` and attach it here as a file. The one you posted is from the UI (zwavejs2mqtt).

jimcortez, 2021-05-31:
My apologies! I found the right log, see below

```
2021-05-31T17:41:41.780Z DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                                  ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                                    ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                                   ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                                  ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                                  ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
2021-05-31T17:41:41.785Z DRIVER   version 7.7.0
2021-05-31T17:41:41.787Z CONFIG   version 7.7.0
2021-05-31T17:41:41.788Z DRIVER   
2021-05-31T17:41:41.788Z DRIVER   starting driver...
2021-05-31T17:41:41.800Z DRIVER   opening serial port /dev/ttyACM0
2021-05-31T17:41:41.827Z DRIVER   serial port opened
2021-05-31T17:41:41.828Z SERIAL » [NAK]                                                                   (0x15)
2021-05-31T17:41:41.835Z DRIVER   Checking for configuration updates...
2021-05-31T17:41:42.354Z DRIVER   No configuration update available...
2021-05-31T17:41:43.392Z DRIVER   loading configuration...
2021-05-31T17:41:43.404Z CONFIG   Using external configuration dir /data/db
2021-05-31T17:41:44.921Z DRIVER   beginning interview...
2021-05-31T17:41:44.923Z DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                                  1 registered
2021-05-31T17:41:44.923Z DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                                  1 registered
2021-05-31T17:41:44.924Z DRIVER   added request handler for ReplaceFailedNode (0x63)...
                                  1 registered
2021-05-31T17:41:44.925Z CNTRLR   beginning interview...
2021-05-31T17:41:44.925Z CNTRLR   querying version info...
2021-05-31T17:41:44.933Z SERIAL » 0x01030015e9                                                         (5 bytes)
2021-05-31T17:41:44.935Z DRIVER » [REQ] [GetControllerVersion]
2021-05-31T17:41:44.937Z SERIAL « [ACK]                                                                   (0x06)
2021-05-31T17:41:44.940Z SERIAL « 0x011001155a2d5761766520332e3935000199                              (18 bytes)
2021-05-31T17:41:44.942Z SERIAL » [ACK]                                                                   (0x06)
2021-05-31T17:41:44.945Z DRIVER « [RES] [GetControllerVersion]
                                    payload: 0x5a2d5761766520332e39350001
2021-05-31T17:41:44.950Z CNTRLR   received version info:
                                    controller type: Static Controller
                                    library version: Z-Wave 3.95
2021-05-31T17:41:44.950Z CNTRLR   querying controller IDs...
2021-05-31T17:41:44.955Z SERIAL » 0x01030020dc                                                         (5 bytes)
2021-05-31T17:41:44.956Z DRIVER » [REQ] [GetControllerId]
2021-05-31T17:41:44.957Z SERIAL « [ACK]                                                                   (0x06)
2021-05-31T17:41:44.959Z SERIAL « 0x01080120f824ba7901c8                                              (10 bytes)
2021-05-31T17:41:44.961Z SERIAL » [ACK]                                                                   (0x06)
2021-05-31T17:41:44.962Z DRIVER « [RES] [GetControllerId]
                                    payload: 0xf824ba7901
2021-05-31T17:41:44.966Z CNTRLR   received controller IDs:
                                    home ID:     0xf824ba79
                                    own node ID: 1
2021-05-31T17:41:44.970Z CNTRLR   querying controller capabilities...
2021-05-31T17:41:44.975Z SERIAL » 0x01030005f9                                                         (5 bytes)
2021-05-31T17:41:44.976Z DRIVER » [REQ] [GetControllerCapabilities]
2021-05-31T17:41:44.982Z SERIAL « [ACK]                                                                   (0x06)
2021-05-31T17:41:44.984Z SERIAL « 0x010401051ce3                                                       (6 bytes)
2021-05-31T17:41:44.985Z SERIAL » [ACK]                                                                   (0x06)
2021-05-31T17:41:44.987Z DRIVER « [RES] [GetControllerCapabilities]
                                    payload: 0x1c
2021-05-31T17:41:44.992Z CNTRLR   received controller capabilities:
                                    controller role:     primary
                                    is in other network: false
                                    is SIS present:      true
                                    was real primary:    true
                                    is a SUC:            true
2021-05-31T17:41:44.992Z CNTRLR   querying API capabilities...
2021-05-31T17:41:44.997Z SERIAL » 0x01030007fb                                                         (5 bytes)
2021-05-31T17:41:44.998Z DRIVER » [REQ] [GetSerialApiCapabilities]
2021-05-31T17:41:45.001Z SERIAL « [ACK]                                                                   (0x06)
2021-05-31T17:41:45.003Z SERIAL « 0x012b0107010000860101005afe81ff884f1f0000fb9f7da067000080008086000 (45 bytes)
                                  000e87300000e0000600000fa
[... 753 lines cut ...]
2021-05-31T17:43:13.655Z SERIAL « 0x0104011301e8                                                       (6 bytes)
[... cut ...]
2021-05-31T17:43:16.347Z CNTRLR   [Node 076] [~] [Manufacturer Specific] productId: 41222 => 41222  [Endpoint 0]
2021-05-31T17:43:16.352Z SERIAL » [ACK]                                                                   (0x06)
2021-05-31T17:43:16.354Z DRIVER « [Node 076] [REQ] [ApplicationCommand]
                                  └─[ManufacturerSpecificCCReport]
                                      manufacturer id: 0x0330
                                      product type:    0x0301
                                      product id:      0xa106

```

AlCalzone, 2021-05-31:
You didn't **re-interview** the device like I asked.

jimcortez, 2021-05-31:
Hey @AlCalzone thanks for following up! I did hit the re-interview button in zwave2mqtt for node 76. Is there something specific in the logs I should look for to make sure it is actually interviewing?

I see the following lines in the log:
```
2021-05-31T17:42:43.123Z CNTRLR   [Node 076] Beginning interview - last completed stage: None
2021-05-31T17:42:43.125Z CNTRLR   [Node 076] new node, doing a full interview...
```

AlCalzone, 2021-06-01:
Sorry - I didn't find that last time. According to your device, it does not support Configuration CC:
```
2021-05-31T17:43:02.566Z CNTRLR « [Node 076] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Association
                                  · Association Group Information
                                  · Multi Channel Association
                                  · Transport Service
                                  · Version
                                  · Manufacturer Specific
                                  · Device Reset Locally
                                  · Powerlevel
                                  · Security
                                  · Security 2
                                  · Supervision
                                  · Central Scene
                                  · Firmware Update Meta Data
                                  controlled CCs:
                                  · Multilevel Switch
                                  · Color Switch
                                  · Scene Activation
                                  · Scene Actuator Configuration
```
The [manual](https://www.sunricher.com/media/resources/manual/SR-ZV9001T3-RGBW-US%20Instruction%20V1.pdf) I found doesn't list it either.

jimcortez, 2021-06-01:
This is the documentation for the device: https://rgbgenie.com/wp-content/uploads/2018/11/ZW-3002-instructions.pdf See "Configuration Parameters"

I see there is has "COMMAND_CLASS_CONFIGURATION". Is this just an issue with the device not actually responding with the proper CCs?

And it seems there is an entry for this in the device DB: https://devices.zwave-js.io/?jumpTo=0x0330:0x0301:0xa106:0.0

Essentially, I am trying to change parameter 4 (SCENE_SEND_TO_WHERE) to 0 so that I get the events sent to home-assistant.

jimcortez, 2021-06-01:
I dug into it a bit and agree with you that the device is not responding with the proper capabilities. I think the instructions provided with the device (and linked on amazon) are for a different unit. RGBGenie seems to be re-selling/re-branding sunricher devices, but the actual chips seem to be mismatched.

closed 2021-06-01 by jimcortez

dorkmatt, 2022-02-10:
> I will chase this with the company and at least post an update here for any future travelers.
> 
> cc @RGBGenie

@jimcortez Did ya ever make contact with RGBgenie to understand the discrepancy?

jimcortez, 2022-03-01:
I did, they agreed it was an issue, then I never followed up. Wish I had better news.

My only idea to salvage the now-unreturnable device is to try and make a dummy node and add it to the same group... Not sure if that is even possible.