zwave-js/zwave-js #2738: Neo Coolcam PIR Sensor Configuration Issue

bwosborne2, 2021-05-30:
**Describe the bug**

Configuration parameter 6 - LED Blink Enable starts with an illegal default value of 5 (should be 1) and cannot be set to 0 to disable blink.

**Device information**
Neo Coolcam PIR Sensor https://devices.zwave-js.io/?jumpTo=0x0258:0x0003:0x0083:0.0
Currently this is Node ID 8.

**Did this use to work before?**

-   [ x  ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ x  ] Other: home Assistant `zwavejs2mqtt` integration

[image]

**To Reproduce**

Steps to reproduce the behavior:

1. Go to 'Control Panel'
2. Click on 'Node 8'
3. Scroll down to 'Configuration' and click on it.
4. Scroll down to '[8-112-0-10] LED Blink Enable'
5. See error, The value is '[5] Custom' instead of '[1] Enable LED blinking (default)'

**Additional context**

This is a test environment. I excluded, factory defaulted the node, and included fresh to collect the attached DEBUG log file.
**Logfile:**
[z2js-inclusion-log.txt](https://github.com/zwave-js/node-zwave-js/files/6566896/z2js-inclusion-log.txt)

AlCalzone, 2021-05-31:
1. Do you have a manual for your exact device? The device ID ends with `0x008d` while all others for this one end in `0x0083`. I think this might be a misindentification on our part, which explains why you're seeing this invalid config value. 

2. That's not the requested logfile. The section you deleted from the template seems pretty explicit to me, but people get it wrong **ALL the time**. If you have any suggetions how to make this clearer, I'm all ears.

```
ATTACH(!) `zwave-js` logfile with DEBUG or VERBOSE loglevel here. Please no links or gists or embedded logs.
Please make sure to upload the correct log. If you're unsure, the correct one is called `zwave-<number>.log` and starts with

███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
  ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
 ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝

If you are using zwavejs2mqtt, this is how you create the logfiles:
* Go to Settings, Z-Wave section
* select log level DEBUG
* enable "log to file"

For HomeAssistant, this is how you do it:
Home Assistant -> settings -> Integrations -> Z-Wave JS -> Configure -> Create dump -> zip the json file and post it here.
```

bwosborne2, 2021-05-31:
I followed the zwavejs2mqtt instructions. and only ended up with a zwavejs2mqtt.log. Since ht ran for a while before my test I only included the fresh interview of the node.

Perhaps the zwavejs2mqtt HA integration changes logging?

bwosborne2, 2021-05-31:
OK I grabbed the log (after excluding & re-including it is nor node 9) and I think I found the issue.  LED Enable is parameter 10. You list Parameter 9 as 2 bytes but the device seems to indicate it is 1 byte, If I read the log correctly.

[zwavejs_2021-05-31.log](https://github.com/zwave-js/node-zwave-js/files/6571055/zwavejs_2021-05-31.log)

AlCalzone, 2021-05-31:
> after excluding & re-including

I didn't have time to look at the log yet, but this is not necessary. You can re-interview (query all capabilities) without excluding the node. This should be possible in both HA and zwavejs2mqtt.

bwosborne2, 2021-05-31:
OK I wanted to get a complete interview so I excluded, factor reset the node, & then included. In my experience some of these cheap devices do not include properly (even insecurely) without a factory reset, for some reason.

AlCalzone, 2021-05-31:
This seems to be a different device than what we have assigned it to in the DB. Do you have a manual for it?
Maybe its this one? https://szneo.com/en/products/show.php?id=197

bwosborne2, 2021-05-31:
That may be it but I looked up the order. 
It is NAS-PD02Z. this OZW xml looks like the one.
https://raw.githubusercontent.com/OpenZWave/open-zwave/master/config/shenzen_neo/nas-pd02z.xml

AlCalzone, 2021-06-01:
[neo-coolcam-motion-sensor-2.pdf](https://github.com/zwave-js/node-zwave-js/files/6580365/neo-coolcam-motion-sensor-2.pdf)

Found a manual that actually includes a description for that parameter. In case anyone wants to go ahead and make a separate config file for the V2 version.

bwosborne2, 2021-06-01:
I am using the HA `zwavejs2mqtt` integration using websickets, Is there any way to develop & test the config with this integration?

I guess I would need to move to a standalone `zwavejs2mqtt` and use MQTT? I am not sure how to interface websockets standalone.

AlCalzone, 2021-06-01:
If you use zwavejs2mqtt (needs to be v4.5.1 or newer), there is a `config` folder in the `store` directory. You can put your file there to test: https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config
Note that you'll need to re-interview for changes to take effect.

bwosborne2, 2021-06-01:
I just looked at the HA integration (4.5.0) and do not see a config folder in store.  

I will move to a standalone version of zwavejs2mqtt. I assume I would also need a custom version of the PR sensor excluding my device ID to avoid conflict.

AlCalzone, 2021-06-02:
> I assume I would also need a custom version of the PR sensor excluding my device ID to avoid conflict.

Yes.

> I guess I need to manually make the config folder? I thought there would be an empty folder created,

@robertsLando?

bwosborne2, 2021-06-02:
Just to clarify, I am not running the latest `zwavejs2mqtt` from Docker and a local store.
I have it successfully communicating with HA using websockets.

AlCalzone, 2021-06-02:
Oh, you **need** 4.5.1 for the custom directory to work correctly.

bwosborne2, 2021-06-02:
I installed it last night.
[image]

bwosborne2, 2021-06-02:
I manually made a config folder. I tried putting the json file directly in there or in a folder with the hex of the manufacturer.
zwavejs2mqtt still takes the value from the internal config entry, not the replacement.

robertsLando, 2021-06-03:
the config folder should be created by the user, this can be easily done from the store UI

bwosborne2, 2021-06-03:
@robertsLando 
I created the config folder and my file seems to be ignored. The device is currently incorrectly defined in the database, grouped with similar but different ones.

bwosborne2, 2021-06-06:
Created PR 2794 to resolve this issue.,

https://github.com/zwave-js/node-zwave-js/pull/2794

blhoward2, 2021-06-07:
I'm not sure what this PR solves? It renames the device but both have the same parameters according to the manual and neither sets the default value for parameter 10 (not 6) to 5? It should allow setting 0. Am I missing something? As it stands this is:

```
		"10": {
			"label": "LED Blink Enable",
			"description": "Defines whether the LED blinking is on / off",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Disable LED blinking",
					"value": 0
				},
				{
					"label": "Enable LED blinking",
					"value": 1
				}
			]
		},
```

closed 2021-06-18 by blhoward2