zwave-js/zwave-js #2755: FIBARO Walli Double Switch FGWDSEU: "Parameter 20" cannot be set to "3"

rnews9, 2021-06-01:
**Describe the bug**

To operate the FIBARO Walli Double Switch FGWDSEU in 2-way/3-way configuration in combination with non-smart switches, the FIBARO switch has to be in "Single Switch Mode" and the parameter 20 has to be set to "3".

But the parameter 20 ("Buttons operation") can only be set to "1" oder "2". The value "3" is out of target range.

**Device information**

FIBARO WALLI DOUBLE SWITCH FGWDSEU (Firmware: 5.2)  

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [X] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X] `HomeAssistant` version core-2021.5.5
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. In Home Assistant preferences select the device,
2. enter Z-Wave Device Configuration,
3. try to set parameter 20 "Buttons operation" to the value 3
4. -> value not accepted.

**Additional context**
Fibaro Manual:
[https://manuals.fibaro.com/content/manuals/en/FGWDSEU-221/FGWDSEU-221-T-EN-v1.1.pdf](https://manuals.fibaro.com/content/manuals/en/FGWDSEU-221/FGWDSEU-221-T-EN-v1.1.pdf)
Page 9: "Diagram 3: Example connection for 2-way switching"
Page 15: "7.2: Single/Double Switch operating mode"
Page 16: "7.3: Multiway switching"
Page 26: "Advanced Parameters: 20. Buttons operation"

**Logfile:**

[zwave_js_dump_fibaro_walli.txt](https://github.com/home-assistant/core/files/6577110/zwave_js_dump_fibaro_walli.txt)

AlCalzone, 2021-06-01:
We don't have the ability to offer this config value conditionally, so we should just add it in. Accepting PRs.

blhoward2, 2021-06-18:
I'm confused by this, as it is in range. @rnews9 what is it that you cannot do?

```
		"20": {
			"label": "Buttons operation",
			"valueSize": 1,
			"minValue": 1,
			"maxValue": 4,
			"defaultValue": 1
		},
```

kpine, 2021-06-18:
The node dump is indicating a different device and config file that defines a smaller range: https://github.com/zwave-js/node-zwave-js/blob/0b7b251cdb8eb1dbce5db198e70b423b96859032/packages/config/config/devices/0x010f/fgwds_5.1.json#L67-L73

Looks like it's just a difference between firmware versions, 5.1 vs 5.2. Here's the alliance DB for [FIBARO Wall Single Switch FGWDSEU-221 5.02](https://products.z-wavealliance.org/products/3662?selectedFrequencyId=-1), with the config param definition for 20 have values 1-3. Here's the [5.01 version](https://products.z-wavealliance.org/products/3393?selectedFrequencyId=-1), same product id & type, that matches the zwave-js db.

```
            "manufacturerId": 271,
            "productId": 4096,
            "productType": 6657,
            "firmwareVersion": "5.2",
            "zwavePlusVersion": 1,
            "deviceConfig": {
              "filename": "/usr/src/node_modules/@zwave-js/config/config/devices/0x010f/fgwds_5.1.json",
              "manufacturer": "Fibargroup",
              "manufacturerId": 271,
              "label": "FGWDS",
              "description": "WALLI SWITCH FIBARO SINGLE SWITCH",
              "devices": [
                {
                  "productType": 6657,
                  "productId": 4096
                }
              ],
              "firmwareVersion": {
                "min": "5.1",
                "max": "255.255"
              },
              "paramInformation": {
                "_map": {}
              },
              "isEmbedded": true
            },
            "label": "FGWDS",
```

kpine, 2021-06-18:
Also, the device described in this issue, "Double Switch" seems to be in conflict with the Z-Wave Alliance DB, which says it's the "Single Switch".

rnews9, 2021-06-18:
Hi @blhoward2, thank you for looking into that.
While you were typing I looked at the firmware version as well. Your are right, in the device manual for the older firmware 5.0-5.1 the value range is from "1" to "2".

According to the device manual for firmware 5.2 or higher the value range is supposed to be from "1" to "4" in "single swith mode" or from "1" to "3" if the device is in "double switch mode".

I checked the firmware on my device, it's "5.2". The Z-Wave network dump confirms version 5.2: 
```
            "manufacturerId": 271,
            "productId": 4096,
            "productType": 6657,
            "firmwareVersion": "5.2",
            "zwavePlusVersion": 1,
```

blhoward2, 2021-06-18:
The manual OP linked though is for a double.

kpine, 2021-06-18:
Gotta love the DB:
[image]

They manual they link is in fact for the "Double Switch".

blhoward2, 2021-06-18:
Yeah, not looking forward to fixing the Fibaro folder in general...

blhoward2, 2021-06-18:
If it's matching the fgwds file then it looks like we need to confirm the parameters and move it over to the correct file. Or combine the two of it should be one file with conditions. We do have a file for the FGWDSEU.

AlCalzone, 2021-06-18:
> Yeah, not looking forward to fixing the Fibaro folder in general...

They have tons of incorrect parameters that really should be partials, so that's going to be fun.

AlCalzone, 2022-03-17:
This was recently fixed

rnews9, 2022-03-17:
Hi @AlCalzone , thanks for your effort! Unfortunately the issue persists. 
The device is using this config:
https://devices.zwave-js.io/?jumpTo=0x010f:0x1a01:0x1000:5.2
(Parameter 20 = min. "1" and max. "2")

According to the device manual for firmware 5.2 or higher the value range is supposed to be from "1" to "4" in "single switch mode" or from "1" to "3" if the device is in "double switch mode".

Or am I missing something?

AlCalzone, 2022-03-17:
> Or am I missing something?

Yes, this PR which was merged recently: https://github.com/zwave-js/node-zwave-js/pull/4370
Its not released yet, so it doesn't show up in the config DB browser yet.