zwave-js/zwave-js #2763: Missing device MCOHome MH-S220

arjen-w, 2021-06-02:
**Which device is missing?**

MCOHome MS-S220

**What are the IDs?**

Manufacturer ID: 0x015f  
Product type: 0x2201
Product ID: 0x1252

**Did you check if we really don't have it?**

- [x] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [x] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [x] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

**Do you have a link to the PDF manual?**
Link to manual: [here]( http://mcohome.com/index.php?c=Front/DownDetail&a=downloadFujian&name=MH-S220&path=L2NvbWRhdGEvOTUzMDEvcHJvZHVjdGZ1amlhbi8yMDIwMDYwMTE4MDEyNjVlZDRkMWY2NzVkZWMucGRm
)
OpenHab config info: [here](https://www.opensmarthouse.org/zwavedatabase/1153)

AlCalzone, 2021-06-02:
@zwave-js-bot import config 1153 from oh

AlCalzone, 2021-06-02:
https://github.com/zwave-js/node-zwave-js/pull/2765 can serve as a template but the original is pretty terrible.

PR #4085 (2022-01-16): feat(config): add MCOHome MH-S220
fixes #2763
closed 2022-01-16 by AlCalzone