zwave-js/zwave-js #2775: Missing configuration: Kwikset 910 Z-Wave Plus Lock

andynbaker, 2021-06-04:
**Which device is missing?**

Kwikset 910 Lock, Z-Wave Plus

**What are the IDs?**

Manufacturer ID: 0x0090 
Product type: 0x0003
Product ID: 0x0000 

**Did you check if we really don't have it?**

- [ X] It is not in the [configuration DB](https://devices.zwave-js.io/)
- [ X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [ X] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

Here's an export of the node from the zwavejs2mqtt interface (forgive me if this is useless):
[node_47.json.zip](https://github.com/zwave-js/node-zwave-js/files/6595196/node_47.json.zip)

Basically, I ditched the old z-wave card in favor of a z-wave plus card I purchased on ebay. My coordinator sees the lock and it pairs without a hitch, except it shows up as an unknown mfg and device. The lock responds to unlock and lock commands issued from home assistant. I can also use zwavejs2mqtt to successfully send new user codes. But the alarm_type and status are not being reported at all, so I can't which code was used to unlock the lock.

Any help is greatly appreciated.

[image]
[image]
[image]

andynbaker, 2021-06-04:
Have a beer on me in advance!

AlCalzone, 2021-06-04:
Please post a driver log (level debug) of a re-interview and doing something that should generate an alarm value. Attach it as a file.

andynbaker, 2021-06-04:
How's this? I re-interviewed it and locked and unlocked via service calls from home assistant. I manually locked it and unlocked it at the door, but those events did not show up in the logs.

[debug.log](https://github.com/zwave-js/node-zwave-js/files/6598573/debug.log)

AlCalzone, 2021-06-06:
> How's this?

Nope, sorry - that's not the correct logfile. If you're using zwavejs2mqtt, here's how you do it:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

Not sure about HA - you'll have to ask in the HA community.

andynbaker, 2021-06-07:
Let's try that again:

[zwavejs_2021-06-06.log](https://github.com/zwave-js/node-zwave-js/files/6605227/zwavejs_2021-06-06.log)

blhoward2, 2021-06-07:
Because you're changing cards, this isn't a Kwikset 910 and the file shouldn't say that it is. That model number comes back to a conversion kit applicable to many models. https://www.kwikset.com/products/detail/kwikset-convert-smart-lock-conversion-kit-with-z-wave-technology

andynbaker, 2021-06-07:
Yeah it looks like they used this module in many of their zwave products, including the 910. You'll have to excuse me, because I have no idea how the module works with the different locks they use it in. 

The most important thing for me is reporting manual lock and unlock, and I'd love to be able to see what user code was used to unlock the lock.

AlCalzone, 2021-06-07:
From the log it looks like the lock supports and uses the standardized Notifications, which are exposed either as values or notification events. However the log only includes two notifications for RF (un)lock, so I can't verify what it does on manual lock.

blhoward2, 2021-06-08:
Right, but my question is does it need its own file as a new version of the 910 or can it be added to the 910 file, which has alarm mapping compat flags?

andynbaker, 2021-06-08:
> From the log it looks like the lock supports and uses the standardized Notifications, which are exposed either as values or notification events. However the log only includes two notifications for RF (un)lock, so I can't verify what it does on manual lock.

Nothing is showing up in the logs when I manually lock and unlock. I'll try fiddling with it tomorrow.

AlCalzone, 2021-06-08:
> or can it be added to the existing 910 file, which has alarm mapping compat flags?

Should be possible

PR #2878 (2021-07-01): fix(config): add fingerprint to Kwikset 910
Closes #2775
closed 2021-07-01 by AlCalzone