zwave-js/zwave-js #2776: Thermostats multiplying and changing IDs

Cyberwizzard, 2021-06-04:
**Checklist:**

-   [x] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [x] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

After a few updates and restarts, the Z-wave nodes seem to switch IDs, causing original integration to break. The Z-TRM3 thermostats not only switched IDs but also have multiplied; they show up to four times in the overview panel. Only the first duplicate seems to have a control slider.

**Device information**

Devices:
HeatIt Z-TRM3 thermostats, Fibaro door sensor and Fibaro light switch

**Did you change anything?**

-   [x] **Yes**: updates of HA core, Zwave addon and a couple of resets without me checking the dashboard until we noticed the automation was fully dead
-   [ ] **No**

**Did this use to work before?**

-   [x] **Don't know**, I had thermostats switch IDs before but thought this was fixed at some point
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version core-2021.6.0
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [x] Other: zwave-js 0.1.24

**To Reproduce**

Steps to reproduce the behavior:

1. Open HomeAssistant
2. 

**Additional context**

Add any other context about the problem here.

**Logfile:**

As requested by zwave channel on Discord, here is the JSON from the network config:

[zwave_js_dump.json.txt](https://github.com/zwave-js/node-zwave-js/files/6598064/zwave_js_dump.json.txt)

Cyberwizzard, 2021-06-04:
[image]

This is what the duplication looks like

Cyberwizzard, 2021-06-04:
This is the view of the light switch and the door sensor after the node IDs changed:
[image]

Is there a way to rename them back to the old IDs so the automation works again?

AlCalzone, 2021-06-06:
> thermostats not only switched IDs but also have multiplied; they show up to four times in the overview panel

From what I see in the dump file, at least node 11 does report 4 thermostat endpoints, each one with values. I'm not sure **why** that happens and I can't tell without a driver logfile (loglevel "debug") of a re-interview.

Cyberwizzard, 2021-06-07:
@AlCalzone I re-ran the interview using debug log level (now the brand name is detected again) - but how do I extract the logs? I only get 100 lines in the integration log UI and there seems to be no button to download the logs?

AlCalzone, 2021-06-07:
No idea... Any chance you got access to zwavejs2mqtt? You can download the logfiles there.

Cyberwizzard, 2021-06-18:
I tried getting away from the MQTT integration as I was told the new method was better...

Will investigate soon if I can get the logs, currently including a new thermostat does not even work.

closed 2021-07-13 by zwave-js-assistant[bot]