zwave-js/zwave-js #2777: External database config directory is not created

kpine, 2021-06-04:
**Checklist:**

-   [X] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

If `ZWAVEJS_EXTERNAL_CONFIG` is set, but the directory does not exist, the external DB path is never created and populated.

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [X] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [X] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Ensure the directory specified by the env var does not exist
2. Start zwavejs2mqtt or your chosen application
3. See the directory is never created

**Additional context**

This was previously logged in #2602 and a fix was submitted in PR #2603. Admittedly my node.js skills are not great, but I don't think the fix was sufficient to addresses the original problem. If the directory does not exist, then the `syncExternalConfigDir` function exits before it tries to create it.

AlCalzone, 2021-06-06:
The location you highlighted was fine. But the code short-circuited in another location when the dir didn't exist.

closed 2021-06-06 by AlCalzone

kickouille, 2021-06-08:
I confirm this is working as expected now, directory is created as it should, thank you very much for your work guys!

PR #2784 (2021-06-06): fix: sync external config dir even when it doesn't exist
fixes: #2777