zwave-js/zwave-js #2783: August SmartLock Pro Added But Not Working

aLTeReGo-SWI, 2021-06-05:
**Checklist:**

-   [ ] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I am able to add/include as well as remove/exclude my August Pro smart lock to Z-Wave JS but it times out on any/all commands. Re-interviewing (sucsesfully) or removing and re-adding (also sucsesfully) does not help

**Device information**

Which device(s) is/are affected (manufacturer/model)?  
NodeID 38 - August Smart Lock Pro
What is/are the node IDs?

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X] **No**

**Did this use to work before?**

-   [X] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.
2021-06-05T19:00:18.671Z CNTRLR   starting inclusion process...
2021-06-05T19:00:18.710Z CNTRLR   handling add node request (status = Ready)
2021-06-05T19:00:18.711Z CNTRLR     the controller is now ready to add nodes
2021-06-05T19:00:20.682Z CNTRLR   handling add node request (status = NodeFound)
2021-06-05T19:00:20.872Z CNTRLR   handling add node request (status = AddingSlave)
2021-06-05T19:01:25.937Z CNTRLR   handling add node request (status = ProtocolDone)
2021-06-05T19:01:25.937Z CNTRLR   stopping inclusion process...
2021-06-05T19:01:26.000Z CNTRLR   handling add node request (status = Done)
2021-06-05T19:01:26.000Z CNTRLR   done called for 38
2021-06-05T19:01:26.002Z CNTRLR   finished adding node 38:
                                    basic device class:    Routing Slave
                                    generic device class:  Entry Control
                                    specific device class: Secure Keypad Door Lock
                                    supported CCs: 
                                    · Basic (0x20)
                                    · Door Lock (0x62)
                                    · User Code (0x63)
                                    · Manufacturer Specific (0x72)
                                    · Security (0x98)
                                    · Version (0x86)
                                    · Z-Wave Plus Info (0x5e)
                                    · Transport Service (0x55)
                                    · Security 2 (0x9f)
                                    controlled CCs: 
2021-06-05T19:01:26.004Z CNTRLR » [Node 038] Assigning SUC return route...
2021-06-05T19:01:26.017Z CNTRLR   the inclusion process was stopped
2021-06-05T19:01:37.394Z CNTRLR   [Node 038] Security bootstrapping failed, the node is included insecurely: a s
                                  ecure inclusion timer has elapsed.
2021-06-05T19:01:37.397Z CNTRLR   [Node 038] Cannot configure Z-Wave+ Lifeline association: Node does not suppor
                                  t associations...
2021-06-05T19:01:37.401Z CNTRLR   [Node 038] Beginning interview - last completed stage: None
2021-06-05T19:01:37.402Z CNTRLR   [Node 038] new node, doing a full interview...
2021-06-05T19:01:37.403Z CNTRLR » [Node 038] querying protocol info...
2021-06-05T19:01:37.426Z CNTRLR « [Node 038] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Entry Control
                                  specific device class: Secure Keypad Door Lock
                                  node type:             Routing End Node
                                  is always listening:   false
                                  is frequent listening: 1000ms
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2021-06-05T19:01:37.891Z CNTRLR   [Node 038] Interview stage completed: ProtocolInfo
2021-06-05T19:01:37.893Z CNTRLR » [Node 038] querying node info...
2021-06-05T19:01:39.213Z CNTRLR « [Node 038] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Transport Service
                                  · Security
                                  · Security 2
                                  controlled CCs:
2021-06-05T19:01:39.601Z CNTRLR   [Node 038] Interview stage completed: NodeInfo
2021-06-05T19:01:39.605Z CNTRLR » [Node 038] querying securely supported commands...
2021-06-05T19:01:49.823Z CNTRLR   [Node 038] Timed out while waiting for a response from the node
2021-06-05T19:01:49.824Z CNTRLR   [Node 038] The node is not included securely. Continuing interview non-securel
                                  y.
2021-06-05T19:01:49.856Z CNTRLR   [Node 038] Interviewing Manufacturer Specific...
2021-06-05T19:01:49.856Z CNTRLR » [Node 038] querying manufacturer information...
2021-06-05T19:02:01.120Z CNTRLR   [Node 038] Timed out while waiting for a response from the node
2021-06-05T19:02:01.160Z CNTRLR   [Node 038] Interviewing Version...
2021-06-05T19:02:01.160Z CNTRLR » [Node 038]   querying the CC version for Version...
2021-06-05T19:02:12.616Z CNTRLR   [Node 038] Timed out while waiting for a response from the node
2021-06-05T19:02:12.616Z CNTRLR   [Node 038] CC version query for Version timed out - assuming the node supports
                                   version 1...
2021-06-05T19:02:12.616Z CNTRLR » [Node 038] querying node versions...

marcus-j-davies, 2021-06-06:
I'm not sure if the Smart Lock **requires** Security (given it is a Lock)
if so, this log entry is very much the answer.

I have devices that will not function (but will still join a network) without being secured.

`2021-06-05T19:01:49.824Z CNTRLR [Node 038] The node is not included securely. Continuing interview non-securely.`

this is just my obersvation.

AlCalzone, 2021-06-06:
> Security bootstrapping failed, the node is included insecurely: a secure inclusion timer has elapsed.

Something went wrong during secure inclusion (I can't see what exactly because you didn't attach a logfile with loglevel "debug"), so `zwave-js` doesn't talk to the node securely. Nonetheless this behavior is in line with the Z-Wave specs.

@aLTeReGo-SWI You'll have to exclude the node, reset it and try to include it again.

closed 2021-06-06 by AlCalzone