zwave-js/zwave-js #2792: [bug] Greenwave Powernode 6 measurements always reported on instance 1

katiuskt, 2021-06-06:
### Version

**Checklist:**

- [ ] I am **not** using HomeAssistant. **Or:** a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is **not** described there.
- [X] I have read the changelog problem was **not** mentioned there.

Build/Run method

- [X] Docker
- [ ] PKG
- [ ] Snap package
- [ ] Manually built (git clone - yarn install - yarn run build )

zwavejs2mqtt version: 4.5.1
zwave-js version: 7.7.1

### Describe the bug

Two Greenwave Powernode 6 devices. Correctly detected by ZWaveJS2MQTT. Nodes 2 & 12.
Attached information: 
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/zwavejs2mqtt/files/6604535/zwavejs2mqtt-store.zip)

Each socket is expected to update their "**Electric Consumed [W]**" information into their own instances (NODE-50-X-66049, with X = 1 to 6).

Unfortunately the electric consumed information is updated regularly but instead to be assigned to the proper socket, the information is assigned to X=1

For instance, in my node 12 I have only one device connected at X=3 right now but information of power consumption is updated on 12-50-**1**-66049, not in 12-50-**3**-66049
[image]
If I click on "Refresh" button, the 12-50-3-value-66049 is updated and 12-50-1-value-66049 goes to 0 for a few seconds. Until new measurement is received which instead to updated the expected 12-50-3-value-66049 updates the 12-50-1-value-66049.

In my other device (node 2) it happens the same but it could be less obvious because I have devices in a lot of more sockets. But in logs you can see only socket 1 is updated periodically but, based on the power consumptions it is obvious to me some times the reports if for one socket or another.

It seems quite similar to me to the issue reported over OpenZWave stack: https://github.com/OpenZWave/open-zwave/issues/1700 but I don't get the technical discussion on it of how it was solved on that ZWave stack. I'm not sure if it could be helpful for ZWaveJS stack.

kWh information (NODE-50-X-value-65537 Ekectric Consumed kWh) is only updated by polling mechanism so I guess I can do the same to get "W" information updated regularly but, as I said, Electric Consumed (W) information is being sent by device periodically and ZWaveJS is aware of that. The issue is ZWaveJS is assigning the values always to socket 1.

### To Reproduce

Steps to reproduce the behavior:

1. Go to 'Control Panel'
2. Click on Greenwave PowerStrip 6 Node
3. Scroll down to 'Meter'
4. You'll see only NODE-50-1-value-66049 Electric Consumed W is updated regularly although that socket (1) isn't populated

### Expected behavior

Each NODE-50-X-value-66049 Electric Consumed W should be updated with the information if its own socket X. X=1 should inform about the electric consumed (W) by socket 1, X=2 should report power consumption of socket 2 and so on.

### Additional context

As mentioned above, it seems to me like this issue raised on other ZWave stack: https://github.com/OpenZWave/open-zwave/issues/1700

katiuskt, 2021-06-06:
Hi @AlCalzone,
Please find it attached (it was inside the .zip file attached in the opening statement but I've reenabled logging and I've attached an updated one): 
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/6604905/zwavejs2mqtt.log)
> 	Line 1821: 2021-06-06 22:39:47.329 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 10 => 11.1
> 	Line 1822: 2021-06-06 22:39:56.328 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 11.1 => 9.7
> 	Line 1844: 2021-06-06 22:40:41.332 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 9.7 => 15.9
> 	Line 1845: 2021-06-06 22:40:50.332 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 15.9 => 9.9
> 	Line 1856: 2021-06-06 22:41:17.330 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 9.9 => 11
> 	Line 1857: 2021-06-06 22:41:26.332 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 11 => 9.5
> 	Line 1864: 2021-06-06 22:41:53.333 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 9.5 => 10.7
> 	Line 1876: 2021-06-06 22:42:02.333 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 10.7 => 9.4
> 	Line 1878: 2021-06-06 22:42:11.332 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 9.4 => 11.2
> 	Line 1883: 2021-06-06 22:42:29.333 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 11.2 => 9.8
> 	Line 1889: 2021-06-06 22:45:29.335 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 9.8 => 10.8
> 	Line 1890: 2021-06-06 22:45:38.334 INFO ZWAVE: Node 12: value updated: 50-1-value-66049 10.8 => 9

Please also find the node JSON and debug_info exported from GUI for node 12 (one of my two greenwave powernode 6 devices)
[node_12.zip](https://github.com/zwave-js/node-zwave-js/files/6604906/node_12.zip)
[node_12_debug_info.txt](https://github.com/zwave-js/node-zwave-js/files/6604907/node_12_debug_info.txt)

AlCalzone, 2021-06-06:
That is not the **driver** log.
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

katiuskt, 2021-06-07:
Oh, my mistake, sorry!

I get you now. Please find it attached: [zwavejs_2021-06-07.log](https://github.com/zwave-js/node-zwave-js/files/6606796/zwavejs_2021-06-07.log)

> 2021-06-07T07:33:29.705Z CNTRLR   [Node 012] [~] [Meter] value[66049]: 9.4 => 10.5                  [Endpoint 1]
> 2021-06-07T07:33:29.705Z SERIAL » [ACK]                                                                   (0x06)
> 2021-06-07T07:33:29.706Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
>                                   └─[MultiChannelCCCommandEncapsulation]
>                                     │ source:      1
>                                     │ destination: 3
>                                     └─[MeterCCReport]
>                                         type:        Electric
>                                         scale:       W
>                                         rate type:   Consumed
>                                         value:       10.5
>                                         time delta:  9 seconds
>                                         prev. value: 9.5

AlCalzone, 2021-06-07:
> │ source: 1
> │ destination: 3

Oh, so the device tells us the report comes from endpoint 1, when in reality it comes from endpoint 3 which it uses as the destination.

katiuskt, 2021-06-07:
Thank you very much for your quick answer! It will be really helpful if it could be managed as you said.

PR #2885 (2021-06-19): feat: add compat flag to treat destination endpoint as source
fixes: #2792
closed 2021-06-19 by AlCalzone