zwave-js/zwave-js #2796: Missing device configuration params for DSC08 AEON Labs / Smart Energy Illuminator.

raym777, 2021-06-07:
**Checklist:**

-   [X] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

As [instructed](https://github.com/zwave-js/node-zwave-js/discussions/2781), providing details for the [DSC08 AEON Labs / Smart Energy Illuminator](https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x0008:0.0) which provides config params **80, 90, 91 and 92** but these are missing from the associated configuration file and can therefore not be set via Z-Wave JS. 
**Device information**

```
  "manufacturer": "AEON Labs",
  "manufacturerId": 134,
  "productType": 3,
  "productId": 8,
  "firmwareVersion": "1.27",
```

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X] **No**

**Did this use to work before?**

-   [X] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X] `HomeAssistant` version core-2021.6.2
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Attempt to set the above missing params in Home Assistant. Error message:
`websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Configuration parameter with value ID 4-112-0-80 could not be found`

-->

AlCalzone, 2021-06-07:
@blhoward2 Do you have some insight here?

blhoward2, 2021-06-08:
Those aren't in the manual provided to me by Aeotec, so they may have been added along the way at some point. I'll email them to verify.

blhoward2, 2021-06-14:
Aeotec confirmed 80, 90-92 exist on this device. They're revising their spec sheet.

raym777, 2021-06-14:
Nice! Thanks a lot for the follow up.

blhoward2, 2021-06-18:
This is incorporated into my Aeotec batch that is in progress.

PR #2896 (2021-06-23): feat(config): template more aeotec devices
Closes #2893
Closes #2796
Closes #2637
Closes #2634
Closed #2664 

After this, I have just three more Aeotec files to go. Bulb 6 Multi-White, Bulb 6 Multi-Color, and Multisensor 7.
closed 2021-06-23 by AlCalzone