zwave-js/zwave-js #2801: Vision ZD2102 door sensor - external switch isn't detected by Home Assistant.

Extreme79, 2021-06-07:
Vision zd2102 door sensor has a main reed and an additional external input where another sensor/reed can be connected.
Unfortunately the external input isn't available in home assistant where I use zwavejs (yes, I already opened an issue on HA core GitHub that [you can read it here](https://github.com/home-assistant/core/issues/51532), where a detailed analysis on the issue is available.
As I understand, a fix by Zwavejs team is needed in order to get it working.

**Device information**

Vision Security ZD2102 door sensor, nodeID 7 in my network

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [X] **No**

**Did this use to work before?**
-   [x ] **No**, it never worked anywhere ( in Ha with Zwavejs).
That sensor, anyway, is working normally in a zipato zipabox controller, since years, with full access to both endpoints.

!-- fill in -->
-   [ ] ioBroker.zwave2 adapter
-   [X] `HomeAssistant` version core-2021.06.2
- -->

**To Reproduce**
Add a vision zd2102 in the network and try to use it in home assistant detecting the open/close for each input.
**Additional context**

Add any o

**Logfile:**
For HomeAssistant, this is how you do it:
Home Assistant -> settings -> Integrations -> Z-Wave JS -> Configure -> Create dump -> zip the json file and post it here.
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6612312/zwave_js_dump.zip)

[node7_log.txt](https://github.com/zwave-js/node-zwave-js/files/6612313/node7_log.txt)

blhoward2, 2021-06-08:
Note that per the comment zwavejs is a secondary controller. I imagine that complicates any troubleshooting...

AlCalzone, 2021-06-08:
Okay from cross-reading the other issue, there are several things that need to be done here:

1\. The device sends the external switch status as an "unknown" notification which contains the V1 alarm values. We ignore the V1 values for V2+ notifications because the Z-Wave specs say so. However we could just not do this for "unknown" notifications, so this part would be an easy fix.

2\. HA doesn't like the "any" metadata for the unknown notification. We don't define any metadata for that event, but I think it's fair to make it a "number" metadata at least.

kpine, 2021-06-08:
Actually it's not only the Unknown sensor that has the "any" metadata, all of the Notification values have this. It seems to happen for V2.

<details>
<summary>Notification dump</summary>

```json
              {
                "endpoint": 0,
                "commandClass": 113,
                "commandClassName": "Notification",
                "property": "Home Security",
                "propertyKey": "Cover status",
                "propertyName": "Home Security",
                "propertyKeyName": "Cover status",
                "ccVersion": 2,
                "metadata": {
                  "type": "any",
                  "readable": true,
                  "writeable": true
                },
                "value": 3
              },
              {
                "endpoint": 0,
                "commandClass": 113,
                "commandClassName": "Notification",
                "property": "Home Security",
                "propertyKey": "Sensor status",
                "propertyName": "Home Security",
                "propertyKeyName": "Sensor status",
                "ccVersion": 2,
                "metadata": {
                  "type": "any",
                  "readable": true,
                  "writeable": true
                },
                "value": 2
              },
              {
                "endpoint": 0,
                "commandClass": 113,
                "commandClassName": "Notification",
                "property": "Home Security",
                "propertyKey": "unknown",
                "propertyName": "Home Security",
                "propertyKeyName": "unknown",
                "ccVersion": 2,
                "metadata": {
                  "type": "any",
                  "readable": true,
                  "writeable": true
                },
                "value": 254
              },
```

</details>

What would be the value for the Unknown notification if it's a number? Is it a static value of 254?

AlCalzone, 2021-06-08:
Please give v7.7.3 a try

Extreme79, 2021-06-09:
I updated the zwave-js integration to v0.1.25 on home assistant, that bumps zwave js 7.7.3.
I rebooted HA.
So I tried to re-interview one of zd2102 I own, but unfortunately this seems not solve the main issue.
Some new entities was created in home assistant, but not related to the secondary switch. 

Later I'll edit this post with some logs to help you to better understand.

kpine, 2021-06-10:
Please post the debug logs from a re-interview. The HA side looks OK based on what's now provided (Home Security sensors have valid metadata types), however there are no V1 values. The interview might explain why.

Extreme79, 2021-06-10:
Here the silly log of the new interview, as I posted also in HA issue side.
[zd2102_new_interview_silly.txt](https://github.com/zwave-js/node-zwave-js/files/6634147/zd2102_new_interview_silly.txt)

and here the additional mini-log of a external switch open (starting with main reed already opened) >> external switch close >> main reed close.
[another_silly_log_zd2102.txt](https://github.com/zwave-js/node-zwave-js/files/6634165/another_silly_log_zd2102.txt)

kpine, 2021-06-10:
```
2021-06-10T19:57:43.498Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[NotificationCCSupportedReport]
                                      supports V1 alarm:            false
                                      supported notification types:
                                      · Home Security
```

The device claims it doesn't support V1 alarms. What a mess of a device.

Extreme79, 2021-06-10:
> ```
> 2021-06-10T19:57:43.498Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
>                                   └─[NotificationCCSupportedReport]
>                                       supports V1 [...]

I wrote to vision security to try the ask about the issue, to ask about a new firmware if available, and I sent the links to the gitub issues (hopeing they'll read..). 
the zd2102 is the unique door sensor I found able to "handle" two inputs within one device.. but as I understand, probably there is a reason if it's the unique...

AlCalzone, 2021-06-10:
> The device claims it doesn't support V1 alarms

Yet it uses them (╯°□°）╯︵ ┻━┻
Well - I'll remove the condition that a device must support V1 alarm for us to preserve them.

Extreme79, 2021-06-27:
I'll check my zd2102 after the latest release as soon as I come back home from holidays (3july).

Extreme79, 2021-07-13:
Now the issue seems solved.
I have an alarmlevel entity that change from 0 (closed) to 255 (opened). So with that value I can detect my secondary switch.
I'll search for a way to convert this entity to a binary sensor, without using templates "disconnected" from the device (can you point me to the right path?), But anyway.. the issue seems solved.