zwave-js/zwave-js #2802: Seemingly random duplication of `MultilevelSwitchCCReport`

kelchm, 2021-06-07:
**Describe the bug**

While trying to chase down some performance problems on my Z-Wave network, I observed a large number of duplicated `MultilevelSwitchCCReport` messages from specific devices. This isn't 100% reproducible and seemingly happens at random.

**Device information**

Which device(s) is/are affected (manufacturer/model)?  
- Inovelli LVW31-SN (firmware v1.56)
- Zooz ZEN27 (firmware v3.3)

What is/are the node IDs?
| Node Id | Model |
| ------- | ------------- |
| 6  | Zooz ZEN27 |
| 8  | Zooz ZEN27 |
| 28  | Inovelli LZW31-SN |
| 29  | Inovelli LZW31-SN |
| 42  | Inovelli LZW31-SN |
| 49  | Zooz ZEN27 |

**How are you using `node-zwave-js`**

-   [x] `zwavejs2mqtt` (Supervisor Add-On v0.18.1)
    -   `node-zwave-js` v7.7.1
    -   `zwavejs2mqtt` v4.5.1
-   [x] `HomeAssistant` core-2021.6.0

**To Reproduce**

The best mechanism I've come up with to reproduce this issue is to repeatedly toggle a group of devices. Sometimes it works absolutely fine, other times a series of ~10 `MultilevelSwitchCCReport` messages show up in a period of less than a second.

**Additional context**

In the attached log file, I'm repeatedly toggling a group of 6 lights _or_ setting this group off when they are already off. If needed, I'm happy to try to capture more logs.

**Logfile:**
- [zwavejs_2021-06-07-7.log](https://github.com/zwave-js/node-zwave-js/files/6612298/zwavejs_2021-06-07-7.log)

blhoward2, 2021-06-08:
Beyond my pay grade. @AlCalzone ? Odd that it happens with two different devices but there isn't much we can do about messages originating from devices.

AlCalzone, 2021-06-08:
This is not only the MultilevelSwitchCCReports, I'm seeing duplicated SupervisionCCReports too. Usually this is an indication of a suboptimal mesh. I don't know exactly what happens, but I think one or more repeaters (and/or the nodes themselves) try a bit too hard to get the message to their destination because they didn't get the ACK from their target.

kelchm, 2021-06-08:
> Unfortunately, this isn't anything we have control over

Thanks for the response @AlCalzone -- I was afraid that was going to be the case. I've tried pretty much everything I can think of at this point (including all of your suggestions).

One thing I will note is that I'm also experiencing intermittent failures when doing a network heal that I've been meaning to open a separate issue for. Much like the issue reported here, sometimes it works 100% fine, sometimes a subset of my network ends up with _failed to update the neighbor list after 5 attempts, healing failed_ (see attached log for an example). 

All the devices on my network are permanently powered, though the majority of them are dimmers/switches which are (unfortunately) installed in metal gang boxes. Interestingly, the devices I seem to be having the most consistent issues with (nodes 2, 6, 8, 49) are the only dimmers/switches which are installed in a plastic gang box. 

With the addition of some dedicated wall plug style repeaters (3x Aeotec Range Extender 7), the mesh at least appears to be fairly strong, but I suppose there's currently no way to know how borderline the links actually are between certain nodes.

[image]

**Attachments:**
- [nodes_dump-2021-06-08.json.txt](https://github.com/zwave-js/node-zwave-js/files/6616667/nodes_dump-2021-06-08.json.txt)
- [zwavejs_2021-06-08.log](https://github.com/zwave-js/node-zwave-js/files/6616675/zwavejs_2021-06-08.log)

AlCalzone, 2021-06-08:
> failed to update the neighbor list after 5 attempts, healing failed

I **think** this is the stick having a problem, but then I'm not very familiar with what happens under the hood there. Maybe it just needs to be power cycled. 🤷🏻‍♂️

You might also have some luck by listening with a Zniffer in different locations (near the controller and near the problematic nodes) and figuring out what happens by looking at those traces.

kelchm, 2021-06-09:
Thanks for sharing your thoughts on this @AlCalzone -- closing this 👍

closed 2021-06-09 by kelchm