zwave-js/zwave-js #2826: Missing alarm mappings for Yale YRD220 lock

vondruska, 2021-06-10:
**Checklist:**

-   [x] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [x] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I have a Yale Z-Wave door lock, Yale YRD220, when the lock is manipulated manually, downstream systems don't get notifications about the changes. Like Home Assistant.

**Device information**

Which device(s) is/are affected (manufacturer/model)?  Yale YRD220 Door Lock

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [x] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [x] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
	-   `node-zwave-js` branch: 
	-   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
	-   `node-zwave-js` branch: 
	-   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
	-   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Open Home Assistant
2. Manipulate lock manually
3. Observe the lock state does not change

**Additional context**

Forcing the lock to lock or unlock programmatically works. And other Z-Wave devices function correctly within the network.

The issue I logged to Home Assistant pointed me here https://github.com/home-assistant/core/issues/51690. As @kpine pointed out in the issue, YRD210 does have the alarm mappings.

**Logfile:**
[zwavejs_2021-06-10.txt](https://github.com/zwave-js/node-zwave-js/files/6633281/zwavejs_2021-06-10.txt)

disforw, 2021-06-10:
Same issue here, I have 3 different models of Yale Zwave door locks. I just merged a PR changing the Primary association group name to “Lifeline” in hopes that it fixes the issue. Let me further explain… 
Events are currently communicating in debug (see screenshot 1) but they are not updating entities downstream.
[image]

In this example:
“113-0-AlarmType > 21” is Lock LOCKED 
“113-0-AlarmType > 22” is Lock UNLOCKED
“113-0-AlarmLevel > 1” is Manually
“113-0-AlarmLevel > 2” is using a code

If someone tells me where I can update these definitions, I dont mind creating the PR.

vondruska, 2021-06-10:
@disforw This is the exact behavior I'm experiencing. Within the HA bug, @kpine pointed out that the [YRD210](https://github.com/zwave-js/node-zwave-js/blob/460a7e962cddaed01e88c975c4fecaab28754e3a/packages/config/config/devices/0x0109/yrd210.json#L59) has the alarmMappings while the [YRD220](https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0129/yrd220.json) does not.

Seems like the alarmType that are apart of the YRD210 line up with YRD220 from what I can tell. But I'm also not finding documentation anywhere specifying the alarmType's. Maybe Yale has standard alarm types across all their locks?

disforw, 2021-06-10:
I have a YRD220, a YRL210, and a YRL220. If someone tells me how to add the alarm mappings, I wouldn't mind doing it.

AlCalzone, 2021-06-10:
@disforw you can take a look at `packages\config\config\devices\0x0129\yrd210.json` to see how it is done:
```json
	"compat": {
		"alarmMapping": [
		    ...
		]
	}
```

The events you mention seem to be in line with what the other Yale locks have - except we don't map the level for types 21/22 to a specific location (Inside/Outside). I don't know if those are consistent between their locks. But when the event comes in you should be able to read the `alarmLevel` and use that to determine where it was locked/unlocked from.

vondruska, 2021-06-10:
I'm happy to test when there's a PR!

AlCalzone, 2021-06-10:
You don't need a PR to test device files now:
https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

Just put the updated file there.

disforw, 2021-06-10:
PR created #2829

closed 2021-06-11 by AlCalzone