zwave-js/zwave-js #2828: MIMO2+ configuration fails, but can be manually set

alarobric, 2021-06-10:
**Checklist:**

-   [X] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

MIMO2+ configuration parameter fails, but can be manually set

The manufacturer documentation (specifically parameters 3 and 9) seem to match with what's in the zwavejs configuration here:
https://devices.zwave-js.io/
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0084/mimo2.json

Manufacturer documentation: https://static1.squarespace.com/static/57363221555986b60e62f0f0/t/5919d919ff7c50802f0b93da/1494866202716/Tech+Appendix+MIMO2%2B+8May2017+removed+MI+address.pdf

But when I try to change parameter 3 from periodic to on-change, or change the interval, I see an error at top-right that it was invalid input.
I came back to this today and found the manufacturer documentation and now the manual configuration option in the zwavejs panel. I've now tried again today setting manually (for instance, parameter 3, size 1, value -24) and it worked! The device took the update, I can see it now behaves differently as I want. 

[image]

So what is wrong in the configuration here that the UI method of setting one of these parameters fails? Am I doing something wrong, or does the configuration need to be tweaked?

Secondly, I note that when setting a parameter manually, the rest of the UI does not get updated. I'm not totally sure, but it updated either on refresh or manually refreshing the node after.

**Device information**

Fortrezz MIMO2+

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [X] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [X] `zwavejs2mqtt` (latest) docker image  Version 4.5.1
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X] `HomeAssistant` version 2021.6.2
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Open Zwavejs2mqtt panel
2. Open Fortrezz MIMO2+ node, Configuration panel
3. Scroll down to parameter "Sensor 1 Report Mode". Change from [0] Periodically to [1] On-change. Send.
4. See error in screenshot. Parameter change is not accepted.

**Additional context**

None

**Logfile:**

Logfile: 
[zwavejs_2021-06-10.log](https://github.com/zwave-js/node-zwave-js/files/6634543/zwavejs_2021-06-10.log)

From UI:
```
2021-06-10 14:28:55.553 INFO APP: GET /api/auth-enabled 304 1.789 ms - -
2021-06-10 14:29:09.625 INFO ZWAVE: Calling api writeValue with args: [
{
nodeId: 13,
commandClass: 112,
endpoint: 0,
property: 9,
propertyKey: 64
},
0,
[length]: 2
]
2021-06-10 14:29:09.628 INFO ZWAVE: Writing 0 to 13-112-0-9-64
2021-06-10 14:29:09.632 INFO ZWAVE: Controller status: Driver: The value 0 is invalid for configuration parameter 9 (size = 1, format = SignedInteger)!
2021-06-10 14:29:09.634 ERROR ZWAVE: Unable to write 0 on 13-112-0-9-64
2021-06-10 14:29:09.635 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
2021-06-10 14:29:13.891 INFO ZWAVE: Calling api writeValue with args: [
{
nodeId: 13,
commandClass: 112,
endpoint: 0,
property: 9,
propertyKey: 64
},
1,
[length]: 2
]
2021-06-10 14:29:13.894 INFO ZWAVE: Writing 1 to 13-112-0-9-64
2021-06-10 14:29:13.899 INFO ZWAVE: Controller status: Driver: The value 1 is invalid for configuration parameter 9 (size = 1, format = SignedInteger)!
2021-06-10 14:29:13.900 ERROR ZWAVE: Unable to write 1 on 13-112-0-9-64
2021-06-10 14:29:13.901 INFO ZWAVE: Success zwave api call writeValue { success: true, message: 'Success zwave api call', result: undefined }
```

AlCalzone, 2021-06-10:
There are two bugs here:
1. The error message shows the incorrect value, leading us down the wrong path. You're setting a partial param, which merges the 0 bit with the existing value. That existing value is incompatible with the definition, not the 0. https://github.com/zwave-js/node-zwave-js/pull/2830

2. The parameter should be defined as `unsigned` but it isn't. https://github.com/zwave-js/node-zwave-js/pull/2831

alarobric, 2021-06-11:
Thanks for the very fast response and fixes too!

Ah 1. makes sense.

I'm not sure I understand for 2. though. If I send through parameter 3 manually, I sent value -24, and that was accepted. That's a signed value...

AlCalzone, 2021-06-11:
Partial parameters are internally treated as unsigned to make the computation easier. Whether you send -24 or 232, the same bits go over the wire. But since the config parameters weren't marked as unsigned, the default (signed) applies. And 232 is not within the range -128..127, so our check failed.

Essentially, it's a technicality - but one that enables us to keep users from having to do bitwise computations before sending a config parameter

closed 2021-06-11 by AlCalzone

AlCalzone, 2021-06-13:
Actually, our handling of signed partial params wasn't correct. https://github.com/zwave-js/node-zwave-js/pull/2846 should fix that part.