zwave-js/zwave-js #2834: Danalock V3-BTZE - device included ok, but not working and no configuration options

MS-Turbo, 2021-06-11:
**Checklist:**

-   [x ] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [x ] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [x ] I have checked the troubleshooting section and my problem is **not** described there.
-   [x ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

https://devices.zwave-js.io/?jumpTo=0x010e:0x0009:0x0001:0.0
Migrated this device from Fibaro to Home assistant. Reset the device and included to HA. Tried same procedure three times. The issue is that the device is not controllable from HA, and the config option on the device is not shown. 

**Device information**

Danalock V3-BTZE, firmware version 0.17.5
Latest ID after several exclude/include runs: 97

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x ] **No**

**Did this use to work before?**

-   [x ] **Don't know**, this is a new device to HA, but worked fine in fibaro
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x ] `HomeAssistant` version 2021.6.3
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...
[zwave_js_dump.zip](https://github.com/zwave-js/node-zwave-js/files/6636710/zwave_js_dump.zip)

'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-06-11:
>             "isSecure": false,

zwave-js thinks the lock isn't included securely, so most functionality just won't work. Without seeing a driver log of the inclusion or a re-interview I cannot tell why that is though. Possible reasons:
- You have no network key configured
- You didn't include the lock securely
- There was a timeout during secure inclusion
- The device wasn't reset correctly, so it still uses the old network key

MS-Turbo, 2021-06-11:
Thanx for the suggestions. I will try them out sunday when i get back home. You can close this issue

closed 2021-06-11 by MS-Turbo