zwave-js/zwave-js #2839: z-wave js integration lost contact with 50 devices after upgrading to Home Assistant OS 6.0

MS-Turbo, 2021-06-11:
**Checklist:**

-   [ ] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X ] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X ] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

Not sure where to post this issue, but after upgrading to Home Assistant OS 6.0 the z-wave js integration lost contact with alot of devices and now only has contact with 26/83. I have tried to re-interview some devices but the interview process fails. Any suggestions on what to try next/ solve this issue?
 
**Device information**

Which device(s) is/are affected (manufacturer/model)?  
What is/are the node IDs?

**Did you change anything?**

-   [X ] **Yes**: _(Upgraded to Home Assistant OS 6.0)_
-   [ ] **No**

**Did this use to work before?**

-   [ ] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [X ] **Yes**, in: _(Before upgrading to Home Assistant OS 6.0)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [X ] `HomeAssistant` version core-2021.6.3
[zwave_js_dump.json.zip](https://github.com/zwave-js/node-zwave-js/files/6640890/zwave_js_dump.json.zip)
[zwave_js_dump.json.zip](https://github.com/zwave-js/node-zwave-js/files/6640892/zwave_js_dump.json.zip)

-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

MS-Turbo, 2021-06-11:
Logs from z-wave js addon/server:

2021-06-11T20:40:13.521Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.522Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.560Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.560Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.614Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.615Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.645Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.646Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.727Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.728Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.741Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:13.742Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:14.239Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:14.239Z DRIVER   Node 76, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.558Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.559Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.616Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.617Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.627Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.627Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.789Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:29.790Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.023Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.024Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.034Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.035Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.061Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.061Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.134Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.135Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.183Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.183Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.251Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.251Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
[... 7 lines cut ...]
                                  nterview
[... cut ...]
                                  nterview
2021-06-11T20:40:30.719Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.951Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview
2021-06-11T20:40:30.952Z DRIVER   Node 28, Endpoint 3: Trying to access endpoint instance before Multi Channel i
                                  nterview

MS-Turbo, 2021-06-11:
Update: rebooted host one more time and now 72/83 devices are up and running. Maybe this issue just fixes itself...

AlCalzone, 2021-06-11:
I can't diagnose this without a driver log.

MS-Turbo, 2021-06-14:
Ok, I will close this and if it happens next time I will post a log with debug level

closed 2021-06-14 by MS-Turbo