zwave-js/zwave-js #2868: Documentation search includes results from zwavejs2mqtt docs

kpine, 2021-06-16:
**Checklist:**

-   [X] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [X] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [X] I have checked the troubleshooting section and my problem is **not** described there.
-   [X] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

When searching the docs website, the results include docs from the zwavejs2mqtt project. Clicking on a result produces a 404 page. I would expect only node-zwave-js docs.

**To Reproduce**

Steps to reproduce the behavior:

1. Open a new private/incognito window so the browser cache is empty
2. Go to the [zwavejs2mqtt docs](https://zwave-js.github.io/zwavejs2mqtt/#/)
3. Search for `hass` in zwavejs2mqtt docs
3. Go to the [node-zwave-js docs](https://zwave-js.github.io/node-zwave-js/#/)
4. Search for `hass` in the node-zwave-js docs
5. See results that are from the zwavejs2mqtt docs
6. Click on a result
7. The result is a 404 page

**Additional context**

Screenshot of mixed results from above steps:

[image]

The problem only occurs _after_ you've visited the zwavejs2mqtt docs. If you go to node-zwave-js docs first and search, the results are fine.

Result with fresh cache, no zwavejs2mqtt:

[image]

AlCalzone, 2021-06-16:
Grml. Looks like the docsify search plugin does not respect the namespace.

AlCalzone, 2021-06-16:
https://github.com/docsifyjs/docsify/issues/1591

closed 2021-06-18 by robertsLando