zwave-js/zwave-js #2881: Yale Config template: rename manual_relock_time and unify the different auto_relock_time templates

AlCalzone, 2021-06-18:
This looks right but that parameter is inconsistent in the template itself. It should be renamed to Auto Relock Time to match the others and the description should be removed. The template reference itself should also change to match the others (and fixed in any other files).

I can look at this later unless someone beats me to it.

_Originally posted by @blhoward2 in https://github.com/zwave-js/node-zwave-js/pull/2842#discussion_r650527635_

asucrews, 2021-07-05:
I have access to zwave Yale locks, what help is needed?

AlCalzone, 2021-07-05:
The locks aren't necessary really. We've only noticed that https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x0129/templates/yale_template.json has a few different definitions for the auto-relock that should be unified. This includes renaming the `manual_relock_time` template there and everywhere it is used.