zwave-js/zwave-js #2883: Add value metadata that indicates that a given value is stateless, e.g. value updates will be received as value notifications

raman325, 2021-06-18:
**Is your feature request related to a problem? Please describe.**
In Home Assistant, we are looking into building a better UI around zwave-js automations. A common use case for automations is to automate off of a Central Scene device which primarily reports button presses through `value notifications`. As I started looking into this, I realized I could determine which devices could potentially send `value notifications` by looking at the Command Classes that are capable of sending stateless values (had to look through the code for this). But even within this Command Class, it seems that some values are still stateful and others are stateless. I started to hardcode rules on a per Command Class and per value basis to identify the stateless values, but that will quickly become unmanageable; I think it would be better for this information to be provided by the driver.

**Describe the solution you'd like**
I'd like a way to look at node state data to determine which nodes and values will send `value notifications`. My latest idea was to add a metadata parameter for values that met this criteria but I am open to alternatives.

**Describe alternatives you've considered**
Hardcoding these rules on a per Command Class and per value basis.

**Additional context**
I am not sure we can make progress on this UI unless we can simplify the discovery logic downstream.

AlCalzone, 2021-06-18:
I've looked into this and we define the `stateful` flag either as part of the `@ccValue` decorator (not the metadata) or when setting a value in the value DB. The former can be queried with the method `isStatefulValue` that is available on each CC instance. The latter can currently not be queried.

IMO this is also something that needs to go into https://github.com/zwave-js/node-zwave-js/issues/552. Otherwise it feels like a band-aid for a band-aid. Until then you'll have to go with hard-coding I fear.

CCs that use the `@ccValue` decorator option:
- Scene Activation CC (scene ID)

CCs that don't
- Central Scene CC (scenes)
- Basic CC (event)

closed 2021-06-18 by AlCalzone

raman325, 2021-06-18:
thanks for the quick follow up. The property for the value is in the parentheses right?

AlCalzone, 2021-06-18:
Yep, but don't quote me on the exact spelling 😅