zwave-js/zwave-js #2887: Popp IP44 Keypad is not sending all codes reliably

Mastiffen, 2021-06-19:
**Checklist:**

-   [x ] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-   [ x] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [ x] I have checked the troubleshooting section and my problem is **not** described there.
-   [ x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

A clear and concise description of what the bug is. Describe what causes the bug, what you observe, and what happens as a result.

All codes are not received. It seems like they are blocked arbitrary. The only code that works all the time is user code 3.

**Device information**

Which device(s) is/are affected (manufacturer/model)?  
What is/are the node IDs?
Popp IP44 keyboard, node ID 11

**Did you change anything?**

-   [x ] **Yes**: _(please describe)_
-   [ ] **No**
I did re-interview. First time the whole user code section dissappeared. Second time it came back.

**Did this use to work before?**

-   [ x] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [ ] `HomeAssistant` version XYZ
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [x ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 

**To Reproduce**

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '...'
3. Scroll down to '...'
4. See error

**Additional context**

Add any other context about the problem here.

**Logfile:**

AlCalzone, 2021-06-19:
> there isn't anything in the log when this happens

Then the messages from the keypad are not reaching the Z-Wave controller. Unreliable communication is often the result of a bad controller placement and/or interference. Try putting it on an USB extension cord, away from metallic surfaces. USB ports (on a RPi for example) are sometimes enough to interfere.

Mastiffen, 2021-06-19:
Thanks, but the keypad is around five meters, with two 15 cm wooden walls and no metal in between, no water pipes and no power cables. The Z-Wave USB device is on an unpowered USB hub around 50 cm from the Pi and anything else that can interfere, and I have a bunch of other stuff around the cabin that has repeater capabilities. It may of course be that the Popp has a ridiculously bad range. I have put a repeating plug-in switch on the inside of the outer wall where the Popp is on the inside and I will see if that helps.

AlCalzone, 2021-06-19:
Distance is not always a factor. I'm not an expert on the radio side of Z-Wave, but from my experience, even moving the controller a few cm can sometimes improve the situation. That being said, maybe it's the Popp that's in a bad spot.

One way to figure that out is by using a Zniffer next to the controller and next to the Popp and to capture the RF traffic that's visible in each location - although that's also no gurarantee the other devices see the same.

If you haven't already, you can also try healing the network.

Mastiffen, 2021-06-19:
I will try to do a new healing. But still with repeaters all around the place (almost all non-battery powered Z-wave devices have the repeater capability, but you most certainly knew that)  and even about two meters from the Popp, with only 15 cm of wood between them the place ment of the actual controller doesn't matter too much. That's the very nice thing about Z-Wave, I have a house with a lot of places were I can't get 433mHz, but Z-Wave works without a hitch. The Namron dimmers are marked as doing that, but it may be that the repeating capabilities are weak.

AlCalzone, 2021-06-19:
All I can do here is try and give you some hints. If the controller doesn't see the messages, it's definitely not a software issue.

Mastiffen, 2021-06-19:
OK, thanks for the help! So even if it sees one reliably (Scene 003) and the other ones not reliably that isn't the software's fault. If you are sure that implementation works I will do some hardware testing with placements of repeaters. :)

AlCalzone, 2021-06-19:
> So even if it sees one reliably (Scene 003) and the other ones not reliably

To be sure: which log are you looking at? The driver log should at least include the raw serial data if something is received.

Mastiffen, 2021-06-19:
I was looking at the debug info in the node. From your question I suspect that's not the log I am looking for.

Mastiffen, 2021-06-19:
Of these only the "Garasjevinsj" should fail because it's not plugged in right now: 
[image]

AlCalzone, 2021-06-19:
> I suspect that's not the log I am looking for.

I'm looking for the one that's requested in the issue template:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

Mastiffen, 2021-06-19:
Aha, I misunderstood something, because I never got to that page. Btw I see that something's wrong here, because even after two heals (which fixed most of the stuff to green) the Popp doesn't seem to have any neighbours. But maybe it's not supposed to?

[image]

I apologize again for misunderstanding, but I have activated the log now and I will be back tomorrow with log, when I have the time to test some more.

closed 2021-06-23 by zwave-js-assistant[bot]

Mastiffen, 2021-06-23:
I have now tested sporadically in three days because I'm totally confused. The healing I did on the Sunday didn't change anything. But the next morning, when I tried to test, all codes worked! And nothing physically has changed, no nodes are moved or anything. Also I have found out that this device isn't supposed to keep track of neighbours because it's considered a "moveable" device, which makes no sense since it's heavy and meant to be bolted to the wall. My Namron 4 channel wall switches do keep track of neigbours, even if they are small and light.

Anyway, I will keep this closed and then get back to it if the messy behaviour starts again.