zwave-js/zwave-js #2893: Aeotec AW111 Dual Nano Dimmer configuration options

ShinyMcGlint, 2021-06-21:
a Home Assistant developer has told me to come here.
https://github.com/home-assistant/core/issues/52011

**Describe the bug**
I have several Aeotec AW111 Dual Nano Dimmers. In the legacy Z-Wave integration I was able to set parameters 123 & 124 to control whether the load was triggered by the physical switch inputs. This functionality appears read-only in Zwave-JS.

It looks like the node-zwave-js/packages/config/config/devices/0x0086/templates/aeotec_template.json file has these parameters set to 'read'. (line 1359 & 1383)

(also there seems to be a typo in the S2 destination field - line 1395 should read "label": "Load and association group 4" rather than 3)

**Device information**
Which device(s) is/are affected (manufacturer/model)?  Aeotec AW111
What is/are the node IDs? 32

**Did you change anything?**
Yes - moved to Zwave-JS from legacy Zwave integration

**Did this use to work before?**
Yes, in: legacy Z-wave integration

**How are you using `node-zwave-js`**
HomeAssistant` core-2021.6.5

**To Reproduce**

Steps to reproduce the behavior:

1. Go to the device page
2. Click on 'Configure Device'
3. Scroll down to '.Control Destonation (S1)'
4. See error 'This parameter is read-only'

**Additional context**
N/A

**Logfile:**
shout if you need the logfile, not sure it's going to help much here!

ShinyMcGlint, 2021-06-21:
Here are the settings from the z-wave back-end manuals webpage
http://manuals-backend.z-wave.info/make.php?lang=en&sku=AEOEZW111&cert=ZC10-17015388

<html>
<body>
<h4 style="font-size: 14px; margin: 10px 0px 0px; color: rgb(51, 51, 51); font-family: Arial, Helvetica, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Parameter 123: Set the control destination for external switch S1</h4><p style="margin: 0px 0px 10px; color: rgb(51, 51, 51); font-family: Arial, Helvetica, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><i>Set the control destination for external switch S1</i><br>Size: 1 Byte, Default Value: 3</p><p style="margin: 0px 0px 10px; color: rgb(51, 51, 51); font-family: Arial, Helvetica, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><xconfigparametervalue></xconfigparametervalue>

Setting | Description
-- | --
1 | control the output loads of itself.
2 | control the other nodes
3 | control the output loads of itself and other nodes.

</p><br class="Apple-interchange-newline">
</body>
</html>

AlCalzone, 2021-06-21:
@blhoward2 are these files on your to do list?

blhoward2, 2021-06-21:
Yes, I've made this change to the template on my end.

PR #2896 (2021-06-23): feat(config): template more aeotec devices
Closes #2893
Closes #2796
Closes #2637
Closes #2634
Closed #2664 

After this, I have just three more Aeotec files to go. Bulb 6 Multi-White, Bulb 6 Multi-Color, and Multisensor 7.
closed 2021-06-23 by AlCalzone