zwave-js/zwave-js #2897: Acceleration for Metered Devices

disforw, 2021-06-22:
**Is your feature request related to a problem? Please describe.**
I have so many automations that wait for a metered plug-in type device to show usage in the amperage or wattage entity (IE: sensor.toaster_oven | float > 0). 

**Describe the solution you'd like**
Until now, I have been creating template binary_sensors that turn on when usage is detected but this method breaks if an entity is renamed and feels a little clunky. Is it possible to create a binary sensor entity that turns on or off based on the load of a metered device? Is this something that is possible to push to every metered device, or is this something I would add in to the Alarm Mapping in a device config file? 

**Additional context**
Below is a device that I physically have/use in my house and can test with. If this is something that needs to be mapped in the config file, I will do so here.
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x021f/dmof1.json

AlCalzone, 2021-06-22:
Sorry, I have read the issue three times now and I still don't understand what you mean.

disforw, 2021-06-22:
I want to have a zwave metered plugin device (such as the Elexa DMOF1) add an additional binary sensor entity to Home Assistant that will show true if it is being used (the device plugged in to it is turned on and drawing power) and will show false if the device is not being used.

AlCalzone, 2021-06-22:
Okay that would be a feature request for home assistant.
In the driver I try to stay close to the specifications and add a convenient API on top.
Doing magic like extra computed values is outside the scope.

closed 2021-06-23 by AlCalzone