zwave-js/zwave-js #2901: [question] Secure Inclusion is successful but Secure shows No?

sofakng, 2021-06-23:
I'm trying to use Secure Inclusion to add some Z-Wave devices to my network.

It appears to be successful because the devices are added but on the control panel under the Secure column they are listed as "No".

Am I misunderstanding the Secure column?

For example, one device is a [GE Outdoor Smart Switch](https://products.z-wavealliance.org/products/2939?selectedFrequencyId=2).  It supports S2 Encryption and I believe when I added this using other Z-Wave hubs I needed to enter a number listed on a sticker inside the box?

However I'm also having the same issue with other devices such as [Zooz ZEN27](https://products.z-wavealliance.org/products/3148?selectedFrequencyId=2) light switches which also support S2.

AlCalzone, 2021-06-23:
>  door sensor say "Unknown" under the Secure column

Please check that the network key is properly set up. This seems to me like it isn't.

And yeah, [driver logs](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs) please :)

smizles, 2021-06-24:
> Please check that the network key is properly configured. This seems to me like it isn't.

The network key is that same one I initially set up the zwavejs add-on with back on 5/24/21, and it's the same one I backed up and successfully used to migrate to zwavejs2mqtt back on 6/1/21. Here is the driver log from the last day my locks responded to remote lock and unlock commands, which is the same day I noticed the "Unknown" status first started appearing later that evening: 
[zwavejs_2021-06-21.log.gz](https://github.com/zwave-js/node-zwave-js/files/6705648/zwavejs_2021-06-21.log.gz)

Also, I have included a lot more information in this zwavejs2mqtt issue I just opened last night in the Home Assistant Core GitHub project: https://github.com/home-assistant/core/issues/52103

sofakng, 2021-06-24:
I'm confused ... @robertsLando said S2 isn't supported but it looks like this issue is being investigated?

smizles, 2021-06-24:
> I'm confused ... @robertsLando said S2 isn't supported but it looks like this issue is being investigated?

I'm new to Zwave, so I'm not sure if my network was running on S0 or S2. However, I am **sure** it was running with some sort of security because the secure command classes of my devices were available for almost a month before this issue started three days ago. Now my devices are barely usable without the advanced features that are only available on a secure Zwave network.

kpine, 2021-06-24:
> 
> 
> I'm confused ... @robertsLando said S2 isn't supported but it looks like this issue is being investigated?

S2 is not supported, therefore a value of "Yes" for the "Secure" column in zwavejs2mqtt indicates the node was included with Security S0.

Newer devices like this GE and Zen27 _only_ support Security S2, and not S0. Some devices support both S0 and S2, but it isn't mandatory. Locks are an exception, the _are_ required to support S0 (at least [until 2020](https://drzwave.blog/2020/11/02/best-practices-for-z-wave-door-locks/), maybe later) for backwards compatibility with "older" gateways.

The problem with the lock is unrelated to this issue.

AlCalzone, 2021-06-24:
@smizles Your log looks relatively normal. Sometimes the devices are a bit overeager in re-requesting a nonce (you're running on S0), which might cause a little bit of delay. To figure out why the nodes's security statuses are shown as "unknown", I'd need to see a log of a startup where it doesn't work anymore.

sofakng, 2021-06-28:
> S2 is not supported, therefore a value of "Yes" for the "Secure" column in zwavejs2mqtt indicates the node was included with Security S0.
>
> Newer devices like this GE and Zen27 _only_ support [...]

Thanks very much for the explanation.  I believe this explains the behavior I'm seeing.  Thanks!

AlCalzone, 2021-06-28:
I've done a bunch of the S2 work already, but some parts are still missing. There are a couple of issues I'll have to tackle first because they make the experience worse for many.

NickHerrig, 2021-07-22:
I also have this behavior with a Direct Wire Indoor/Outdoor smart switch from GE which only supports S2. 

@AlCalzone Let me know If I can help at all with testing devices when the Zwavejs PR completed or up for review.

AlCalzone, 2021-07-22:
I should have enough S2-compatible devices to test, but you can follow the S2 PR to stay updated #1136. Once it is merged, a couple tests won't hurt.

closed 2021-07-26 by zwave-js-assistant[bot]