zwave-js/zwave-js #2904: Kwikset 912 (again)

peng1can, 2021-06-23:
**Which device is missing?**

Enter Manufacturer and Device name here

**What are the IDs?**

Manufacturer ID: 0x0090
Product type: 0x0003
Product ID: 0x0336

**Did you check if we really don't have it?**

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)
- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)
- [X] It is not in the [TODO list](https://github.com/zwave-js/node-zwave-js/issues/1600)

[image]

I'm very sorry.  In my last request on the Kwikset 912, I typo'ed the ProductID. (0x0036 instead of 0x0336).  I have included a screenshot to extra-confirm the numbers this time.

Also, in trying to test this, I have tried editing the 912.json file and changing the 0x0036 to 0x0336, restarted the extension, did a refresh and re-interview, and the device remains unknown (the screenshot is after I did those steps).  Is there a different method for getting the config to apply to an existing device?  Just want to be extra sure I've given you the correct config info.

Thanks!

AlCalzone, 2021-06-24:
To test new device files, follow this guide: https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

Just restarting afterwards should be enough in your case.

To be sure: This change was not correct then?
https://github.com/zwave-js/node-zwave-js/pull/2879/files

peng1can, 2021-06-24:
Yes, the change in #2879 was incorrect for my lock.  I typoed the Product ID.  The Product ID above (0336) and in the screenshot should be correct.  

I see now that OZW has 0036 in their code sets for the Kwikset 912, and I think that's where I got confused when reporting it last time.  It may therefore be worth leaving the other Product ID in as well?

AlCalzone, 2021-06-24:
I don't trust what OZW has in their config files 😅

AlCalzone, 2021-06-24:
If you could provide a PR where you change the config file from 0x0036 to 0x0336, that would be nice.

PR #2999 (2021-07-11): fix(config): correct device fingerprint for Kwikset 912
fixes: #2904
closed 2021-07-11 by AlCalzone