zwave-js/zwave-js #2907: Trouble interviewing Schalge Connect (BE469ZP)

jreed509, 2021-06-24:
**Checklist:**

-   [x] My problem is **not** within Home Assistant or the ZWave JS integration. **Or:** a Home Assistant developer has told me to come here.
-  Home Assistant dev directed me here from [this issue](https://github.com/home-assistant/core/issues/50496)
-   [x] My problem is **not** within ZWaveJS2MQTT. **Or:** a ZWaveJS2MQTT developer has told me to come here.
-   [x] I have checked the troubleshooting section and my problem is **not** described there.
-   [x] I have read the changelog and my problem was **not** mentioned there.

**Describe the bug**

I am unable to fully interview any of my 3 Schlage locks. During the initial inclusion (although sometimes it takes a few tries), _most_ of the lock configuration parameters are accessible. After the initial inclusion, my locks are usable (I can lock and unlock them and their status updates) and some of the usercodes are available through zwave_js. However some parameters (usercodes in particular) are missing (see nodes 24 and 45 for an example).

Re-interviewing seems to make the problem worse as after re-interviewing most configuration parameters are removed (looking at the zwave_js logs, I see lots of "timed out waiting for response from the nodes"). Node 58 is an example of what I see after re-interviewing. 

So far I've tried excluding/including, re-interviewing, healing the network, and factory resetting my locks several times with no change in behavior. I've also tried including/interviewing the lock with it very close to the controller. 

I'm seeing identical behavior on all three locks.

**Device information**

Which device(s) is/are affected (manufacturer/model)?  
* Schlage Connect BE469ZP (z-wave+ version)
What is/are the node IDs?
24, 45, 58

**Did you change anything?**

-   [ ] **Yes**: _(please describe)_
-   [x] **No**

**Did this use to work before?**

-   [x] **Don't know**, this is a new device
-   [ ] **No**, it never worked anywhere
-   [ ] **Yes**, in: _(specify application with versions)_

**How are you using `node-zwave-js`**

-   [ ] `zwavejs2mqtt` (latest) docker image
-   [ ] `zwavejs2mqtt` (dev) docker image
-   [ ] `zwavejs2mqtt` Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] ioBroker.zwave2 adapter
-   [x] `HomeAssistant` version (see below)
-   [ ] Pkg
-   [ ] Manual Docker build
    -   `node-zwave-js` branch: 
    -   `zwavejs2mqtt` branch: 
-   [ ] `node-red-contrib-zwave-js`
    -   Module version (double click node): 
-   [ ] Manually built (as described in the [docs](https://zwave-js.github.io/node-zwave-js/#/development/installing-from-github))
-   [ ] Other: 
## Home Assistant Version Info

version | core-2021.6.6
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 4.19.0-16-amd64
arch | x86_64
timezone | America/New_York

<details><summary>Home Assistant Community Store</summary>

[... 4031 lines cut ...]
              {
                "id": 90,
                "name": "Device Reset Locally",
                "version": 1,
                "isSecure": true
              },
              {
                "id": 135,
                "name": "Indicator",
                "version": 1,
                "isSecure": true
              }
            ],
            "interviewStage": "Complete"
          }
        ]
      }
    }
  }
]

```

</p>
</details>

AlCalzone, 2021-06-25:
I really need a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs), loglevel `debug` or `silly` to investigate this. Since this is going to involve lock codes, you can upload an encrypted archive and send me the password via e-mail (see my profile).

jreed509, 2021-06-25:
Thanks - logs are uploaded below and I sent you an email with the password. 

This zip folder contains two log files: one recorded during the initial inclusion and another during the re-interview. It looks like, this time, in both cases the lock failed to provide usercodes at all. 

Finally, one additional note: I needed to use zwavejs2mqtt to generate this log file, but I typically use the official zwavejs HA integration to control my devices. 

[zwavejs2mqtt_logs.zip](https://github.com/zwave-js/node-zwave-js/files/6716583/zwavejs2mqtt_logs.zip)

AlCalzone, 2021-06-25:
There's definitely something wrong with the communication. The lock is responding with encrypted commands that re-use a previous nonce (one-time key). The first example is when the `Basic CC` version is queried:
```
# zwave-js sends VersionCCGet 
2021-06-25T12:41:25.371Z DRIVER » [Node 059] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      47
                                  └─[SecurityCCCommandEncapsulation]
                                    │ nonce id: 94
                                    └─[VersionCCGet]

# Node requests nonce for the response
2021-06-25T12:41:25.445Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceGet]

# zwave-js responds with nonce ID 0xea
2021-06-25T12:41:25.446Z DRIVER » [Node 059] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      48
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xea437d392184a2ce

# Command is received by node
2021-06-25T12:41:25.454Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-06-25T12:41:26.714Z DRIVER « [REQ] [SendData]
                                    callback id:     48
                                    transmit status: OK

# Node responds with encrypted response to VersionCCGet
2021-06-25T12:41:26.799Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation]
                                    │ sequenced: false
                                    └─[VersionCCReport]
                                        library type:      Routing Slave
                                        protocol version:  6.3
                                        firmware versions: 3.3, 11.0
                                        hardware version:  3

# zwave-js requests CC version for Basic CC
2021-06-25T12:41:26.802Z DRIVER » [Node 059] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      49
                                  └─[SecurityCCNonceGet]
2021-06-25T12:41:26.811Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-06-25T12:41:26.940Z DRIVER « [REQ] [SendData]
                                    callback id:     49
                                    transmit status: OK

2021-06-25T12:41:27.016Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0x263bc669942e9115

2021-06-25T12:41:27.017Z DRIVER » [Node 059] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      50
                                  └─[SecurityCCCommandEncapsulation]
                                    │ nonce id: 38
                                    └─[VersionCCCommandClassGet]
                                        CC: Basic

# Node requests nonce to encrypt response
2021-06-25T12:41:27.243Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceGet]

# zwave-js responds with nonce ID 0xe3
2021-06-25T12:41:27.244Z DRIVER » [Node 059] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      51
                                  └─[SecurityCCNonceReport]
[... 110 lines cut ...]
2021-06-25T12:42:21.798Z SERIAL « 0x01200004003b1898816c14e0121558b3b0d346a524e7ed44a0ae262806db01c90 (34 bytes)
                                  06a
2021-06-25T12:42:21.798Z DRIVER   Dropping message with invalid payload
2021-06-25T12:42:21.798Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation] [INVALID]
2021-06-25T12:42:21.798Z SERIAL » [ACK]                                                                   (0x06)
2021-06-25T12:42:21.870Z SERIAL « 0x01200004003b1898816c14e0121558b3b0d346a524e7ed44a0ae262806db01c90 (34 bytes)
                                  06a
2021-06-25T12:42:21.870Z DRIVER   Dropping message with invalid payload
2021-06-25T12:42:21.870Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation] [INVALID]
2021-06-25T12:42:21.870Z SERIAL » [ACK]                                                                   (0x06)
2021-06-25T12:42:22.003Z SERIAL « 0x01200004003b1898816c14e0121558b3b0d346a524e7ed44a0ae262806db01c30 (34 bytes)
                                  060
2021-06-25T12:42:22.003Z DRIVER   Dropping message with invalid payload
2021-06-25T12:42:22.003Z DRIVER « [Node 059] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation] [INVALID]
```

The normal command flow is:
- Send SecurityCCNonceGet,
- wait for SecurityCCNonceReport,
- send encrypted command with that nonce (once!)

TL;DR: The device is misbehaving. That could be due to a connectivity issue which seems to be the most frequent reason for this spamming. You can try healing the network and/or moving permanently-powered nodes to improve the mesh between the lock and the controller.

jreed509, 2021-06-25:
Thanks for looking into this. I'll see if there's anything I can do to improve the connection, although I already have a fairly dense network of permanently-powered light switches in my network - several of which are physically located between my locks and the controller. 

Do you think it would be worthwhile to move the lock close to the controller during initial inclusion to rule out any signal issues? I'm assuming I could follow this up with a heal network afterward once the lock is returned to the door to correct any routing issues caused by moving it.

AlCalzone, 2021-06-26:
Could be worth a try.

sybohy, 2021-06-29:
We've played with the Schlage's BE469ZP a bunch. It's our experience that this device does NOT return access codes in clear (i.e. returns "******" instead of "123456"). Not sure you're going to be able to solve this one.

jreed509, 2021-06-30:
@sybohy Thanks for the info. Interestingly, I am able to read and edit user codes (the half that show up after a good interview) just fine through the ui when they successfully interview. 

The issue seems to be reliably completing the interview to get all the codes in the first place.

AlCalzone, 2021-06-30:
I'm planning to **not** query the user codes during the interview any more, because that may take excessively long and take a toll on the battery: #2727

That means you'll have to query specific user codes instead of doing the interview over and over. Which you can already do by the way, maybe that will help you.

jreed509, 2021-07-01:
@AlCalzone Ahh that makes sense. I've noticed the interview spends a lot of time on user codes - especially in my case where each request waits the full timeout length waiting for the lock to respond. 

I didn't realize you could query individual parameters - I'll give that a try. Could you point me to instructions on how to query a single usercode? 

I don't think the official Home Assistant add-on has this functionality yet - makes sense because its relatively new, but I don't see that functionality in zwavejs2mqtt either (I just see the option to refresh all values for the node or all existing usercodes, but not a single code in particular).

AlCalzone, 2021-07-01:
zwavejs2mqtt has the ability to let you access an arbitrary API. Here's an example for `startLevelChange`, but it works for any API:
https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=send-command

You'll want to use this API:
https://zwave-js.github.io/node-zwave-js/#/api/CCs/UserCode?id=get
so...
```js
{
  args: [
    {
      nodeId: 1 /* <- your node ID here */,
      commandClass: 0x63,
      endpoint: 0 /* <- probably 0, maybe another one */,
    },
    "get",
    [1 /* <- this is the user ID */],
  ],
}
```

kpine, 2021-07-01:
The usercodes are not provided as values?

sybohy, 2021-07-01:
@jreed509  here's a function I recently wrote to get the node state to update a specific user code. Note that you don't really need the msgOptions persay. I just wanted to be very specific about the polling behavior and the cmd priority (be careful with this). 

```typescript
import {
  UserCodeCCAPI,
  UserIDStatus,
  getUserCodeValueID,
  getUserIdStatusValueID,
} from "zwave-js/build/lib/commandclass/UserCodeCC"

function refreshUserCode(node: ZWaveNode, user_code_id: number) {
  const msgOptions: SendCommandOptions = {
    maxSendAttempts: 1,
    priority: MessagePriority.Normal,
  }
  const userCodeValueId: ValueID = getUserCodeValueID(0, user_code_id)
  const userIdStatusValueId: ValueID = getUserIdStatusValueID(0, user_code_id)
  node.pollValue(userCodeValueId, msgOptions)
  node.pollValue(userIdStatusValueId, msgOptions)
}
```

jreed509, 2021-07-01:
@kpine I may be missing something, but I don't think the `refresh_value` service allows for refreshing individual values. It either refreshes all values for a node (with `refresh_all_values: true`) or only the primary value, which I believe is the lock state (with `refresh_all_values: false`).

Regarding KeyMaster - I actually haven't been able to get KeyMaster working. That's actually what got me started trying to debug the missing usercodes in the first place. Calling `keymaster.refresh_codes` returns an error that `userIdStatus for code slot 1 not found`. I assume this is because zwave_js (the HA integration) can't find any registered usercodes to begin with - that's probably beyond the scope of this issue though.

jreed509, 2021-07-01:
Thanks @AlCalzone & @sybohy - I was able to run this code snippet using the "custom driver function" feature of zwavejs2mqtt to query individual user codes:
```
const nodeNum = 24
const userCodeID = 1;
const endpointNum = 0;

const node = driver.controller.nodes.get(nodeNum);
const endpoint = node.getEndpoint(endpointNum);
const userCodeCCApi = endpoint.commandClasses["User Code"];
if (userCodeCCApi.isSupported()) {
 await userCodeCCApi.get(userCodeID, false);
}
```

This worked great to grab the missing user codes which were then editable in the zwavejs2mqtt ui. Unfortunately, these values don't seem to carry over to the official HA zwave_js addon so I can't actually do anything with them with my current setup. I suppose the official HA addon will need to support querying individual parameters for this to work there.

kpine, 2021-07-01:
> I may be missing something, but I don't think the `refresh_value` service allows for refreshing individual values. It either refreshes all values for a node (with `refresh_all_values: true`) or only the primary value, which I believe is only the lock state (with `refresh_all_values: false`).

You're right. I was thinking `refresh_value` allowed specifying the value ID properties like `set_value` does, but it only allows entities. That's unfortunate.

> Regarding KeyMaster - I actually haven't been able to get KeyMaster working. That's actually what got me started trying to debug the missing usercodes in the first place. Calling keymaster.refresh_codes returns an error that userIdStatus for code slot 1 not found. I assume this is because zwave_js (the HA integration) can't find any registered usercodes to begin with - that's probably beyond the scope of this particular issue though.

Well, I just learned that the `refresh_codes` service call is only for OZW, it doesn't apply to zwave_js. You might try submitting an issue there and asking for the capability? KeyMaster only responds to value updates, and if nothing is triggering a value update, then I suppose it will never see the code values, and you're back to square one. When zwave_js stops automatically requesting user codes, the problem will be noticed by all those users.

jreed509, 2021-07-01:
> that the refresh_codes service call is only for OZW, it doesn't apply to zwave_js
Ahh good catch - yep that explains that error then. I just opened an issue with KeyMaster requesting this functionality as well.

raman325, 2021-07-02:
How about the [`node.refreshCCValues`](https://zwave-js.github.io/node-zwave-js/#/api/node?id=refreshccvalues) command? Will that accomplish the same thing if that were to be called on the User Code CC? We could explore adding a service for that in HA, and we can definitely call that via the library directly from Keymaster in the `refresh_codes` service

AlCalzone, 2021-07-02:
@raman325 it would, but I strongly suggest to query individual user codes on demand, rather than all of them.

raman325, 2021-07-02:
fair enough. The only challenge there is that we would then need to add support for the ConfigurationCC.set command to `zwave-js-server`. This is doable, it just becomes a slippery slope to duplicate the CC specific APIs within `zwave-js-server` which I would prefer not to have to maintain. Something to think about

AlCalzone, 2021-07-02:
> This is doable, it just becomes a slippery slope to duplicate the CC specific APIs within `zwave-js-server` which I would prefer not to have to maintain

Would https://github.com/zwave-js/node-zwave-js/issues/2799 avoid that?

raman325, 2021-07-02:
> > This is doable, it just becomes a slippery slope to duplicate the CC specific APIs within `zwave-js-server` which I would prefer not to have to maintain
> 
> Would #2799 avoid that?
> 
> Edit: Why `ConfigurationCC.set` though? I thought this is about user codes?
> You can also query their individual value IDs with the `pollValue` method.

Sorry meant `UserCodeCC.get`, was thinking about something else as I was writing. With that being said, yes, a function like this would be great! I think this means that we would only have to expose one generic command to cover any CC API function? If so, sign me up!

AlCalzone, 2021-07-14:
Version 8 is going to drop the user code query during the interview, so you shouldn't be seeing this behavior anymore. If you need to read an individual user code, you can do that on demand then.

closed 2021-07-14 by AlCalzone