zwave-js/zwave-js #2908: Zwaveproducts.com/Zlink WD-100   doesn't ever respond with updated values after turning on or off

continuousmining, 2021-06-25:
I have several WD-100 which is the same hardware as the HomeSeer HS-WD100+'s however they don't ever respond with a status after toggling the switch.   It always goes on and the flicks right back to off.  

2021-06-25T01:25:38.613Z SERIAL » 0x010e001308076c0181032601ff2506f9                                  (16 bytes)
2021-06-25T01:25:38.614Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      6
                                  └─[SupervisionCCGet]
                                    │ session id:      1
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 255
2021-06-25T01:25:38.621Z SERIAL « [ACK]                                                                   (0x06)
2021-06-25T01:25:38.628Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-06-25T01:25:38.629Z SERIAL » [ACK]                                                                   (0x06)
2021-06-25T01:25:38.629Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-06-25T01:25:38.648Z SERIAL « 0x011800130600000200c47f7f7f7f000003000000000301000035              (26 bytes)
2021-06-25T01:25:38.649Z SERIAL » [ACK]                                                                   (0x06)
2021-06-25T01:25:38.649Z DRIVER « [REQ] [SendData]
                                    callback id:     6
                                    transmit status: OK
2021-06-25T01:25:38.665Z SERIAL « 0x010d00040008056c02010000c40050                                    (15 bytes)
2021-06-25T01:25:38.667Z SERIAL » [ACK]                                                                   (0x06)
2021-06-25T01:25:38.668Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          1
                                      more updates follow: false
                                      status:              NoSupport
                                      duration:            0s

AlCalzone, 2021-06-25:
Please re-interview the device, create a logfile and **attach** it here as a file.

continuousmining, 2021-06-25:
Here you go.  It is a zwaveproducts.com / ZLink WD-100 Switch

[ZlinkWD-100.log](https://github.com/zwave-js/node-zwave-js/files/6718789/ZlinkWD-100.log)

AlCalzone, 2021-06-25:
Guess I should have been more specific - loglevel **debug** please and upload the actual logfile, not some weird console redirection which messes up all special characters:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

continuousmining, 2021-06-26:
[WD-100 Zwave Logs.zip](https://github.com/zwave-js/node-zwave-js/files/6719429/WD-100.Zwave.Logs.zip)

Here you go!  Thank you!

AlCalzone, 2021-06-26:
@zwave-js-bot add compat flag to `0x0315:0x4447:0x3034:0.0`
```json
"commandClasses": {
	"remove": {
		"0x6c": {
			// The device does not support supervision although it is mandatory for the device type
			"endpoints": "*"
		}
	}
}
```

AlCalzone, 2021-06-26:
@zwave-js-bot add compat flag to 0x0315:0x4447:0x3034:0.0
```json
"commandClasses": {
	"remove": {
		"0x6c": {
			// The device does not support supervision although it is mandatory for the device type
			"endpoints": "*"
		}
	}
}
```

AlCalzone, 2021-06-28:
@continuousmining do you have a chance to test this config file?
https://raw.githubusercontent.com/zwave-js/node-zwave-js/add-compat-flag-1624740612/packages/config/config/devices/0x0315/wd-100.json

The process is described here:
https://zwave-js.github.io/zwavejs2mqtt/#/development/intro?id=testing-custom-devices-config

continuousmining, 2021-06-28:
I tried but how do I do this running Home Assistant ?

AlCalzone, 2021-06-28:
Apparently that's not possible (yet). You'll have to use the zwavejs2mqtt addon.

continuousmining, 2021-06-28:
I am.. but I can’t edit the driver file I create.

AlCalzone, 2021-06-28:
Hmm, can you upload the complete file there?

continuousmining, 2021-06-28:
No, I can't.

AlCalzone, 2021-06-28:
Okay then... I'll merge this and you can report back after the next update.

closed 2021-06-28 by AlCalzone

continuousmining, 2021-06-28:
Awesome, thanks!

AlCalzone, 2021-06-28:
FYI, zwave-js 7.10.0 is **not** the update that includes this.

continuousmining, 2021-06-29:
I got it working!  Thx a lot it works great!

AlCalzone, 2021-06-29:
What did you do? Would be good if I have something to link other HA users to

PR #2914 (2021-06-28): fix(config): remove Supervision support from zwaveproducts.com WD-100
Seems to be an OEM device from HomeSeer, where we have the flag

fixes: #2908