zwave-js/zwave-js #2909: Flag devices that should be re-interviewed

blhoward2, 2021-06-25:
Add metadata for devices that are unknown in the cache allowing the UI to alert the user that a config file is now available and that the device should be reinterviewed.

Consider doing the same for new custom device files (which would require comparing the custom device file to the device cache each time).

AlCalzone, 2021-06-25:
Thinking about this in a broader sense, we could add a maintenance task list for nodes. Points could include:
- [ ] New config file -> re-interview necessary
- [ ] Bad node health (building on the statistics, e.g. when a node has a high error count)
- [ ] Firmware update available (based on the firmware service we've been discussing)
- [ ] ... other ideas?

robertsLando, 2021-06-25:
Hehehe implementing all that points would be awesome! Point 3 require an entire new project to handle firmwares uploads by manufacturers, I can do that but I need to know what specific features we need there

blhoward2, 2021-06-25:
What about flagging that the device spams us with excessive traffic?

robertsLando, 2021-06-25:
Is there a way to get the 'signal' status of a device, I mean how good is the communication with it

AlCalzone, 2021-06-25:
There is - zwave-js doesn't use that infe yet.

blhoward2, 2021-06-25:
How about a known critical flaw that disables the device if we push a flag in a config file? Just as an insurance policy. I’m thinking like on ozw there was a bug that caused all Leviton devices to crash the driver