zwave-js/zwave-js #2910: ZMNHDA Qubino Flush Dimmer multilevel switch current value not propagating from zwavejs to zwavejs2mqtt

NINE78, 2021-06-25:
### Version

Build/Run method

- [X] Docker -> running a Home Assistant supervised add-on

zwavejs2mqtt version: 1.1.0
zwavejs version: 6.1.3

### Describe the bug

For my Qubino (GOAP) ZMNHDA dimmers, I see multilevel switch ("current value") updates sent to Endpoints 0 and 1, but only Endpoint 1 is received in zwavejs2mqtt, this when manually toggling the light switch.

From zwavejs log:
```
15:34:46.396 SERIAL « 0x0109000400400326030094                                            (11 bytes)
15:34:46.398 CNTRLR   [Node 064] [~] [Multilevel Switch] currentValue: 63 => 0          [Endpoint 0]
15:34:46.399 SERIAL » [ACK]                                                                   (0x06)
15:34:46.402 DRIVER « [Node 064] [REQ] [ApplicationCommand]
                      └─[MultilevelSwitchCCReport]
                          current value: 0
15:34:46.407 CNTRLR   [Node 064] [~] [Multilevel Switch] currentValue: 63 => 0          [Endpoint 1]
15:34:47.336 SERIAL « 0x0110000400400a320221340000023b0000bd                              (18 bytes)
```

However in zwavejs2mqtt log, only 38-1 value is updated, which does not get picked up by the "Current Value" entity in HA (there is only one for the light):
`2021-02-10 15:34:46.411 INFO ZWAVE: Node 64: value updated: 38-1-currentValue 63 => 0`

### To Reproduce

Steps to reproduce the behavior:

1. toggle lights
2. observe only endpoint 1 coming through to zwavejs2mqtt in the logs

### Expected behavior

Also expect 64-38-0-currentValue to be updated

### Additional context

Logs attached
[zwavejs2mqtt.log](https://github.com/zwave-js/zwavejs2mqtt/files/5959164/zwavejs2mqtt.log)
[zwavejs_578.log](https://github.com/zwave-js/zwavejs2mqtt/files/5959168/zwavejs_578.log)

AlCalzone, 2021-07-11:
This **should** be fixed with https://github.com/zwave-js/node-zwave-js/pull/2997 in the upcoming 8.x release. If not, feel free to reopen with new logs.

bratanon, 2021-07-25:
Also seeing this issue with all ZMNHDD. 

Currently using z-wave js addon in HA 0.1.32 which includes Z-Wave JS 8.0.5

AlCalzone, 2021-07-25:
Z-Wave JS 8.x includes some changes to endpoints/associations which might require compat flags for some devices. Please share a driver log of a re-interview, so I can use that to figure out which changes are necessary.

bratanon, 2021-07-26:
Is "driver log" the same as I get from the Home Assistant addon for z-wave js?

Is there a way to filter the logs for a specific node? I have a lot of  nodes reporting all the time which clutter the logs.

AlCalzone, 2021-07-26:
Its this one - loglevel "debug" please:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs

> Is there a way to filter the logs for a specific node?

zwavejs2mqtt should let you do that.

bratanon, 2021-07-26:
[node014.log](https://github.com/zwave-js/node-zwave-js/files/6877608/node014.log)

Hope this helps.

bratanon, 2021-07-26:
Not sure if this also helps, but it seems like if i start with what I think is brightness level 50 and turn off the light, its correctly reflected in the UI. However, If i start with brightness level 30, its not working. it will still be turned on in the UI.
[node014_turn_off_30.log](https://github.com/zwave-js/node-zwave-js/files/6877698/node014_turn_off_30.log)
[node014_turn_off_50.log](https://github.com/zwave-js/node-zwave-js/files/6877699/node014_turn_off_50.log)

AlCalzone, 2021-07-26:
@bratanon Please download the logfile as indicated in the documentation I linked you. The re-interview log is just a small chunk and those logs are mixed with the application logs that just create noise.

bratanon, 2021-07-26:
> @bratanon Please download the logfile as indicated in the documentation I linked you. The re-interview log you posted is just a small chunk of the whole thing and those logs are mixed with the application logs that just create noise.

Ok got it. Here is that.
[zwavejs_2021-07-26.log](https://github.com/zwave-js/node-zwave-js/files/6878646/zwavejs_2021-07-26.log)

AlCalzone, 2021-07-26:
Ok, the interview log looks fine. The 50->0 log also looks fine, but in the 40->0 log the intermediate values are not sent by the device after the step to 30.

Honestly, this looks more like those reports either get lost or the device doesn't send them sometimes. I don't think we can realistically do something about it. zwave-js already polls if the device does not report on its own, but in this case it does (once).

You could try to use a Zniffer next to the device to figure out if it is actually sending the reports. 
If it is, you have a connectivity issue you should try and fix by re-arranging or adding some mains-powered device and/or healing the network.
If it doesn't, asking the Qubino support for help would be my recommentation

robertsLando, 2021-10-15:
You should be able to do it from associations dialog (try to write down endpoint manually if not listed), if not driverFunction is the only alternative

AlCalzone, 2021-10-15:
> try to write down endpoint manually if not listed

That's what I wanted to hear :)

fbietlot, 2021-10-15:
Thank you so much for taking the time to help me.
I have read a little bit about association and I try to add the association via the advanced driver function (I made some prinstcreen as it could be helpfull for newbie like me ;-) )
`const source = {
nodeId: 10,
endpoint: 1
}
const groupId = 1

const ass = {
nodeId: 1,
endpoint: 1
}
await driver.controller.addAssociations(source, groupId, [ass])`

Unfortunately I receive an error sayng that endpoint 1 does'nt exist for node10 .
Actually, I think that it's what I'm trying to do : add the endpoint 1 :-( (if it makes sense)

I **tryied** to check in the node dump .
[nodes_dump.json.txt](https://github.com/zwave-js/node-zwave-js/files/7354109/nodes_dump.json.txt)
I see somethink like : 
`endpointsCount | 2`
but can't find any reference to endpoint1

I must admit I'm not sure the issue is coming from zwavejs. But I have no clue how I can investigate more. 
I read about zwave pc controller. I don't know if it can help but I will try to understand more

</body>
</html>

[image]
[image]
[image]

robertsLando, 2021-10-15:
@AlCalzone

robertsLando, 2021-10-18:
Seems you have followed all the steps described in: https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files

Have you also tried to stop and restart application process? 

cc @AlCalzone

AlCalzone, 2021-10-18:
The problem is that we recently switched the config file format. You copied a new file which is not yet understood by the driver version installed on your system.
If @robertsLando pushes a new update of zwavejs2mqtt, you should be able to use that to load the new file.

AlCalzone, 2021-10-18:
Regarding Qubino's response
> The command class that you use 37 [...] The command class 37 refers to COMMAND_CLASS_HRV_STATUS 0x37

They are confusing decimal 37 (hexadecimal `0x25`) with hexadecimal `0x37` (decimal 55).

blhoward2, 2021-10-18:
The only way to use a built docker image is by installing the Portainer addon, or you may be able to install the ssh addon and manually run docker (I'm not sure on that one). Otherwise he's limited to the zwavejs addon or the zwavejs2mqtt addon.

fbietlot, 2021-10-19:
Me again but with good news this time ;-)
@QubinoHelp : your suggestion was actually good for marameter 100 ;-)
@AlCalzone there is a confusion in the config file packages/config/config/devices/0x0159/zmnhtd.json

Here is the config

```
{
			"#": "100",
			"label": "Enable / Disable endpoints relays",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 3,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "EP IR external relay and External relay disabled",
					"value": 0
				},
				{
					"label": "EP IR external relay dsbld, External relay enbld",
					"value": 1
				},
				{
					"label": "EP IR external relay enbld, External relay dsbld",
					"value": 2
				},
				{
					"label": "EP IR external relay and External relay enabled",
					"value": 3
				}
			]
		}
```

Versus the manual

> Parameter no. 100 – Enable / Disable endpoints IR external relay (BICOM) and TRIAC External
> relay (IKA)
> Enabling IR external relay (BICOM) and Triac External relay (IKA) or both of them, means that
> endpoint (IR external relay) and endpoint (External relay) or both will be present on UI. Disabling
> them will result in hiding endpoints according to Parameter set value. Note that hiding endpoint
> has no impact on its functionality.
> Values (size is 1 byte dec):
> • default value 0
> • 0 - Endpoints IR external relay and External relay disabled
> • 1 - Endpoints IR external relay (BICOM) enabled, Triac External relay (IKA) disabled
> • **2- Endpoints IR external relay (BICOM) disabled, Triac External relay (IKA) enabled**
> • 3- Endpoints IR external relay and External relay enabled

Value 1 and 2 are inverted.

commit 10e53aff (2021-07-13): feat: rework how endpoints and lifeline associations are handled (#2997)

fixes: #2286
fixes: #2257
fixes: #1989
fixes: #2714
fixes: #1860
fixes: #2910