zwave-js/zwave-js #2916: [bug] Danalock square BTZW125 - wrong device config loaded

kfazon, 2021-06-28:
After adding this danalock lock it loads the wrong device config. The link to zwavejs db on Device ID
270-2-3 (0x010e-0x0003-0x0002) is ok and there is a right config in zwavejs database. But the loaded cofig is from an unknown lock

AlCalzone, 2021-06-28:
@kfazon we need some more info to diagnose. What config gets loaded?

kfazon, 2021-06-28:
I do not know from witch lock is it.. Here is the code from debug info.. 
Here is the link from what should have been loaded (https://devices.zwave-js.io/?jumpTo=0x010e:0x0003:0x0002:2.4)

This is loaded:

{
  "id": 9,
  "name": "Vjezbaona Vrata Lock",
  "loc": "Vjezbaona",
  "values": [
    {
      "id": "9-98-0-currentMode",
      "nodeId": 9,
      "commandClass": 98,
      "commandClassName": "Door Lock",
      "endpoint": 0,
      "property": "currentMode",
      "propertyName": "currentMode",
      "type": "number",
      "readable": true,
      "writeable": false,
      "label": "Current lock mode",
      "stateless": false,
      "min": 0,
      "max": 255,
      "list": true,
      "states": [
        {
          "text": "Unsecured",
          "value": 0
        },
        {
          "text": "UnsecuredWithTimeout",
          "value": 1
        },
        {
          "text": "InsideUnsecured",
          "value": 16
        },
        {
          "text": "InsideUnsecuredWithTimeout",
          "value": 17
        },
        {
          "text": "OutsideUnsecured",
          "value": 32
        },
        {
          "text": "OutsideUnsecuredWithTimeout",
          "value": 33
        },
        {
          "text": "Unknown",
          "value": 254
        },
        {
          "text": "Secured",
          "value": 255
        }
      ],
      "isCurrentValue": true,
      "targetValue": "98-0-targetMode"
    },
    {
      "id": "9-98-0-targetMode",
      "nodeId": 9,
      "commandClass": 98,
      "commandClassName": "Door Lock",
      "endpoint": 0,
      "property": "targetMode",
[... 375 lines cut ...]
  "protocolVersion": 2,
  "nodeType": 1,
  "endpointsCount": 0,
  "endpointIndizes": [],
  "isSecure": false,
  "supportsSecurity": false,
  "supportsBeaming": true,
  "isControllerNode": false,
  "isListening": false,
  "isFrequentListening": "1000ms",
  "isRouting": true,
  "keepAwake": false,
  "maxDataRate": 40000,
  "deviceClass": {
    "basic": 4,
    "generic": 64,
    "specific": 2
  },
  "deviceId": "270-2-3",
  "status": "Alive",
  "interviewStage": "Complete",
  "lastActive": 1624864564246,
  "healProgress": "done",
  "_name": "Vjezbaona Vrata Lock (Vjezbaona)"
}

kfazon, 2021-06-28:
And this should be in config lines..
https://github.com/zwave-js/node-zwave-js/blob/master/packages/config/config/devices/0x010e/btzw125.json

blhoward2, 2021-06-28:
A log of the interview would be much more helpful. Please do so at the debug level and then post.

kfazon, 2021-06-28:
I started reinterview and this is debug of this node during reinterview

2021-06-28 15:07:28.362 INFO ZWAVE: Calling api refreshInfo with args: [ 9, [length]: 1 ]
2021-06-28 15:07:28.380 INFO ZWAVE: Node 9: interview started
2021-06-28 15:07:28.384 INFO ZWAVE: Success zwave api call refreshInfo undefined
2021-06-28 15:07:28.409 INFO ZWAVE: Node 9: interview stage PROTOCOLINFO completed
2021-06-28 15:07:29.680 INFO ZWAVE: Node 9: interview stage NODEINFO completed
2021-06-28 15:07:39.864 INFO ZWAVE: Node 9: value added: 114-0-manufacturerId => 270
2021-06-28 15:07:39.867 INFO ZWAVE: Node 9: value added: 114-0-productType => 3
2021-06-28 15:07:39.868 INFO ZWAVE: Node 9: value added: 114-0-productId => 2
2021-06-28 15:07:39.961 INFO ZWAVE: Node 9: value added: 134-0-libraryType => 3
2021-06-28 15:07:39.962 INFO ZWAVE: Node 9: value added: 134-0-protocolVersion => 2.78
2021-06-28 15:07:39.963 INFO ZWAVE: Node 9: value added: 134-0-firmwareVersions => 2.4
2021-06-28 15:07:40.440 INFO ZWAVE: Node 9: value added: 128-0-level => 0
2021-06-28 15:07:40.442 INFO ZWAVE: Node 9: value added: 128-0-isLow => false
2021-06-28 15:08:00.543 INFO ZWAVE: Node 9: interview stage COMMANDCLASSES completed
2021-06-28 15:08:00.545 INFO ZWAVE: Node 9: interview stage OVERWRITECONFIG completed
2021-06-28 15:08:00.577 INFO ZWAVE: Node 9: interview stage NEIGHBORS completed
2021-06-28 15:08:00.579 INFO ZWAVE: Node 9: interview stage COMPLETE completed
2021-06-28 15:08:00.585 INFO ZWAVE: Node 9: value added 9-98-0-currentMode => undefined
2021-06-28 15:08:00.585 INFO ZWAVE: Node 9: value added 9-98-0-targetMode => undefined
2021-06-28 15:08:00.587 INFO ZWAVE: Node 9: value added 9-98-0-outsideHandlesCanOpenDoor => undefined
2021-06-28 15:08:00.587 INFO ZWAVE: Node 9: value added 9-98-0-insideHandlesCanOpenDoor => undefined
2021-06-28 15:08:00.588 INFO ZWAVE: Node 9: value added 9-98-0-latchStatus => undefined
2021-06-28 15:08:00.588 INFO ZWAVE: Node 9: value added 9-98-0-boltStatus => undefined
2021-06-28 15:08:00.589 INFO ZWAVE: Node 9: value added 9-98-0-doorStatus => undefined
2021-06-28 15:08:00.589 INFO ZWAVE: Node 9: value added 9-98-0-lockTimeout => undefined
2021-06-28 15:08:00.589 INFO ZWAVE: Node 9: value added 9-98-0-operationType => undefined
2021-06-28 15:08:00.590 INFO ZWAVE: Node 9: value added 9-98-0-outsideHandlesCanOpenDoorConfiguration => undefined
2021-06-28 15:08:00.590 INFO ZWAVE: Node 9: value added 9-98-0-insideHandlesCanOpenDoorConfiguration => undefined
2021-06-28 15:08:00.590 INFO ZWAVE: Node 9: value added 9-98-0-lockTimeoutConfiguration => undefined
2021-06-28 15:08:00.591 INFO ZWAVE: Node 9: value added 9-114-0-manufacturerId => 270
2021-06-28 15:08:00.591 INFO ZWAVE: Node 9: value added 9-114-0-productType => 3
2021-06-28 15:08:00.592 INFO ZWAVE: Node 9: value added 9-114-0-productId => 2
2021-06-28 15:08:00.592 INFO ZWAVE: Node 9: value added 9-134-0-libraryType => 3
2021-06-28 15:08:00.592 INFO ZWAVE: Node 9: value added 9-134-0-protocolVersion => 2.78
2021-06-28 15:08:00.593 INFO ZWAVE: Node 9: value added 9-134-0-firmwareVersions => 2.4
2021-06-28 15:08:00.593 INFO ZWAVE: Node 9: value added 9-128-0-level => 0
2021-06-28 15:08:00.594 INFO ZWAVE: Node 9: value added 9-128-0-isLow => false
2021-06-28 15:08:00.595 INFO ZWAVE: Node 9 ready: Poly-control - BTZW125 (Danalock v2 circle)
2021-06-28 15:08:00.599 INFO ZWAVE: Node 9: interview COMPLETED, all values are updated

blhoward2, 2021-06-28:
That's not the debug level. And please post it in a file and attach it. No one can read pages long, unformatted posts.

AlCalzone, 2021-06-28:
I need the [driver logs](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/bug_report?id=driver-logs) to be sure, but this looks terribly like you have no network key configured (or a different one than on the lock), or included the lock insecurely.

kfazon, 2021-06-29:
I fixed by secure add.. It was not secure added because it did not go. Tried over 10 times and then added insecure. in 1st attempt.. 
I excluded it and now It succeeded to secure add on 6th or 7th attempt.. 

Thank you for your time

closed 2021-06-29 by AlCalzone