zwave-js/zwave-js #2923: MCOHome A8-9

digitalambiance, 2021-06-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

MCOHome A8-9

### Manufacturer ID

MCOHome

### Product Type

Air Quality, Multiple Sensor

### Product ID

A8-9

### Firmware Version

All Versions

### Do you have a manual?

[A8-9.pdf](https://github.com/zwave-js/node-zwave-js/files/6735341/A8-9.pdf)

AlCalzone, 2021-06-30:
We need the Manufacturer ID, Product Type, Product ID as hexadecimal numbers like the template showed you:
[image]

digitalambiance, 2021-06-30:
Hey Al,
I'm struggling to find these hex values.  Where would they be listed?  Can
I contact the manufacturer about it?  Please lmk where this info is
generally found.  Thanks man

AlCalzone, 2021-06-30:
Which application are you using? It should show you, otherwise it can be found in the driver logs during the interview.

digitalambiance, 2021-07-01:
Got it, I'll add these details now.  Just found them in the ZWaveJS config

digitalambiance, 2021-07-01:
Hex values added.  Firmware version is unknown.  I'll ask the manufacturer about this....  Let me know if there's anything else you need from my side.

AlCalzone, 2021-07-01:
> Firmware version is unknown

We can start with the default range of "everything", that's okay

AlCalzone, 2021-07-01:
@zwave-js-bot import config 3995 from zwa

AlCalzone, 2021-07-01:
@digitalambiance The linked PR can serve as a starting point, but the parameters need to be cleaned up a bit to match our [style guide](https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide).

If you like, you can give it a shot (you can create a PR on the PR branch import-config-1625159522) - that will speed things up.

digitalambiance, 2021-07-06:
Hey Al,
Never done this before so I can't promise that it's going to turn out
"correctly" but I'll give it a shot!  Maybe a situation where I end up
asking a lot of simple questions tho >_<

AlCalzone, 2021-07-06:
That's fine, we'll try to help you

AlCalzone, 2021-07-08:
Sorry I mixed this one up - I had already started on the cleanup. The PR is now merged

closed 2021-07-08 by AlCalzone

digitalambiance, 2021-07-08:
AMAZING! so that device will be incorporated into the next update?  Thanks
so much man!

AlCalzone, 2021-07-08:
Yep

digitalambiance, 2021-07-15:
Hey Al, wondering when this update is anticipated to push?  Not trying to
rush anyone, I just need to update the ppl I'm working with and get a sense
of when we'll be able to start working with this sensor.

Thanks man!

AlCalzone, 2021-07-16:
This is included in v7.12.0 and most likely in the config update `7.11.1-20210709`

PR #2931 (2021-07-08): feat(config): add config file for MCOHome A8-9
fixes: #2923

**TODO:**
- [x]	Change PR title to be more specific