zwave-js/zwave-js #2924: Heatit ZM Single Relay 16A

henrikeri, 2021-06-29:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Heatit ZM Single Relay 16A

### Manufacturer ID

0x019b

### Product Type

0x0004

### Product ID

0x3500

### Firmware Version

1.0

### Do you have a manual?

[1651.pdf](https://github.com/zwave-js/node-zwave-js/files/6735554/1651.pdf)

henrikeri, 2021-07-12:
@AlCalzone, could this potentially be auto-generated from the zwave alliance db as it exists there (https://products.z-wavealliance.org/products/4062)?

AlCalzone, 2021-07-12:
Yeah we can do that. The resulting file needs to be manually cleaned up to match our style guide though. If you want to help out, you can make a PR onto the PR our bot is going to create in a minute.

@zwave-js-bot import config 4062 from zwa

henrikeri, 2021-07-12:
@AlCalzone, done. Although bear in mind it's my first PR if I messed something up :)

AlCalzone, 2021-07-12:
I've left you a few comments in the PR

PR #3004 (2021-07-19): feat(config): add Heatit ZM Single Relay 16A
Adds a config file for Heatit ZM Single Relay 16, and thus fixes #2924. It is based on the autogenerated config in #3003.

closes: #3003
closed 2021-07-19 by AlCalzone