zwave-js/zwave-js #2938: [question] ZWaveError: Failed to send the command after 3 attempts

clinoid, 2021-07-02:
Having a lot of issues with my z wave network. Frequently it will start up ok, and then after a while it will abruptly stop working, and I get this error. When this happens, _nothing_ on the network will work in terms of commands. It still receives notifications from sensors, but it is unable to turn on any lights/fans/etc.

This is what I see in the debug window.

2021-07-01 12:30:49.249 ERROR ZWAVE-SERVER: Z-Wave error Failed to send the command after 3 attempts (Status Fail)
ZWaveError: Failed to send the command after 3 attempts (Status Fail)
at Driver.sendMessage (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:2271:23)
at Driver.sendCommand (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:2405:28)
at Driver.trySendCommandSupervised (/usr/src/app/node_modules/zwave-js/src/lib/driver/Driver.ts:2494:15)
at MultilevelSwitchCCAPI.set (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/MultilevelSwitchCC.ts:172:47)
at Proxy.MultilevelSwitchCCAPI.<computed> (/usr/src/app/node_modules/zwave-js/src/lib/commandclass/MultilevelSwitchCC.ts:311:33)
at ZWaveNode.setValue (/usr/src/app/node_modules/zwave-js/src/lib/node/Node.ts:752:14)
at Function.handle (/usr/src/app/node_modules/@zwave-js/server/dist/lib/node/message_handler.js:19:44)
at Object.node (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:37:91)
at Client.receiveMessage (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:91:99)
at WebSocket.<anonymous> (/usr/src/app/node_modules/@zwave-js/server/dist/lib/server.js:44:45

I am running the zwavejs2mqtt docker on an unraid system. The usb stick is the zooz 700 stick.

A reboot will seem to fix the issue for a little bit, but I am frequently right back where I started. I have tried to heal the network, I have rebooted, unplugged the USB and plugged it back in. Sometimes the issue will go away on its own. Any ideas?

AlCalzone, 2021-07-02:
According to the specifications, Status **Fail**:
> Indicates that the network is busy (jammed).

There's something interfering with your stick so it cannot send anything. You could try putting it on an extension cord, away from metallic surfaces. Apparently, the USB ports themselves can be enough to cause interference.

0rsa, 2021-07-09:
Same issue here with my z-wave network (happening after a couple of hours/days)
I use a conbee 2 for zigbee and I have the impression that both are starting to malfunction at the same time, maybe a way to explore.
I will try to get some logs next days.

kelchm, 2021-07-14:
Interesting anecdote that may be helpful to others -- despite using an extension cable, I was experiencing this on a daily basis when using my UZB-7 with an Intel NUC. I've since moved the Z-Wave stick it to a dedicated Raspberry Pi 4 in a different location and have not experienced this a single time since.

closed 2021-07-14 by AlCalzone