zwave-js/zwave-js #2973: Issue with Greenwave PowerNode 6 on first port

Aohzan, 2021-07-03:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Try to turn on the port 1

What do you observe?
Port 1 status report that it is already on but no power on the outlet. I have to turn it off and on directly again

What did you expect to happen?
Port 1 status is off, and I can turn it on

Steps to reproduce the behavior:
Watch the switch entity of the first outlet

### Device information

Manufacturer:  GreenWave
Model name: PowerNode 6
Node ID in your network: 16

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.28
Z-Wave JS 7.10.0
Z-Wave JS Server 1.8.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Update to 7.7.4 or 7.9.0, I don't know on which version it started to bug

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Ok with version before

### Attach Driver Logfile

```2021-07-03T19:45:50.194Z CNTRLR   [Node 016] Timed out while waiting for a response from the node
2021-07-03T19:45:50.236Z CNTRLR   [Node 016] Mapping unsolicited report from root device to first supporting end
                                  point #1```

Aohzan, 2021-07-03:
When I turn off outlet 2->6, the status of outlet 1 become off, but the outlet is still on.
So I turn on and off the switch of outlet 1, and 10-15s later, the outlet turn really off.
The log :
```
2021-07-03T20:12:37.242Z SERIAL « 0x01180004001012600d010632022134000002180009000001e25e              (26 bytes)
2021-07-03T20:12:37.244Z CNTRLR   [Node 016] [~] [Meter] value[66049]: 48.2 => 53.6                 [Endpoint 6]
2021-07-03T20:12:37.244Z SERIAL » [ACK]                                                                   (0x06)
2021-07-03T20:12:37.246Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      6
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       W
                                        rate type:   Consumed
                                        value:       53.6
                                        time delta:  9 seconds
                                        prev. value: 48.2
2021-07-03T20:12:38.741Z SERIAL « 0x01180004001012600d01013202213400000023000900000036b5              (26 bytes)
2021-07-03T20:12:38.743Z CNTRLR   [Node 016] [~] [Meter] value[66049]: 5.4 => 3.5                   [Endpoint 1]
2021-07-03T20:12:38.745Z SERIAL » [ACK]                                                                   (0x06)
2021-07-03T20:12:38.746Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       W
                                        rate type:   Consumed
                                        value:       3.5
                                        time delta:  9 seconds
                                        prev. value: 5.4
2021-07-03T20:13:19.907Z SERIAL » 0x010e00131007600d00062501002560ff                                  (16 bytes)
2021-07-03T20:13:19.918Z DRIVER » [Node 016] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      96
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 6
                                    └─[BinarySwitchCCSet]
                                        target value: false
2021-07-03T20:13:19.924Z SERIAL « [ACK]                                                                   (0x06)
2021-07-03T20:13:19.933Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-07-03T20:13:19.935Z SERIAL » [ACK]                                                                   (0x06)
2021-07-03T20:13:19.939Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-07-03T20:13:19.948Z SERIAL « 0x01050013600089                                                     (7 bytes)
2021-07-03T20:13:19.951Z SERIAL » [ACK]                                                                   (0x06)
2021-07-03T20:13:19.954Z DRIVER « [REQ] [SendData]
                                    callback id:     96
                                    transmit status: OK
2021-07-03T20:13:19.962Z CNTRLR   [Node 016] [~] [Binary Switch] currentValue: true => false        [Endpoint 6]
2021-07-03T20:13:20.968Z SERIAL » 0x010d00131006600d000625022561ff                                    (15 bytes)
2021-07-03T20:13:20.971Z DRIVER » [Node 016] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      97
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 6
                                    └─[BinarySwitchCCGet]
2021-07-03T20:13:20.977Z SERIAL « [ACK]                                                                   (0x06)
2021-07-03T20:13:20.987Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2021-07-03T20:13:20.990Z SERIAL » [ACK]                                                                   (0x06)
2021-07-03T20:13:21.001Z DRIVER « [RES] [SendData]
                                    was sent: true
2021-07-03T20:13:21.012Z SERIAL « 0x01050013610088                                                     (7 bytes)
2021-07-03T20:13:21.012Z SERIAL » [ACK]                                                                   (0x06)
2021-07-03T20:13:21.013Z DRIVER « [REQ] [SendData]
                                    callback id:     97
                                    transmit status: OK
[... 198 lines cut ...]
2021-07-03T20:14:01.264Z DRIVER « [Node 016] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 0
                                    └─[BinarySwitchCCReport]
                                        current value: false
[... cut ...]
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       W
                                        rate type:   Consumed
                                        value:       0
                                        time delta:  9 seconds
                                        prev. value: 0
2021-07-03T20:14:11.251Z CNTRLR   [Node 016] Timed out while waiting for a response from the node
```

Zucht, 2021-07-04:
Experiencing exactly the same after updating  the Home Assistant Z-Wave JS Add-on to 0.1.28

AlCalzone, 2021-07-04:
Fixed on master in https://github.com/zwave-js/node-zwave-js/pull/2919
This is going into the next patch release of the driver.

The device is sending invalid multi channel commands, so we needed to add a workaround. The behavior before the update was also not correct btw.

AlCalzone, 2021-07-04:
Duplicate of #2917

closed 2021-07-04 by AlCalzone